pyvisa/pyvisa #377: How do you query but expect a binary response?

jason-s, 2018-11-13:
What's the right way to query but expect a binary response? I don't know the length so I can't use `read_bytes`... This code used to work with an earlier version of PyVISA a couple of years ago, now it's broken because of a UnicodeDecodeError. All I want to do is query but obtain a binary string in response.

```
>>> oscope.ask("*IDN?") # Query the IDN properties of the scope
u'AGILENT TECHNOLOGIES,MSO-X 3034A,MY53510287,02.37.2014052001\n'

>>> wfm_ascii = oscope.query(":WAV:DATA?") # Get the waveform data
---------------------------------------------------------------------------
UnicodeDecodeError                        Traceback (most recent call last)
<ipython-input-4-13c026ae81c6> in <module>()
----> 1 wfm_ascii = oscope.query(":WAV:DATA?") # Get the waveform data

c:\app\python\anaconda\2\lib\site-packages\pyvisa\resources\messagebased.pyc in query(self, message, delay)
    567         if delay > 0.0:
    568             time.sleep(delay)
--> 569         return self.read()
    570 
    571     def ask(self, message, delay=None):

c:\app\python\anaconda\2\lib\site-packages\pyvisa\resources\messagebased.pyc in read(self, termination, encoding)
    411         if termination is None:
    412             termination = self._read_termination
--> 413             message = self._read_raw().decode(enco)
    414         else:
    415             with self.read_termination_context(termination):

UnicodeDecodeError: 'ascii' codec can't decode byte 0xbc in position 10: ordinal not in range(128)
```

jason-s, 2018-11-13:
Oh, never mind, it looks like this works:

    oscope.write(":WAV:DATA?")
    data = oscope.read_raw()

closed 2018-11-13 by jason-s

MatthieuDartiailh, 2018-11-13:
I would suggest you use `query_binary_values` that takes care of converting the binary data and can even directly store them in a numpy array.