pyvisa/pyvisa #378: No Device Found

martin2250, 2018-11-15:
Hi,
I'm trying to use pyvisa with a Tek DPO4104B over USB on Archlinux.
I installed the packages `ni-visa` (17.0.0-3) and `python-pyvisa` (1.9.1-1) from the AUR, added my user to the two groups recommended by ni-visa and have rebooted since the install. pyvisa still fails to list the device.

`python -m visa info`:
```
Machine Details:
   Platform ID:    Linux-4.19.1-arch1-1-ARCH-x86_64-with-arch
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.7.1
   Compiler:       GCC 8.2.1 20180831
   Bits:           64bit
   Build:          Oct 22 2018 10:41:28 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: /opt/ni-visa/usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
```

`lsusb -v`:
```
Bus 001 Device 008: ID 0699:0405 Tektronix, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0699 Tektronix, Inc.
  idProduct          0x0405 
  bcdDevice            3.22
  iManufacturer           2 Tektronix
  iProduct                3 DPO4104B
  iSerial                 4 C030141
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          5 Standard
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      3 Test and Measurement
      bInterfaceProtocol      1 TMC
      iInterface              6 USBTMC-USB488
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               2
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

```
python -m visa shell

Welcome to the VISA shell. Type help or ? to list commands.

(visa) list
(visa) 
```

```
sudo python -m visa shell
[sudo] password for martin: 
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3.7/site-packages/visa.py", line 55, in <module>
    visa_main()
  File "/usr/lib/python3.7/site-packages/visa.py", line 46, in visa_main
    shell.main('@' + args.backend if args.backend else '')
  File "/usr/lib/python3.7/site-packages/pyvisa/shell.py", line 427, in main
    VisaShell(library_path).cmdloop()
  File "/usr/lib/python3.7/site-packages/pyvisa/shell.py", line 104, in __init__
    self.resource_manager = ResourceManager(library_path)
  File "/usr/lib/python3.7/site-packages/pyvisa/highlevel.py", line 1535, in __new__
    obj.session, err = visa_library.open_default_resource_manager()
  File "/usr/lib/python3.7/site-packages/pyvisa/ctwrapper/functions.py", line 1227, in open_default_resource_manager
    ret = library.viOpenDefaultRM(byref(session))
  File "/usr/lib/python3.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
```
Does any one else have similar issues or a solution to this problem?
Thanks!

MatthieuDartiailh, 2018-11-17:
Looking at the error message it looks like Visa does not manage to create the resource manager. Did you restart after installing NI-VISA ?

martin2250, 2018-11-18:
> I installed the packages `ni-visa` (17.0.0-3) and `python-pyvisa` (1.9.1-1) from the AUR, added my user to the two groups recommended by ni-visa and have rebooted since the install.

MatthieuDartiailh, 2018-11-19:
Sorry for missing that. The only other issue in which AUR was discussed is https://github.com/pyvisa/pyvisa/issues/272. But it seems you did everything well.
I will try to run a test on a VM again, just in case.

MatthieuDartiailh, 2020-09-06:
Closing for lack of activity (for which I am to blame). In order to run on Linux, some users have had good results with Keysight which however requires a specific kernel version. Feel free to re-open if necessary.

closed 2020-09-06 by MatthieuDartiailh