pyvisa/pyvisa #385: visa.ResourceManager() got function 'viOpen' not found in Windows

tobegit3hub, 2018-12-13:
Now we use `pyvisa` to load the visa library in Windows.

But when we call `visa.ResourceManager()` or `visa.ResourceManager(r'C:\Windows\System32\visa32.dll')`, it throws exception of "AttributeError: function 'viOpen' not found".

MatthieuDartiailh, 2018-12-13:
Could you please post the output of `pyvisa-info` or `python -m visa info` ?

tobegit3hub, 2018-12-14:
Thanks for your reply @MatthieuDartiailh . I got this error when running `python -m visa info`.

[image]

Should I change to use the pyvisa-py backend?

MatthieuDartiailh, 2018-12-14:
How did you install Visa ? It looks like your install is broken somehow. I would suggest uninstalling Visa and reinstalling. And be sure to restart your computer after re-installing.

tobegit3hub, 2018-12-17:
Thanks @MatthieuDartiailh . We install `pyvisa` with `pip` and download `visa` from <https://www.keysight.com> .

It does not work after rebooting.

MatthieuDartiailh, 2018-12-17:
Could you post the full output of `python -m visa info` ? I would need to know what bitness is your OS and your python install.

tobegit3hub, 2018-12-18:
Thanks and here is the complete log.

[image]

MatthieuDartiailh, 2018-12-18:
Ok now I am lost. Could you explain in details your use case ?

tobegit3hub, 2018-12-18:
Sorry for the lack of detailed explaination. We have to use cases.

1. Run the Python script to use `pyvisa` directly in Windows. We did not get the error message like that.
2. Use Jenkins to run the script in agent server which is the same server above. It does not work with the same command in the same server. It's straight and I'm not sure if it is related to `pyvisa` itself.

MatthieuDartiailh, 2018-12-18:
I am not familiar with Jenkins and so I do not have any idea why it would cause one function to appear missing ... 
By curiosity could you post the output of `pyvisa-info` when run directly (not through Jenkins) ?

arthur00, 2019-05-30:
I'm seeing the same issue - similar conditions: keysight NI on Windows using Python 3.6.8 and pyvisa 1.9.1

MatthieuDartiailh, 2019-05-30:
Thanks for the update. I am starting to wonder if it may be related to using a 32 bits version. Could you post the traceback you get when running ?
```python
import visa
rm = visa.ResourceManager()
print(rm.list_resources())
```

arthur00, 2019-05-30:
I suspected the same thing, so I was already installing the 64 bit version to try it out. Turns it works on 64 bit Python. As far as I'm concerned, this ticket can be closed - but it would be nice if pyvisa caught this issue with a more descriptive message by checking if the library can be opened first. It is very confusing since the dll is called visa32.dll. That's on keysight I guess.

Falling back to another adapter might also be helpful - I don't want to uninstall Keysight VISA since I use it for other purposes, but I also have Python 32 bit on my Python projects, so I can't just upgrade to 64 at this point. 

Here is the traceback:
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 1526, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 1498, in open_visa_library
    return cls(argument)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 97, in __new__
    raise OSError('Could not open VISA library:\n' + '\n'.join(errs))
OSError: Could not open VISA library:
function 'viOpen' not found

MatthieuDartiailh, 2019-05-30:
Can you try pointing pyvisa to the dll located in SysWoW64 instead of system32 ? I am wondering if somehow the detection of the bitness is defective.

arthur00, 2019-05-30:
Tried it, same error (I replaced "library_path" list in the debugger to match SysWoW64 path)
SyntaxError: invalid syntax
Traceback (most recent call last):
  File "C:\Users\arthur.valadares\dev\speedit\rpl_instrument\__main__.py", line 73, in OnInit
    win = MainWindow(frame, Log())
  File "C:\Users\arthur.valadares\dev\speedit\rpl_instrument\rpl_autotest.py", line 97, in __init__
    self.current_test = self.create_test_type(self.mode)
  File "C:\Users\arthur.valadares\dev\speedit\rpl_instrument\rpl_autotest.py", line 788, in create_test_type
    return mode_class[test_type](self, self.output_folders[test_type], test_name)
  File "C:\Users\arthur.valadares\dev\speedit\rpl_instrument\auto_optspec.py", line 56, in __init__
    self.rm = visa.ResourceManager()
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 1526, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 1498, in open_visa_library
    return cls(argument)
  File "C:\Users\arthur.valadares\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\highlevel.py", line 97, in __new__
    raise OSError('Could not open VISA library:\n' + '\n'.join(errs))
OSError: Could not open VISA library:
function 'viOpen' not found

MatthieuDartiailh, 2019-05-30:
Thanks for testing. Why do you have a SyntaxError in your traceback ?

This situation is weird. Could you just try to directly pass the path to the ResourceManager ? In case the debugger cause some issue.

hany1024, 2020-06-03:
I don't know if the thread is continuing but have you tried down-grading to 3.6.0? 
I've done that from 3.8v and got it to work. Make sure to go check with Keysight IO Libraries IDE whether it correctly identify the connection through USB link.

MatthieuDartiailh, 2020-09-07:
Closing for lack of activity. Feel free to reopen if necessary. 

Note that this kind of issue can be caused by a bad Python install or a bad library install. On Windows and Python 3.8 if the dll is not in system 32 one should manually add the path to it using `os.add_dll_directory` or the relevant entry in .pyvisarc.

closed 2020-09-07 by MatthieuDartiailh

ar5000, 2020-09-24:
I just ran into the same issue. I'm running 3.8.5 64-bit in a dedicated venv.

If I open the Keysight Connection Expert, I can see and communicate with the device with no errors.

See the output below-
>>> import pyvisa as visa
>>> rm = visa.ResourceManager()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "E:\Documents\Python\DMM\lib\site-packages\pyvisa\highlevel.py", line 3014, in __new__
    obj.session, err = visa_library.open_default_resource_manager()
  File "E:\Documents\Python\DMM\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1871, in open_default_resource_manager
    ret = library.viOpenDefaultRM(byref(session))
  File "E:\Documents\Python\DMM\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "E:\Documents\Python\DMM\lib\site-packages\pyvisa\highlevel.py", line 249, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_LIBRARY_NFOUND (-1073807202): A code library required by VISA could not be located or loaded.

MatthieuDartiailh, 2020-09-24:
That looks like an issue related to the new dll mechanism access on Python 3.8.  However I have a windows system running 3.8 and Keysight with no issue. It may still be an installation issue though. If you do not want to re-install you can try to use `os.add_dll_directory` to add the Keyisght directory found under Program files.

amurzeau, 2020-11-17:
I get this too, on 32 bits Windows 8.1 installation, with 32 bits `python 3.8.6`, `PyVISA==1.11.3` and `Keysight IO Libraries suite`.
Doing `os.add_dll_directory(r"C:\Program Files\Keysight\IO Libraries Suite\bin")` fixes the issue for me.

Maybe it might still work for some OS/python version combination or when we are upgrading python instead of a fresh install.
https://docs.python.org/3/library/os.html#os.add_dll_directory
https://docs.python.org/3/whatsnew/3.8.html#bpo-36085-whatsnew

I guess there is no other workaround for this outside of using `os.add_dll_directory(r"C:\Program Files\Keysight\IO Libraries Suite\bin")`.

MatthieuDartiailh, 2020-11-18:
You can also set this in a configuration file as described in the documentation but yes as of Python 3.8 this is how it has to be done on Windows at least in some cases.

wolfg-2020, 2021-01-13:
Hi, I have a similar error working with pyvisa. 
I'm using a Keysight GPIB adapter 82357B to connect a Keitheley DMM model 2000. I have installed its GPIB driver. With NI-Max or Keysight connection expert I can actually read the serial number of the mentioned instrument. But i have never successfully connect it via pyvisa. 

my os:
Microsoft Windows [Version 10.0.18363.1256]
with pip list command I can see:
Package           Version
----------------- ---------
-ip               20.3.1
certifi           2020.12.5
chardet           3.0.4
idna              2.10
numpy             1.19.4
pandas            1.1.5
pip               20.3.3
PyMeasure         0.8.0
PyQt6             0.1
pyqtgraph         0.11.0
pyserial          3.5
python-dateutil   2.8.1
pytz              2020.4
PyVISA            1.10.0
requests          2.25.0
setuptools        51.1.0
six               1.15.0
typing-extensions 3.7.4.3
urllib3           1.26.2
visa              1.0.0
wheel             0.36.2

python on my computer:
Python 3.8.3 (tags/v3.8.3:6f8c832, May 13 2020, 22:20:19) [MSC v.1925 32 bit (Intel)] on win32
>>>import platform
>>>platform.architecture()
('32bit', 'WindowsPE')

Here is the lines I tried today after going through the above comments for this topic:

import pyvisa as visa
rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")
os.add_dll_directory("C:/Program Files/Keysight/IO Libraries Suite/bin")
os.add_dll_directory("C:/Program Files (x86)/Keysight/IO Libraries Suite/bin")
rm.list_resources()

But I still got this Attribute Error: function 'viOpen' not found.

Details as:
Traceback (most recent call last):
  File "D:/Python/myProjects/InstrumentCont/check_visa.py", line 5, in <module>
    rm = visa.ResourceManager("C:/Windows/System32/visa32.dll")
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pyvisa\highlevel.py", line 1551, in __new__
    visa_library = open_visa_library(visa_library)
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pyvisa\highlevel.py", line 1523, in open_visa_library
    return cls(argument)
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pyvisa\highlevel.py", line 109, in __new__
    obj._init()
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 144, in _init
    functions.set_signatures(self.lib, errcheck=self._return_handler)
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 168, in set_signatures
    apply("viOpen", [ViSession, ViRsrc, ViAccessMode, ViUInt32, ViPSession], required=True)
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 95, in _internal
    set_signature(library, function_name, argtypes, restype, errcheck_)
  File "D:\Python\myProjects\InstrumentCont\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 241, in set_signature
    func = getattr(library, function_name)
  File "D:\Python\Python38-32\lib\ctypes\__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "D:\Python\Python38-32\lib\ctypes\__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'viOpen' not found

I've been struggling with this problem for quite a long time, and would you please give me some advice ? Thank you so much!

MatthieuDartiailh, 2021-01-13:
The calls to os.add_dll_directory should cipe before opening the resource manager since opening the resource manager access the dll.

wolfg-2020, 2021-01-14:
Yes, you're amazing~~ I've connected to the instrument according to your advice. It has been a pain in the neck. Thank you so much!

MatthieuDartiailh, 2021-01-14:
Happy I could help. I will try to keep in mind that I could clarify the docs.