pyvisa/pyvisa #386: Wrapper not found: No package named pyvisa-py

HITTIH, 2019-01-02:
hello, I encountered this problem too,and I tried the above methods,and it didn't work ,But once I run it the way below, it works.
```
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> eload = rm.open_resource('TCPIP0::10.44.90.19::INSTR')
>>> eload.query('*IDN?')
'Chroma,63210A-150-1000,63210AL00087,1.25,1.01,1.01\n'
```
if I run it in this way "python xxxx.py"(which has the same code ), it didn't work and return “Wrapper not found: No package named pyvisa-py”。And it's no use doing what I did with another issue post.
this is the output of python -m visa info：
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      x86 Family 6 Model 94 Stepping 3, GenuineInt

Python:
   Implementation: CPython
   Executable:     C:\Users\m2717\AppData\Local\Programs\Python
on.exe
   Version:        3.6.1
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 21 2017 17:54:52 (#v3.6.1:69c0db5)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

hgrecco, 2019-01-02:
It might be that the version or the environment are different depending on how you run things.

Please do the following. In your script, add the following lines before anything else:
```python
from pyvisa import util
util.get_debug_info()
```
run it and post the output.

Then execute the same lines in your interpreter and post the output

The purpose is to compare between these two (and also to the otput that you have pasted already)

HITTIH, 2019-01-02:
I'm sorry I wrote '@PY' in my code instead of '@py',I changed it and then worked. Thank you for your kindly help.