pyvisa/pyvisa #387: No USB instrument appearing

masbab, 2019-01-07:
Hi ,
I am trying to use pyvisa with Agilent DSA91304A over USB on Centos 6.10. I installed **VISA** packages NI-VISA-17.0.0 and pyvisa 1.8. 

python -m visa info
Machine Details:
   Platform ID:    Linux-2.6.32-754.9.1.el6.x86_64-x86_64-with-centos-6.10-Final
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.6.6
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-17)
   Bits:           64bit
   Build:          Aug 18 2016 15:13:37 (#r266:84292)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

But I am dont see the oscilloscope on list_resources 

>>> import visa
>>> rm = visa.ResourceManager()
>>> rm.list_resources()
(u'ASRL1::INSTR',)

Does anybody has solution to this problem?
Thanks

MatthieuDartiailh, 2019-01-08:
Have a look at the Linux section of this page: http://www.ni.com/tutorial/4478/en/, it may fix your issue.

masbab, 2019-01-24:
Thanks for reply!
I tried that link before but for linux users it does not say much. I checked /proc/bus/usb/devices access and it seems ok.

MatthieuDartiailh, 2019-01-24:
The NI package is still experimental. Can you access your resource using pyvisa-py (Creating the resource manager with '@py' as argument) ?

masbab, 2019-01-31:
InstrOsc = rm.get_instrument('USB0::10893::36865::MY51500127::0::INSTR')
>>> print(InstrOsc.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 334, in read
    raw_write(req)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 227, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
>>> InstrOsc = rm.open_resource('USB0::10893::36865::MY51500127::0::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/highlevel.py", line 1644, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/resource.py", line 203, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/lib/python2.6/site-packages/pyvisa/highlevel.py", line 1601, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/sessions.py", line 213, in __init__
    self.after_parsing()
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 201, in after_parsing
    self.parsed.serial_number)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 256, in __init__
    super(USBTMC, self).__init__(vendor, product, serial_number, **kwargs)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 186, in __init__
    raise Exception('failed to set configuration\n %s' % e)
Exception: failed to set configuration
 [Errno 16] Resource busy

masbab, 2019-01-31:
Do I need pyusb? At the moment I dont have pyusb installed!

Regards

masbab, 2019-01-31:
Now I dont know what special thing happened now even without using @py I can see the instrument
rm=visa.ResourceManager()
>>> rm.list_resources()
(u'USB0::0x2A8D::0x9001::MY51500127::INSTR', u'ASRL1::INSTR')

MatthieuDartiailh, 2019-01-31:
According to `pyvisa-info` you do have pyusb. That is weird. Does ni works now ?

masbab, 2019-01-31:
I installed pysub lately. Now I also uninstalled pyvisa-py. I am just having the following 
python -m visa info
Machine Details:
   Platform ID:    Linux-2.6.32-754.9.1.el6.x86_64-x86_64-with-centos-6.10-Final
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.6.6
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-17)
   Bits:           64bit
   Build:          Aug 18 2016 15:13:37 (#r266:84292)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904

masbab, 2019-01-31:
rm=visa.ResourceManager()
>>> rm.list_resources()
(u'USB0::0x2A8D::0x9001::MY51500127::INSTR', u'ASRL1::INSTR')
>>> InstrOsc = rm.get_instrument('USB0::0x2A8D::0x9001::MY51500127::0::INSTR')
>>> print(InstrOsc.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/lib/python2.6/site-packages/pyvisa/ctwrapper/functions.py", line 1864, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "/usr/lib/python2.6/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>> InstrOsc.write("*RST")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/lib/python2.6/site-packages/pyvisa/ctwrapper/functions.py", line 1864, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "/usr/lib/python2.6/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2019-02-04:
Getting a timeout on a write is unusual. Can you communicate with your instrument using NI-Max ?

masbab, 2019-02-15:
Hi Matthieu,
I did not try with NI_MAX but I played around with keysight command expert in Windows  and used SCPI commands which is working fine. In linux my current environment setup look like this

python -m visa info
Machine Details:
   Platform ID:    Linux-2.6.32-754.9.1.el6.x86_64-x86_64-with-centos-6.10-Final
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.6.6
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-17)
   Bits:           64bit
   Build:          Aug 18 2016 15:13:37 (#r266:84292)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

I started with basic python commands 

>>> import visa
>>> rm=visa.ResourceManager('@py')
>>> rm.list_resources()
(u'USB0::10893::36865::MY51500127::0::INSTR',)
>>> InstrOsc = rm.get_instrument('USB0::10893::36865::MY51500127::0::INSTR')
>>> print(InstrOsc.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 334, in read
    raw_write(req)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 227, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out

I don't know if the pyvisa version I am using is bug free or not?

Thanks in advance.

MatthieuDartiailh, 2019-02-15:
Surprisingly now you are seeing a timeout when trying to read back the instrument response not when writing the query like with NI-VISA. My usual advice would be to check the manual of the instrument to determine the proper read and write termination characters (which can be set using the `read_termination` and the `write_termination` attribute of the resource) and see if that fixes the issue.

masbab, 2019-02-18:
@MatthieuDartiailh 
I have tried termination characters too but did not help

>>> import visa
>>> rm=visa.ResourceManager('@py')
>>> rm.list_resources()
(u'USB0::10893::36865::MY51500127::0::INSTR',)
>>> InstrOsc = rm.open_resource('USB0::0x2A8D::0x9001::MY51500127::INSTR', write_termination='\r', read_termination='\r')
>>> print(InstrOsc.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 334, in read
    raw_write(req)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 227, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out

Results with different termination character 

>>> import visa
>>> rm=visa.ResourceManager('@py')
>>> rm.list_resources()
(u'USB0::10893::36865::MY51500127::0::INSTR',)
>>> InstrOsc = rm.open_resource('USB0::0x2A8D::0x9001::MY51500127::INSTR', write_termination='\n', read_termination='\n')
>>> print(InstrOsc.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 334, in read
    raw_write(req)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 227, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out

>>> import visa
>>> rm=visa.ResourceManager('@py')
>>> rm.list_resources()
(u'USB0::10893::36865::MY51500127::0::INSTR',)
>>> InstrOsc = rm.open_resource('USB0::0x2A8D::0x9001::MY51500127::INSTR', write_termination='\r\n', read_termination='\r\n')
>>> print(InstrOsc.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/usr/lib/python2.6/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/usr/lib/python2.6/site-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 334, in read
    raw_write(req)
  File "/usr/lib/python2.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 227, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out

Thanks

masbab, 2019-02-19:
I am curious if USB drivers are creating problem, here is the message log

Feb 19 15:26:11 cobralabpc-3 kernel: usb 2-1: New USB device found, idVendor=2a8d, idProduct=9001
Feb 19 15:26:11 cobralabpc-3 kernel: usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 19 15:26:11 cobralabpc-3 kernel: usb 2-1: Product: DSO91304A
Feb 19 15:26:11 cobralabpc-3 kernel: usb 2-1: Manufacturer: Keysight Technologies
Feb 19 15:26:11 cobralabpc-3 kernel: usb 2-1: SerialNumber: MY51500127
Feb 19 15:26:11 cobralabpc-3 kernel: usb 2-1: configuration #1 chosen from 1 choice
Feb 19 15:26:11 cobralabpc-3 kernel: usb 2-1: ep 0x1 - rounding interval to 128 microframes, ep desc says 255 microframes
Feb 19 15:26:11 cobralabpc-3 kernel: usb 2-1: ep 0x82 - rounding interval to 128 microframes, ep desc says 255 microframes
Feb 19 15:31:21 cobralabpc-3 kernel: usb 2-1: ep 0x1 - rounding interval to 128 microframes, ep desc says 255 microframes
Feb 19 15:31:21 cobralabpc-3 kernel: usb 2-1: ep 0x82 - rounding interval to 128 microframes, ep desc says 255 microframes
Feb 19 15:31:21 cobralabpc-3 kernel: usb 2-1: reset high speed USB device number 6 using xhci_hcd
Feb 19 15:31:21 cobralabpc-3 kernel: xhci_hcd 0000:00:14.0: Setup ERROR: setup context command for slot 5.
Feb 19 15:31:22 cobralabpc-3 kernel: usb 2-1: reset high speed USB device number 6 using xhci_hcd
Feb 19 15:31:22 cobralabpc-3 kernel: xhci_hcd 0000:00:14.0: Setup ERROR: setup context command for slot 5.
Feb 19 15:31:22 cobralabpc-3 kernel: usb 2-1: reset high speed USB device number 6 using xhci_hcd
Feb 19 15:31:22 cobralabpc-3 kernel: xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88021814c4c0
Feb 19 15:31:22 cobralabpc-3 kernel: xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88021814c508
Feb 19 15:31:22 cobralabpc-3 kernel: xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88021814c550
Feb 19 15:31:22 cobralabpc-3 kernel: usb 2-1: ep 0x1 - rounding interval to 128 microframes, ep desc says 255 microframes
Feb 19 15:31:22 cobralabpc-3 kernel: usb 2-1: ep 0x82 - rounding interval to 128 microframes, ep desc says 255 microframes

Thanks!

lilichen0, 2021-01-10:
@MatthieuDartiailh 

I  want to use Pyvisa to control the Keithley instrument in Linux, I have got the device name by NI-MAX in windows platform.
but when I run "rm.open_resources('name')", the error is "no device found", rm.list_resources() also can not list any USB device. I have tried usb.core.find(), the device can be seen. I do not know why Pyvisa can not list it, I also tried "rm=pyvisa.ResourceManager('@py')", it still does not work.

MatthieuDartiailh, 2021-01-10:
Can you post the output of `pyvisa-info` so that we get more context ? 

You can also look at https://github.com/pyvisa/pyvisa-py/blob/master/pyvisa_py/protocols/usbutil.py for how pyvisa-py identify USBTMC devices. Also did you check the documentation regarding udev rules ? https://pyvisa-py.readthedocs.io/en/latest/installation.html#usb-resources-usb-instr-raw those can affect the listed instruments.

lilichen0, 2021-01-11:
@MatthieuDartiailh
the output is:

/home/dell/anaconda3/lib/python3.6/site-packages/visa.py:23: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  FutureWarning,
/home/dell/anaconda3/lib/python3.6/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-4.15.0-130-generic-x86_64-with-debian-buster-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/dell/anaconda3/bin/python
   Version:        3.6.4
   Compiler:       GCC 7.2.0
   Bits:           64bit
   Build:          Jan 16 2018 18:10:19 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available

lilichen0, 2021-01-11:
>>> rm.list_resources()
('ASRL/dev/ttyS0::INSTR',)
>>> rm=pyvisa.ResourceManager('@py')
>>> rm.list_resources()
('ASRL/dev/ttyS0::INSTR',)