pyvisa/pyvisa #388: list_resources() doesn't list the device but open_resources works

gomelody, 2019-01-11:
I am writing some script to control Tektronix oscilloscope. With USB connected scope, it works perfectly. 
However when I change to a scope connected with Ethernet cable. 
list_resources() doesn't list the device though it shows up in TekVisa or NIVisa manager. 
With typing TCPIP address, open_resources works. 
Surprisingly, running the script with another laptop works. 
I put my script below. Thanks for your time. 

import visa
import sys

rm = visa.ResourceManager()
visa_addresses = rm.list_resources()
print(visa_addresses)

for device in range (len(visa_addresses)):
    try:
        scope = rm.open_resource(visa_addresses[device])

        if 'TEKTRONIX' in scope.query("*IDN?"):
            print('device found')
            scope_address = visa_addresses[device]
            scope.close()
            break

    except:
        print('Failed to open %s' % visa_addresses[device])
        if device == len(visa_addresses) - 1:
            input('No Scope found. Press any key to exit')
            exit()
        continue
    scope.close()

scope = rm.open_resource(scope_address)

MatthieuDartiailh, 2019-01-11:
This is surprising. Could you post the output of `pyvisa-info` ? Also try making Ni-VISA the primary visa implementation for TCPIP. I have some bad experience with TekVisa in the past.

gomelody, 2019-01-12:
Thanks for your reply! Sorry, I just checked my NI MAX and found it was not listed there. I wrongly assume when TekVisa recognize it, NIVisa would do it either. I just add the device on NI MAX. My script works now! 

it's interesting. The other laptop which worked doesn't even have python and NI MAX installed. It has only TekVisa and run the executive file changed from the script. 

Below is my pyvisa-info output: 

`Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 69 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\program files (x86)\python37-32\python.exe
   Version:        3.7.1
   Compiler:       MSC v.1915 32 bit (Intel)
   Bits:           32bit
   Build:          Oct 20 2018 14:05:16 (#v3.7.1:260ec2c36a)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 18874368
         Spec. Version: 5244928
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.0.2). Backend: N/A
      USB RAW: Available via PyUSB (1.0.2). Backend: N/A
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
`

MatthieuDartiailh, 2019-01-12:
I guess that NiVISA was the primary VISA on the system where both exists. I am surprised that Tek does better than Ni on this case. Is the instrument using the socket protocol (address ends in SOCKET) or vxi11 (address ends in INSTR) ? Could you also post the instrument model for future references ?

gomelody, 2019-01-12:
It's TDS3012C V4.26 using VXI11 protocol.

closed 2019-04-09 by MatthieuDartiailh