pyvisa/pyvisa #390: AttributeError: 'GPIBInstrument' object has no attribute 'flush'

KM4YRI, 2019-02-05:
## Problem

The `flush()` method cannot be called on a `GPIBInstrument` object

I can do this from Labview, and the official NI VISA documentation says that viFlush() is valid for the following resource classes: `GPIB INSTR`, `GPIB INTFC`, `Serial INSTR`, `TCPIP INSTR`, `TCPIP SOCKET`, `VXI INSTR`, `VXI SERVANT`

From looking at the pyvisa code, it appears as though the only resource class that has `flush()` implemented is asynchronous serial.

## Steps to reproduce

```
import visa
rm = visa.ResourceManager()
session = rm.open_resource('GPIB0::5::INSTR')
session.flush()
```

returns this error:

```
Traceback (most recent call last):
session.flush(0b00100000)
AttributeError: 'GPIBInstrument' object has no attribute 'flush'
```

## Proposed fix

Implement `flush()` by calling ``viFlush()`` within the `GPIBInstrument` class, and any other objects in the list above where it's safe to be called.

If you'd like that in the form of a pull request I can handle that, but wanted to ensure that this is indeed a bug and not a mis-interpretation on my end.

MatthieuDartiailh, 2019-02-05:
Indeed this is missing from most of the resources even though the library object does wrap the proper method.
Yes a PR would be welcome.

KM4YRI, 2019-02-05:
Awesome, thanks for the quick reply.  I'll try to throw something together tonight.  I'm most familiar with GPIB and TCP/IP when it comes to VISA, and those are the only ones that I have equipment for testing purposes, besides serial.  So I'll probably just implement those and not the VXI.  In my experience it's better to leave things unimplemented, then implemented but untested.

MatthieuDartiailh, 2019-02-05:
VXI is not widely used with PyVISA (at least I never heard of anybody using it), so it is fine to skip the implementation.

KM4YRI, 2019-02-06:
I've tested it and flush() works with the following mask values on GPIB devices:

When I went back and tried to see why I was trying to send a flush in the first place, it was only sending it for serial devices, not GPIB.  So there might be some case where you might actually need to send a flush() to a GPIB device, but in hindsight, I don't know of one.

MatthieuDartiailh, 2020-07-23:
Closed by #391

closed 2020-07-23 by MatthieuDartiailh

KM4YRI, 2020-07-24:
Nice, thanks!