pyvisa/pyvisa #392: AttributeError: module 'visa' has no attribute 'ResourceManager'

mcinquino, 2019-02-08:
Hello,

I have a strange problem that is giving me the following error: AttributeError: module 'visa' has no attribute 'ResourceManager'

What makes this strange is that when I run this program from IDLE I do not get this error. When I run the program from an application I have written in Lazarus using python for lazarus I get the above error.  Prior to getting this error the same code has been running flawlessly through the lazarus app. The lazarus app basically takes the code from a memobox and executes it as a script, it then returns the results to another memobox. 

Can anyone point me to what might be happening with this error. I am guessing it is not finding a library file or dll but they should all be in place as the code runs from IDLE. It may be useful to note that other python code runs fine. For example running code to pull the sys.path info runs fine. 

Below is the code I am trying to run.

Thanks,
Mike

import sys
import visa

if not hasattr(sys, 'argv'):
    sys.argv  = ['']

from ThorlabsPM100 import ThorlabsPM100
rm = visa.ResourceManager()
PM100D = rm.open_resource('USB0::0x1313::0x8078::P0010695::INSTR')
#power_meter = ThorlabsPM100(inst=inst)

print (PM100D.read)

print(PM100D.query('measure:power?'))
print(PM100D.query('power:dc:unit?'))
print(PM100D.query('sense:corr:wav?'))

MatthieuDartiailh, 2019-02-08:
Do you have a module named visa that may shadow pyvisa that is on the path when you run from lazarus ?  You can try to replace `import visa` by `import pyvisa as visa` to see if that fixes the issue. Otherwise could you post add `print(dir(visa))` to your script and post the output so that we can figure out what is in that module ?

mcinquino, 2019-02-11:
Hello,
import pyvisa as visa gave the same result. 

This is what was returned after running print(dir(visa)) (from Lazarus)
['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '__version__', 'absolute_import', 'compat', 'constants', 'division', 'log_to_screen', 'logger', 'logging', 'pkg_resources', 'print_functio
n', 'unicode_literals']

mcinquino, 2019-02-11:
I will add that attempting to execute import visa by itself returned the following. 

    from pyvisa.highlevel import ResourceManager
ImportError: cannot import name 'ResourceManager' from 'pyvisa.highlevel' (C:\Users\mcinquino\AppData\Local\Programs\Python\Python37\Lib\site-packages\pyvisa\highlevel.py)

MatthieuDartiailh, 2019-02-11:
This is really weird ... Somehow the content of the module seems altered. Could you post the output of `print(dir(pyvisa.highlevel))`  after importing pyvisa.highlevel?

mcinquino, 2019-02-11:
import pyvisa.highlevel
print(dir(pyvisa.highlevel))

returned 

Traceback (most recent call last):
  File "<string>", line 2, in <module>
AttributeError: module 'pyvisa' has no attribute 'highlevel'

import pyvisa.highlevel did not cause an error by itself.

MatthieuDartiailh, 2019-02-11:
Since the error happens only in Lazarus, is it possible that Lazarus built a cache that may have been corrupted at some point ?

mcinquino, 2019-02-11:
That is a good idea. I will have to look into this. I have uninstalled the python4Lasarus package and reinstalled it a few times in Lazarus to see if that would clear any hidden retained information with no luck.

MatthieuDartiailh, 2019-02-11:
Let me know if you find anything.

mike9755, 2019-04-07:
So, have you fixed this problem now? I met same as you, I got this message when I run a python file, but it's ok when run from python shell, but the file runs well in another computer, it's really weird

mcinquino, 2019-04-07:
Hi, no I was not able to resolve it.

changhexuefei, 2019-08-09:
Hello, I used "import visa visa.ResourceManager()" in May this year, it can work, but after re-installing pyvisa in the new project, use "import visa, visa.ResourceManager()" to report AttributeError: module ' Visa 'has no attribute 'ResourceManager' error, I changed to "import pyvisa as visa", it will work, why?

MatthieuDartiailh, 2019-08-09:
The visa.py file was last updated in November 2017. In your new project do you have a file named visa.py that could shadow the one provided by PyVISA ? You can inspect the `__file__` attribute of the module to determine what file Python picked.

MatthieuDartiailh, 2019-08-17:
I believe your issue is related to https://github.com/pyvisa/pyvisa/issues/447. The short fix is do not use the visa module but import from pyvisa.

changhexuefei, 2019-08-19:
As you might guess, there is a visa.py file in my project. I haven't noticed this problem, so after I modified this file, there is no 'AttributeError: module 'visa' has no Attribute 'ResourceManager'' error like this, thank you

| |
高景为
|
|
17701273631@163.com
|
签名由网易邮箱大师定制

closed 2019-08-19 by MatthieuDartiailh