pyvisa/pyvisa #393: GPIB instrument not detected by pyvisa

bserra, 2019-02-12:
I am using a raspberry with pyvisa installed on it. I installed linux-gpib and tried this configuration a few times already (with NI or agilent 82357a/b GPIB-USB adapters).

First time having this issue: the device is responding to ibtest but list_resources() from pyvisa does not list it.

pyvisa-info
```
Machine Details:
   Platform ID:    Linux-4.14.79-v7+-armv7l-with-debian-9.6
   Processor:

Python:
   Implementation: CPython
   Executable:     /home/irlabf/berryconda3/envs/pymont-env/bin/python
   Version:        3.6.6
   Compiler:       GCC 6.3.0 20170516
   Bits:           32bit
   Build:          Sep  6 2018 10:56:14 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR: Available via Linux GPIB (b'4.2.0')
```
Trying to list instruments using pyvisa
```
import visa
rm = visa.ResourceManager('@py')
rm.list_resources()
('ASRL/dev/ttyAMA0::INSTR',)
```
lsusb, my device (010) is on Bus 001
```
Bus 001 Device 007: ID 1d57:0008 Xenta
Bus 001 Device 004: ID 258a:6a88
Bus 001 Device 010: ID 0957:0107 Agilent Technologies, Inc.
Bus 001 Device 006: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Any idea of what I am doing wrong?

MatthieuDartiailh, 2019-02-13:
Usually those adapters rely on custom GPIB drivers that gpib-linux will definitively not pick-up. However those driver should be binary compatible with standard gpib drivers. You can try to use/hack https://github.com/tivek/gpib_ctypes to load this driver and see if you can communicate with your instrument. If so we should definitively expose ways to select the gpib drivers to use from pyvisa-py/gpib_ctypes. This also has some link to #337 but there we would love a solution broader than simply pyvisa-py.

bserra, 2019-02-13:
Actually, I have the drivers as well as some procedures for installing them on both agilent 82357a/b and the adapter seems to be working perfectly (green light on 'ready' instead of red light on 'fail'). I test them using ibtest and I can communicate with the instrument.
I was able to communicate with pyvisa and ibtest using this setup the week before, apparently someone switched instruments (same GPIB address) and now only ibtest is working.

I tried rebooting the instrument controller, reseting the usb port where the GPIB-USB adapter is plugged to, and rebooting the raspberry.

dirkenstein, 2019-07-11:
Just wanted to resurrect this thread and advise that `rm.list_resources() `may not work reliably on 82357a/b. You can still access resources by explicitly specifying their address in open_resource.

e.g.

`rm.open_resource('GPIB0::15::INSTR') 
`
The underlying `ibln()` (and `FindLsnr()`) functions in linux-gpib do not seem to work correctly with the 82357a. The problem is that they need a working` iblines()` function and that seems to not register changes in NDAC when the device is addressed.

bserra, 2019-09-19:
That's it @dirkenstein. I can access the instrument but have to guess the GPIB address if I do not have the documentation. So it is not critical but a bit surprising the first time. 
Replacing the GPIB to USB adapter with a NI one allowed me to see the ressource.

MatthieuDartiailh, 2019-09-25:
Since this is resolved (at least from pyvisa side) I am closing. Feel free to reopen if necessary.

closed 2019-09-25 by MatthieuDartiailh

qolor, 2021-06-14:
Hi floks, Im having a similar issue. When I plug the 82357b agilent gpib/usb interface, the raspy negociates well and the green light of ready is turned on. But the resource manager of pyvisa is not giving me nothing more that ASRL/dev/ttyAMA0::INSTR.
What am I doing wrong? the solution of giving the address to open resource didnt work.

MatthieuDartiailh, 2021-06-14:
Can you post the ouput of `pyvisa-info` ?

qolor, 2021-06-14:
pyvisa-info
Machine Details:
   Platform ID:    Linux-5.10.17-v7+-armv7l-with-debian-10.9
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Jan 22 2021 20:04:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.4 r[1984]')
      GPIB INTFC: Available via Linux GPIB (b'4.3.4 r[1984]')

Thank you for helping me.

MatthieuDartiailh, 2021-06-14:
Can you post the script you tried when directly providing the address and the error message you have been seeing ? I fail to see how your situation differs from the original report.