pyvisa/pyvisa #395: Pyvisa 1.9.1 ressource manager problem with NRPZ sensors

Mathieu1972, 2019-02-19:
Hi,
I have tested with two generations of power sensors and the old one that was working has problem to be detected and the new one seems to stay in OS registry even if not connected, here is the trace of the debug mode:
##############################################################################
2019-02-19 10:08:00,191 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2019-02-19 10:08:00,192 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2019-02-19 10:08:00,202 - pyvisa - DEBUG - No user defined library files
2019-02-19 10:08:00,210 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll']
2019-02-19 10:08:00,210 - pyvisa - DEBUG - The NI implementation available
2019-02-19 10:08:00,212 - pyvisa - DEBUG - No user defined library files
2019-02-19 10:08:00,219 - pyvisa - DEBUG - Automatically found library files: ['C:\\WINDOWS\\system32\\visa32.dll']
2019-02-19 10:08:00,230 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2019-02-19 10:08:00,231 - pyvisa - DEBUG - Created library wrapper for C:\WINDOWS\system32\visa32.dll
2019-02-19 10:08:00,235 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x10F98760>',) -> 0
2019-02-19 10:08:00,235 - pyvisa - DEBUG - Created ResourceManager with session 256
2019-02-19 10:08:00,243 - pyvisa - DEBUG - viFindRsrc(256, '?*::INSTR', '<ViObject object at 0x10F98760>', 'c_ulong(2)', <ctypes.c_char_Array_256 object at 0x10F98EE0>) -> 0
2019-02-19 10:08:00,243 - pyvisa - DEBUG - viFindNext(<ViObject object at 0x10F98760>, <ctypes.c_char_Array_256 object at 0x10F98EE0>) -> 0
2019-02-19 10:08:00,243 - pyvisa - DEBUG - viClose(<ViObject object at 0x10F98760>,) -> 0
La liste d instrument est :('USB0::2733::313::101542::0::INSTR', 'USB0::2733::351::100921::0::INSTR')
2019-02-19 10:08:00,244 - pyvisa - DEBUG - viParseRsrcEx(256, 'USB0::2733::313::101542::0::INSTR', 'c_ushort(7)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x10F98F80>, <ctypes.c_char_Array_256 object at 0x10F987B0>, <ctypes.c_char_Array_256 object at 0x10F98D50>) -> 0
2019-02-19 10:08:00,245 - pyvisa - DEBUG - USB0::2733::313::101542::0::INSTR - opening ...
2019-02-19 10:08:00,253 - pyvisa - DEBUG - viOpen(256, 'USB0::2733::313::101542::0::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x10F98D50>') -> -1073807264
Traceback (most recent call last):
........
    my_instrument =rm.open_resource(name)
  File "C:\Python37\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Python37\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Python37\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Python37\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "C:\Python37\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_NCIC (-1073807264): The interface associated with this session is not currently the controller in charge.
>>>
############################################################################

The problem is that none of the seen power sensors are plugged (NRP40S and NRP18SN) and the NRP Z21 power sensors are not seen at all, i had 2 of them plugged an not seens during this test.

ressource manager seems not work properly.

hgrecco, 2019-02-19:
Are the sensors visible to NI-MAX?

Mathieu1972, 2019-02-19:
No, NI MAX does not see the old power sensors NRP Z21 type, but Visa control in NI MAX sees the new generation like NRP40S (USB type)..I m using python 3.7.2 and NI VISA 18 (last version).

Mathieu1972, 2019-02-19:
It sees also a device that is another US sensor that is not plugged anymore from the same type as the NRP18SN from R&S.
[image]

hgrecco, 2019-02-19:
PyVISA is just a wrapper of NI-VISA (and other VISA libraries). My suggestion is that you make it work first in NI-VISA and then try it in PyVISA.

Mathieu1972, 2019-02-19:
The problem will be forwarded to NI and R&S to solve this issue, Pyvisa sees the same problem as NI VISA.

closed 2019-03-06 by MatthieuDartiailh