pyvisa/pyvisa #397: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed. (Keithley2260B)

riya1813, 2019-02-26:


riya1813, 2019-02-26:
my_inst = rm.open_resource('TCPIP0::192.168.18.73::2268::SOCKET', write_termination = '\n',read_termination='\n')
print(my_inst.query('*IDN?'))

MatthieuDartiailh, 2019-02-26:
Could you post the full error traceback and the output of the `pyvisa-info` command (run in a terminal).

MatthieuDartiailh, 2019-02-26:
If things work with NI I am surprised it does not work with pyvisa with the same settingd. Did you close the NI connection before trying pyvisa ? Next you can check whether or not the instrument understood your query by doing the following:
```python
my_inst.write('*IDN?')
my_inst.read_bytes(1)
```
If once again you get a time out, it means the instrument did not understand the query (most likely it did not detect the end of the message). Otherwise it means the end of the response is not detected. You can try to read one char till you get a timeout and see if there is an unexpected termchar.

riya1813, 2019-02-27:
I tried with commands as you suggested but it still throws the Timeout error.

Traceback (most recent call last):
  File "<pyshell#5>", line 1, in <module>
    c.read_bytes(1)
  File "C:\Users\2225\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\resources\messagebased.py", line 342, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\2225\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\2225\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_CONN_LOST (-1073807194): The connection for the given session has been lost.

riya1813, 2019-02-27:
I found d solution.
my_inst.read('\n') works and returns the result

MatthieuDartiailh, 2019-03-04:
This is weird since this simply set the termination character during the read. Can you check the value of _read_termination after creating the resource ?

gitanjali1990, 2019-05-24:
I tried my_inst.read('\n') it doesnot work. I am  using Keithely 2614B.
Thanks a lot

MatthieuDartiailh, 2019-05-24:
@gitanjali1990 please post the output of `pyvisa-info` and you script, otherwise it is hard to help. I suggest you also have a look at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#getting-the-instrument-configuration-right that details how to investigate how to configure PyVISA to communicate with your instrument.

rcnijzink, 2019-05-28:
The code @gitanjali1990 refers to is like this:

`import visa
import pyvisa
import time

rm = visa.ResourceManager()
rm.list_resources()
#('ASRL1::INSTR', 'ASRL2::INSTR', 'GPIB0::12::INSTR')
inst1 = rm.open_resource('TCPIP::k-2614B-4407318.private.list.lu::23::SOCKET',write_termination= '\n', read_termination='\n')
#inst1.timeout = None
print(inst1.write("smua.reset()"))
time.sleep (2)
print(inst1.write("smua.nvbuffer1.clear()"))
print(inst1.write("smua.nvbuffer1.collecttimestamps = 1"))
print(inst1.write("smua.nvbuffer1.collectsourcevalues = 1"))
print(inst1.write("smua.source.limiti = 0.01"))
time.sleep (1)
print(inst1.write("smua.source.limitv = 20"))
time.sleep (1)
print(inst1.write("smua.sense = smua.SENSE_REMOTE"))
time.sleep (1)
print(inst1.write("SweepILinMeasureV(smua, -5e-4, 5e-4, 0.1, 10)"))
time.sleep (5)
print(inst1.write("printbuffer(1,10,smua.nvbuffer1.timestamps)"))
print(inst1.write("printbuffer(1,10,smua.nvbuffer1.sourcevalues)"))
print(inst1.write("printbuffer(1,10,smua.nvbuffer1.readings)"))

time.sleep (2)
while True:
   time.sleep (1)
    print(inst1.read_bytes(1))

time.sleep (2)
print(data)`

The error:
`Traceback (most recent call last):
  File "read_visa3.py", line 32, in <module>
    print(inst1.read_bytes(1))
  File "/home/remko/miniconda2/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 342, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "/home/remko/miniconda2/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 350, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
`
This happens after generally reading a couple of bytes, on Ubuntu 18.04.2 LTS and Windows 7.

MatthieuDartiailh, 2019-05-28:
What do you expect to happen here but a timeout ? The while loop is infinite and I do not believe your  instrument is sending data forever. Could you describe what you are attempting to do ?

rcnijzink, 2019-05-28:
The while-loop was added to test if it would read any bytes at all, as you suggested by this link:

https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#getting-the-instrument-configuration-right

Before, we tried with what @gitanjali1990 mentioned: 
```
inst1.read('\n')
```

Or just a plain
```
inst1.read()
```

MatthieuDartiailh, 2019-05-28:
What bytes do you read ? and what is the format of the expected answer ?

rcnijzink, 2019-05-29:
It should be ascii, and it should be a termination of \r\n (which we tried, just as other terminations). The timeout seems to occur when reading a specified number of bytes, that is bigger than the real number of bytes. But when we add inst1.timeout = None, it just hangs and nothing happens. We tried to get the bytes_in_buffer to work, but this is apparently not implemented for a tcpip device.

MatthieuDartiailh, 2019-05-29:
Could you post the ourput of your script where you print one byte at a time ? I will try to have a look at the instrument manual.

rcnijzink, 2019-05-29:
Thank you for the help! But we solved it now. The solution was quite simple in the end:

```
data = inst.read_bytes(1000, break_on_termchar='\r\n')
```

jbehura, 2019-09-04:
I also get a timeout error on query. As you have mentioned earlier, it has something to do with the termination characters. However, I can't figure out why I get an error when using 
```
scope.query('channel1:data:header?')
```

Here is my code:
```
import visa
import numpy
import matplotlib.pyplot as plt

visa.ResourceManager('@py')

rm = visa.ResourceManager()

scope = rm.open_resource("TCPIP::192.168.1.1::INSTR")
scope.read_termination = '\n'
scope.write_termination = '\n'
print(scope.query('*IDN?'))

scope.write('*IDN?')
data = scope.read_bytes(69)
print(data)

scope.query('channel1:data:header?')
scope.write("format real,32")
values = scope.query_binary_values("channel1:data:values?", datatype='f')
plt.plot(values)
plt.show()
```
Here is the output along with the error:
```
TEKTRONIX,TDS 3014B,0,CF:91.1CT FV:v3.41 TDS3FFT:v1.00 TDS3TRG:v1.00
b'TEKTRONIX,TDS 3014B,0,CF:91.1CT FV:v3.41 TDS3FFT:v1.00 TDS3TRG:v1.00\n'
Traceback (most recent call last):
  File "/home/jbehura/Work/Projects/Wild/Code/acquireData.py", line 18, in <module>
    scope.query('channel1:data:header?')
  File "/usr/local/lib/python3.6/dist-packages/pyvisa/resources/messagebased.py", line 600, in query
    return self.read()
  File "/usr/local/lib/python3.6/dist-packages/pyvisa/resources/messagebased.py", line 414, in read
    message = self._read_raw().decode(enco)
  File "/usr/local/lib/python3.6/dist-packages/pyvisa/resources/messagebased.py", line 387, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python3.6/dist-packages/pyvisa-py/highlevel.py", line 350, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Process finished with exit code 1
```

MatthieuDartiailh, 2019-09-04:
Based on your instrument model manual (as reported in the IDN) (the manual can be found there https://www.tek.com/oscilloscope/tds3014b-manual/tds3000-tds3000b-tds3000c-series), your instrument does not support any command starting by 'channel'. I suggest you have a look at the manual to figure out how to achieve your goal.

closed 2019-09-04 by MatthieuDartiailh

EngJob, 2021-05-17:
Hello, 
I need  help to communicate with with Keithley 22321-30-3 Power Supply. Commands shown here https://download.tek.com/manual/2200S-903-01_A_Oct_2012_2200S-903-01A.pdf 
I have the resources name and used this code

```
import pyvisa
import time
rm = pyvisa.ResourceManager()

power= rm.open_resource('ASRL5::INSTR')

power.query("INST:SEL	CH2")
power.query("SOURCE:OUTP:ENAB	ON")
power.query("SOURCE:VOLT	2.22V")
power.query("SOURCE:CURR	1.5A")
power.query("SOURCE:OUTP	ON")
```
Getting Error:
C:\Users\EngJob\Documents>power2.py
Traceback (most recent call last):
  File "C:\Users\EngJob\Documents\power2.py", line 7, in <module>
    power.query("INST:SEL       CH2")
  File "C:\python\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\python\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\python\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\python\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\python\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\python\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
**pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.**
please help.

MatthieuDartiailh, 2021-05-18:
`query` should only be used with commands for which you expect a response from the instrument. Typically (I did not check for your specific instrument but I have worked with Keithley instruments before), none of the command you are using is expected to generate a response from the instrument, which is why you get a timeout when reading.

EngJob, 2021-05-18:
The commands I am sending **to set the voltage and curren**t. **it is NOT setting the values.** 
why should we expect a respond?

MatthieuDartiailh, 2021-05-18:
Your script basically stops at `power.query("INST:SEL	CH2")`  because you attempt to read a response that does not exist, replace your `query` with `write`. If you want to check for proper termination character try querying "*IDN?".

EngJob, 2021-05-18:
I run the following
```
import pyvisa
import time
rm = pyvisa.ResourceManager()

power= rm.open_resource('ASRL5::INSTR')
power.query('*IDN?')
```

and getting the same error
**pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.**

EngJob, 2021-05-18:
Thank you @MatthieuDartiailh 
this is so frustrating to see keithley has so many manual refences for the same power supply
Here are what I got so far
1. 2200 series https://download.tek.com/manual/2200S-903-01_A_Oct_2012_2200S-903-01A.pdf 
2. and this which say for 2200 seris https://download.tek.com/manual/2220S-907-01B_Dec_2013_Prog.pdf 
3. and the one from you https://download.tek.com/manual/077100401_Referencemanual.pdf 

I am using Serial to USB and and driver is working on device manger and I get the using **rm = pyvisa.ResourceManager()**
**print(rm.list_resources())** I get the **ASRL5::INSTR** which is correct.  I disconnect and connect to make sure it is the right device. 
Now so many people mention about the new line  \n character but now one says how to use. Everyone should give simple example. How would I use it?  like this
 `power.query('*IDN?\n')` 
or after running the command, like this
 `power.query('\n')`

The  SYST:REM seems the most important. I will try it going back to the lab and get back to you.

MatthieuDartiailh, 2021-05-18:
Regarding the termination character setting you can do the following:
```python
import pyvisa
import time
rm = pyvisa.ResourceManager()

power= rm.open_resource('ASRL5::INSTR')
power.write_termination = "\n"
power.read_termination = "\n"
```

I will try to have a look at the other resources you shared too.

EngJob, 2021-05-18:
Hello @MatthieuDartiailh 
I tried this
```
import pyvisa
import time
rm = pyvisa.ResourceManager()

power= rm.open_resource('ASRL5::INSTR')
power.write_termination = "\n"
power.read_termination = "\n"
power.write("syst:rem")
power.write("Chan:outp 1")
power.write("Volt 19")
```

Error is gone, but the **voltage is not set** and **channe 1** is not selected as output.
and this 
```
power.write_termination = "\n"
power.read_termination = "\n"
power.write('*IDN?')
```
or  `power.query('*IDN?')` both gives the same timeout error. 

The manual https://download.tek.com/manual/077100401_Referencemanual.pdf is confusing I still did not get how to use it. may be I don't understand how to set voltage.

MatthieuDartiailh, 2021-05-18:
The main issue I see is that we do not know what serial parameters to use (baud rate, parity, stop bit, protocol). If the voltage is not set it probably mens the instrument does not register the command. I will try to provide a more detailed plan tomorrow, I just wish the manual was more explicit...

MatthieuDartiailh, 2021-05-18:
Actually this repo while old contains useful bits https://github.com/tektronix/Keithley_2231A_Examples/blob/master/PyVISA_Examples/Keithley_2231A_PyVISA_Examples.py

Note however that stop_bits should be set using the appropriate enumeration found in pyvisa.constants.

Hope this helps.

EngJob, 2021-05-19:
@MatthieuDartiailh Thanks for your help. I appreciate it.
Had already tested the code in the repository. But did not work. Gives the same error :  **VI_ERROR_TMO (-1073807339)**
```
import pyvisa
import time
rm = 0
my_PS = 0

def KEI2231_Connect(rsrcString, getIdStr, timeout, doRst):
    my_PS = rm.open_resource(rsrcString, baud_rate = 9600, data_bits = 8)	#opens desired resource and sets it variable my_instrument
    my_PS.write_termination = '\n'
    my_PS.read_termination = '\n'
    my_PS.send_end = True
    my_PS.StopBits = 1
    # my_PS.flow_control =      # only available in PyVisa 1.9
    #my_PS.baud_rate = 9600
    if getIdStr == 1:
        print(my_PS.query("*IDN?"))
        #time.sleep(0.1)
    my_PS.write('SYST:REM')
    #print(my_PS.timeout)
    my_PS.timeout = timeout
    #print(my_PS.timeout)
    if doRst == 1:
        my_PS.write('*RST')
        #time.sleep(0.1)
    return my_PS

def KEI2231A_Disconnect():
    my_PS.write('SYST:LOC')
    my_PS.close
    return

def KEI2231A_SelectChannel(myChan):
    my_PS.write("INST:NSEL %d" % myChan)
    #time.sleep(0.25)
    return

def KEI2231A_SetVoltage(myV):
    my_PS.write("VOLT %f" % myV)
    #time.sleep(0.24)
    return

def KEI2231A_SetCurrent(myI):
    my_PS.write("CURR %f" % myI)
    #time.sleep(0.24)
    return

def KEI2231A_OutputState(myState):
    if myState == 0:
        my_PS.write("OUTP 0")
        #time.sleep(0.25)
        #my_PS.write("OUT:ENAB 0")
    else:
        my_PS.write("OUTP 1")
        #time.sleep(0.25)
        #my_PS.write("OUT:ENAB 1")
    #time.sleep(0.25)
    return

def KEI2231_Send(sndBuffer):
    my_PS.write(sndBuffer)
    return

def KEI2231_Query(sndBuffer):
    return my_PS.query(sndBuffer)
#================================================================================
#    MAIN CODE GOES HERE
#================================================================================
t1 = time.time()    # Capture start time....
rm = pyvisa.ResourceManager()	# Opens the resource manager and sets it to variable rm

my_PS = KEI2231_Connect("ASRL5::INSTR", 1, 20000, 1)

KEI2231A_SelectChannel(1)
KEI2231A_SetVoltage(12.0)
KEI2231A_SetCurrent(1.0)
KEI2231A_OutputState(1)

time.sleep(0.25)

KEI2231A_OutputState(0)

KEI2231A_Disconnect()

rm.close

t2 = time.time() # Capture stop time...
print("{0:.3f} s".format((t2-t1)))
```
Modified the code as follow. but not error, and does not set the voltage
```
import pyvisa
import time
t1 = time.time()    # Capture start time....
rm = pyvisa.ResourceManager()

my_PS = rm.open_resource('ASRL5::INSTR', baud_rate = 9600, data_bits = 8)	
my_PS.write_termination = '\n'
my_PS.read_termination = '\n'
my_PS.send_end = True
my_PS.StopBits = 1
# my_PS.flow_control =  # only available in PyVisa 1.9
my_PS.baud_rate = 9600

#print(my_PS.query("*IDN?")) # this gives error
#time.sleep(0.1)
my_PS.write('SYST:REM')
my_PS.write("INST:NSEL 1")
my_PS.write("VOLT 12.0")
my_PS.write("OUTP 1")
#time.sleep(0.25)
#my_PS.write("OUT:ENAB 1")
#time.sleep(0.25)

my_PS.write('SYST:LOC')
my_PS.close
```
and changed the `my_PS.write("OUTP 1")` to `my_PS.write("OUT:ENAB 1")` but still did not work. 
it seems the person who has created this code had doubt to wither use **_OUTPUT 1_** or _**OUTPUT:ENAB 1**_

MatthieuDartiailh, 2021-05-20:
Not sure it will help but for `stop_bits ` the appropriate syntax would be:
```python
from pyvisa.constants import StopBits
my_PS.stop_bits = StopBits.one
```
You can also try the different possible value of the control_flow and see if anything works
```python
from pyvisa.constants import ControlFlow
my_PS.control_flow= ControlFlow.none # ControlFlow.xon_xoff, ControlFlow.rts_cts, ControlFlow.dtr_dsr
```

EngJob, 2021-05-20:
Hello, 
I get this error `NameError: name 'StopBits' is not defined`
I am using RS232 to USB. would that be the issue. I am not getting anything back not anything can be set.

MatthieuDartiailh, 2021-05-20:
What version of pyvisa are you using ? Try updating to 1.11.

wy7318, 2021-05-26:
Dear @MatthieuDartiailh,

I have been reading all the comments regarding timeout and nothing could resolve my issue. I am using HP 8153A ([http://www.doe.carleton.ca/~nagui/labequip/lightwave/8153A_Programming%20Manual.pdf]) and I could read '*IDN' just fine. 
However, when I try to read the output power, it keeps giving me an timeout error. So far, I was able to read power **once** but it suddenly start giving me errors since then. 

`import pyvisa
rm = pyvisa.ResourceManager()
inst = rm.open_resource("GPIB0::5::INSTR")
# This part works
inst.write("*IDN?")
name = inst.read()
print(name)
# This part doesn't work. Was able to read just one time so I know the command is correct
inst.write("FETch2:SCALar:POWER:DC?")
readPower = inst.read()
print(readPower)`

Hoping to get your help, Thank you in advance.

MatthieuDartiailh, 2021-05-27:
Looking at the manual it appears you misspelled the command. It should be Fetch not Fetch2. I only gave this a quick so there may be other typos I missed.

wy7318, 2021-05-27:
Dear @MatthieuDartiailh,

Thank you for your time and reading the manual for this. However, 'fetch2' means the channel number, 1 or 2 could be placed for this command. I have tried 'fetch' but still having timeout error.  Would there be any other solution I could try? I really appreciate your time!

Best Regards,

MatthieuDartiailh, 2021-05-27:
Looking at the issue on a computer rather than my phone I believe the issue comes from the fact that fetch simply query the last measurement performed but if the instrument was not trigged it cannot answer. I suggest either enabling continuous triggering, or using the read command which trigger the instrument.

The relevant portions of the manual are found page 157 for fetch and page 161 for read.

wy7318, 2021-05-27:
@MatthieuDartiailh,
Thank god, I am embarrassed missing such an info but you are my hero on this application! 
I really appreciate it.

MatthieuDartiailh, 2021-05-27:
Happy to help and sorry for the noise earlier

EngJob, 2021-05-28:
> Looking at the manual it appears you misspelled the command. It should be Fetch not Fetch2. I only gave this a quick so there may be other typos I missed.

I copied it from the manual. it refers to the ch1 or ch2

LydieNsangou, 2021-07-21:
hi please i know it is not the same issue but i have a question 
i have been trying to remotely control my tektronix oscilloscope via usb . here is my code

import pyvisa
import time
from time import sleep
rm = pyvisa.ResourceManager()
print(rm)
rm.list_resources()
oszi = rm.open_resource('USB0::0x0699::0x03A4::C021051::INSTR')
 Value= 0.003E-4
print(oszi.write('HORIZONTAL:SCALE  Value  ')

i am trying to find out if it is possible to add some variables in the commands .forexample i had to directly write the value of the horizontal scale in the command above but i will need to vary the horizontal scale in my test case . has someone try this before
thanks for your help :)

MatthieuDartiailh, 2021-07-21:
The following should do what you want. You simply need to format the string:
```python
import pyvisa
import time
from time import sleep
rm = pyvisa.ResourceManager()
print(rm)
rm.list_resources()
oszi = rm.open_resource('USB0::0x0699::0x03A4::C021051::INSTR')
Value= 0.003E-4
oszi.write(f'HORIZONTAL:SCALE {Value}')
```

LydieNsangou, 2021-07-21:
Hi @MatthieuDartiailh !thanks for your Quick reponse 
Does it work by you because I tried that already and it didn't work.

MatthieuDartiailh, 2021-07-21:
I have definitely often formatted values into strings passed to pyvisa. However I cannot test with your instrument. It may be an issue of how the float is formatted. I assume `oszi.write(f'HORIZONTAL:SCALE 0.003E-4')`. Does using `oszi.write(f'HORIZONTAL:SCALE 3.000E-7')` directly works ? If yes, then try oszi.write(f'HORIZONTAL:SCALE {Value:.3E}') which will resolve to the exact same string.

LydieNsangou, 2021-07-21:
Yes when using oszi.write(f'HORIZONTAL:SCALE 3.000E-7') it works .but I will like to vary 3.000E-7  that why I want to create à variable which I can insert there

MatthieuDartiailh, 2021-07-21:
Then try `oszi.write(f'HORIZONTAL:SCALE {Value:.3E}')` which will enforce the proper formatting. If it fails please post a traceback or a detailed description of the error.

LydieNsangou, 2021-07-22:
Hallo @MatthieuDartiailh good morning!
i just tried with oszi.write(f'HORIZONTAL:SCALE {Value}') and it worked thanks

MatthieuDartiailh, 2021-07-22:
Happy to help

LydieNsangou, 2021-07-22:
@MatthieuDartiailh do you know the command to create anew folder ? i have been writing this oszi.write("FILESYSTEM:MAKEDIR E:/Test") but it seems not to work

MatthieuDartiailh, 2021-07-22:
Check the manual but usually strings need to be quoted and I assume you need a \ since the system is using windows. Try `oszi.write(r'FILESYSTEM:MAKEDIR "E:\Test"')` but note it is just a guess.

Shaang98, 2021-12-08:
Hi @MatthieuDartiailh , I'm currently trying to remotely control my OSA AQ6370D via LAN cable. The code runs fine but when I input "Save" as a command, it shows a timeour error 
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Heres the traceback:
Traceback (most recent call last):
  File "C:\Users\Shaan G\Desktop\Bangkit Academy\BA 3\BA 4\BA 5\BA 6\BA 7\BA 8\BA 9\YOKO AQ6370 LAN (1).py", line 76, in <module>
    wl, intensity = get_trace(trace)
  File "C:\Users\Shaan G\Desktop\Bangkit Academy\BA 3\BA 4\BA 5\BA 6\BA 7\BA 8\BA 9\YOKO AQ6370 LAN (1).py", line 21, in get_trace
    wl = YOKO.query('WDAT' + trace).strip().split(',')[1:] # basically commands from device library to get data from device like wavelenght and intensity
  File "C:\Users\Shaan G\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 638, in query
    self.write(message)
  File "C:\Users\Shaan G\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Users\Shaan G\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]
  File "C:\Users\Shaan G\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2796, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))
  File "C:\Users\Shaan G\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\Shaan G\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
  
  Really hope to hear your input, thanks!

LongnoseRob, 2021-12-08:
Very probably the command string is not recognized.
I would recommend to confirm if the specific command is supported by your OSA.
Yokogawa Document IMAQ6370C-17EN might be a good starting point..

SengerM, 2021-12-22:
Don't know if this is the best place to ask for help, but this already looks like a long question-and-answer site so I will proceed. I am trying to communicate with an [EA-PS 5040-20 A](https://elektroautomatik.com/shop/en/products/programmable-dc-laboratory-power-supplies/dc-laboratory-power-supplies/series-ps-5000-br-160-up-to-640-w/705/laboratory-power-supply-0..40v/0..20a/320w) power supply from Ubuntu. This is my code
```Python
import pyvisa

power_supply = pyvisa.ResourceManager().open_resource('ASRL/dev/ttyACM3::INSTR') # Found the address using https://unix.stackexchange.com/a/144735/317682

termination = '\n' # From the SCPI manual that was shipped with the power supply and which I cannot locate on the web.
power_supply.write_termination = termination
power_supply.read_termination = termination

print(f"Number of bytes written: {power_supply.write('*IDN?')}")
while True: # Whatever I read with this `while` loop is independent of what I write, but I have to write something in order to read something.
	try:
		print(power_supply.read_bytes(1))
	except Exception as e:
		print(e)
		break

print(power_supply.query('*IDN?')) # Always timeout error.
```
and this is my output
```Bash
Number of bytes written: 15
b'*'
b'\xc9'
b'\x02'
b'\x87'
b'\x99'
VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Traceback (most recent call last):
  File "deleteme.py", line 17, in <module>
    print(power_supply.query('*IDN?'))
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 644, in query
    return self.read()
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/home/tct/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```
I have a PDF document that came in a USB stick together with the power supply describing the SCPI interface. I could not find it in their website. In this manual it says:

> In general, SCPI does not require a termination character when using it via USB, Ethernet or other interfaces, ex-
cept for GPIB. The IEEE standard defines the use of a termination character. Hence devices with installed option
3W, i. e. GPIB interface, require to send this character or else a but timeout error will occur.
> Supported termination character(s): 0xA (LF, line feed)

I am using USB so it seems it is not needed, but anyway I set it to `'\n'`. I have already tried with other popular options like `'\r'` and `chr(0)`, `'\n\r'` and `'\r\n`.

The power supply is not ["lock to local control"](https://elektroautomatik.com/shop/media/pdf/9d/6a/61/05100300_EN.pdf#page=30) so it should be possible to control it via USB.

Any help with this is appreciated.

LongnoseRob, 2021-12-23:
@SengerM 
Ideally this would be a new issue, but I also see that it might be connected to the issue discussed here before.

In this case two thing sfeel strange to me:

1.
> Number of bytes written: 15
This should be less, it feels strange that `*IDN?` is written in 15bytes, I would expect 6/7 bytes (with termination).
Which is what I see if do the same with a device (R&S signal generator) connected via GPIB:
```
In [16]: rm = pyvisa.ResourceManager()

In [17]: sfm = rm.open_resource("GPIB0::20::INSTR")

In [18]: print(f"Number of bytes written: {sfm.write('*IDN?')}")
Number of bytes written: 7
```

2.
> ```
> b'*'
> b'\xc9'
> b'\x02'
> b'\x87'
> b'\x99'
> ```
These are not standard ASCII encodings.

Have you tried to talk to the device with a terminal program e.g. minicom (as yoiu mention you are on ubuntu)?

@MatthieuDartiailh could it have to do with wrong en/decodings to unicode somewhere on the line pyvisa --> pyvisa-py --> pyserial?

SengerM, 2021-12-23:
@LongnoseRob sorry, the number of bytes written is 5. This was the output from a test I did `print(f"Number of bytes written: {power_supply.write('*IDN?ansodndjas')}")` (or similar) to check if the `while` loop reads something different. It does not. I will fix 15→5 in [my original post](https://github.com/pyvisa/pyvisa/issues/397#issuecomment-999841942).

LongnoseRob, 2021-12-23:
@SengerM  ok, 5 bytes sent makes much more sense, so the transmit side seems to be ok

SengerM, 2021-12-23:
@LongnoseRob I never used minicom, I am giving a look at it right now. The idea is to try randomly different communication setups to see if I can find one that works? With the "default setup" anything I type shows noise, for example if I type `hello` this is what shows up:
```
Welcome to minicom 2.7.1

OPTIONS: I18n                                                                
Compiled on Dec 23 2019, 02:06:26.                                           
Port /dev/ttyACM3, 09:31:52                                                  
                                                                             
Press CTRL-A Z for help on special keys                                      
                                                                             
h�'ɍe�ɶNl�fLl�fLo�ɖL 
```
Is this any hint for you?

LongnoseRob, 2021-12-23:
@SengerM 
I would recommend to use different serial speed e,g, 9600
Also you should check if you are on the right interface, ctrl-a o will open the settings dialog.

SengerM, 2021-12-23:
Have tested the three available serial speeds (9600, 38400 and 115200) but it does not seem to change anything. When you say "right interface" you mean `/dev/ttyACM3` or something else?

MatthieuDartiailh, 2021-12-23:
Going through your instrument manual it appears it does not rely on VISA for communication but on the MODBUS protocol which is a binary protocol. As a consequence PyVISA cannot help. If you can link the programming manual for the manufacturer, I will double check my conclusions.

SengerM, 2021-12-24:
Thanks. [Here](https://cernbox.cern.ch/index.php/s/RCSSFkAQZileD65) is the programming manual shipped in a USB stick with the power supply.

MatthieuDartiailh, 2022-04-05:
Can you create a new issue for this and post the output of `pyvisa-info` ?

[8 more replies]