pyvisa/pyvisa #398: visa.ResourceManager('@ni') returns either visa64.dll or visa32.dll, choosing randomly

HansJA, 2019-03-05:
visa.ResourceManager('@ni') returns either visa64.dll or visa32.dll, choosing randomly.

one attempt:
C:\Users\Hans\Desktop>python
Python 3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 23:09:28) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> visa.ResourceManager('@ni')
<ResourceManager(<VisaLibrary('C:\\WINDOWS\\system32\\visa64.dll')>)>
>>> ^Z

next attempt
C:\Users\Hans\Desktop>python
Python 3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 23:09:28) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> visa.ResourceManager('@ni')
<ResourceManager(<VisaLibrary('C:\\WINDOWS\\system32\\visa32.dll')>)>
>>> ^Z

In 10 attempts visa32 was chosen 6 times, visa64 4 times.

visa32 gets chosen more often, my feeling is that in the longer perspective visa32.dll is chosen 7 or 8 times out of 10.

MatthieuDartiailh, 2019-03-05:
Do you see any resulting issue ? According to the following thread it is possible for both to actually be 64 bits and should not cause any trouble in your application? https://forums.ni.com/t5/Instrument-Control-GPIB-Serial/NI-MAX-requires-32-bit-VISA-missing-in-64-bit-installs/m-p/3896195

You can confirm the bitness by running multiple times `pyvisa-info` from the command line.

HansJA, 2019-03-05:
Is bitness your only criterion for selecting a resource manager?
I think you should assume that NI chose different names for good reason.

Other than that, I have multiple issues with PyVISA, but I cannot safely
attribute any one problem with this bug report.
Most if my issues have one specific form:
I do
pip install -U <all pyvisa related packages>
python
import visa
<Get an instrument and try to use it>
^Z
pip uninstall <all pyvisa related packages>

pip install - U <all pyvisa related packages>
python
import visa
<Get an instrument and try to use it>

And the instrument will behave very differently each time.

python-vxi11 works consistently...

Because of my wobbly experience with pyvisa I probably will forget all
about it for the time being.

Hans J. Albertsson
From my Nokia 8

MatthieuDartiailh, 2019-03-05:
Could you describe your problem in a more detailed manner ? I cannot really help with such a description.

FYI visa32 and visa64 found in system 32 are equivalent per https://forums.ivifoundation.org/questions/904/64-bit-c-visa-clarrification.html 
Hence bitness matching is sufficient.

HansJA, 2019-03-05:
I didn't actually intend that whining complaint of mine to be a bug report:
sorry. :-) I was just getting a bit fed up with nothing ever working for a
whole day.

I shall get on with my project if building a new lab, and return to the
automation problem later on!

I can give you one handwave on a minor surprise:

I have a Siglent SPD3303X power supply, and I did

import visa
rm = visa.ResourceManager('ni')
rm.list_resources_info()

That returned the name
'USB0::0x0483::0x7450::SPD3X2R4938::INSTR0'
and the alias 'PS1'.

myi = rm.open_resource('USB0::0x0483::0x7450::SPD3X2R4938::INSTR0'

then returns error lines, finally saying "location information missing" (or
something like that).

Opening the alias 'PS1' instead worked ok.

Opening the TCPIP0 resource worked with the name.

Length issue?

I could never talk to the 3303X over either resource anyway, it just beeps
at all writes and reads, and query always hangs until global timeout

Vxi11 works with the 3303X, but I have other instruments that won't talk
over vxi, so no joy for my automation project.

Hans J. Albertsson
From my Nokia 8

HansJA, 2019-03-05:
'ni' was really '@ni' in that rm.ResourceManager() call, I just misspelt
the word.

Hans J. Albertsson
From my Nokia 8

MatthieuDartiailh, 2019-03-05:
For communication issues, my usual advice is to check for termination character settings. I am in the process of updating the docs in that direction actually. Whenever you come back to working on your automation, do not hesitate to ping me for help.
I will leave this open as a reminder to mention the equivalence of visas in the docs.

HansJA, 2019-03-05:
Any comment on "opening the long name failing"?

Hans J. Albertsson
From my Nokia 8

HansJA, 2019-03-05:
And thank you for your interest in my problem!

Hans J. Albertsson
From my Nokia 8

Den tis 5 mars 2019 14:34Hans J Albertsson <hans.j.albertsson@gmail.com>
skrev:

MatthieuDartiailh, 2019-03-05:
Do you mean the issue with opening the usb resource ? I don't have an answer off the top of my head but I will try to look into it (the third part of the address should be the serial number and it does look long in your case). Since you are using NI can you quickly confirm it works with NI-Max ?

HansJA, 2019-03-05:
It does!

Hans J. Albertsson
From my Nokia 8

MatthieuDartiailh, 2019-03-05:
Thanks. Could you post the full traceback of the error ? I t may help me figure out what the issue is about.

HansJA, 2019-03-05:
Ok, in an hour or so

Hans J. Albertsson
From my Nokia 8

HansJA, 2019-03-05:
Tomorrow. Sorry.

Hans J. Albertsson
From my Nokia 8

Den tis 5 mars 2019 15:10Hans J Albertsson <hans.j.albertsson@gmail.com>
skrev:

MatthieuDartiailh, 2019-03-05:
Sure no problem

HansJA, 2019-03-05:
Here.
First I startup, import and define the rm

C:\Users\Hans\Desktop>python
Python 3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 23:09:28) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm = visa.ResourceManager('@ni')
>>> rm
<ResourceManager(<VisaLibrary('C:\\WINDOWS\\system32\\visa32.dll')>)>

Next, I look at the complete resource info.

>>> rm.list_resources_info()
{'USB0::0x0483::0x7540::SPD3XHBD2R4938::INSTR': ResourceInfo(interface_type=<InterfaceType.usb: 7>, interface_board_number=0, resource_class='INSTR', resource_name='USB0::0x0483::0x7540::SPD3XHBD2R4938::0::INSTR', alias='PS1'), 'TCPIP0::192.168.1.161::inst0::INSTR': ResourceInfo(interface_type=<InterfaceType.tcpip: 6>, interface_board_number=0, resource_class='INSTR', resource_name='TCPIP0::192.168.1.161::inst0::INSTR', alias=None), 'ASRL3::INSTR': ResourceInfo(interface_type=<InterfaceType.asrl: 4>, interface_board_number=3, resource_class='INSTR', resource_name='ASRL3::INSTR', alias='COM3'), 'ASRL10::INSTR': ResourceInfo(interface_type=<InterfaceType.asrl: 4>, interface_board_number=10, resource_class='INSTR', resource_name='ASRL10::INSTR', alias='LPT1')}

I try opening the first string in the info for the first resource on the USB0 bus
>>> ps1 = rm.open_resource('USB0::0x0483::0x7450::SPD3XHBD2R4938::INSTR')

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

That didn’t work, but wait: In the resourceinfo there’s a different string in the resource_name()! An extra 0::
How about opening THAT??
>>> ps1 = rm.open_resource('USB0::0x0483::0x7450::SPD3XHBD2R4938::0::INSTR')

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1213, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
  File "C:\Program Files\Python37\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

No, that worked no better.

>>> ps1 = rm.open_resource('PS1')

The alias from the resource_info could however be opened!
Yay!

>>> ^Z
Skickades från E-post för Windows 10

Från: Hans J Albertsson
Skickat: den 5 mars 2019 15:51
Till: pyvisa/pyvisa
Kopia: pyvisa/pyvisa; Author
Ämne: Re: [pyvisa/pyvisa] visa.ResourceManager('@ni') returns eithervisa64.dll or visa32.dll, choosing randomly (#398)

Tomorrow. Sorry.
Hans J. Albertsson
From my Nokia 8

Den tis 5 mars 2019 15:10Hans J Albertsson <hans.j.albertsson@gmail.com> skrev:
Ok, in an hour or so
Hans J. Albertsson
From my Nokia 8

hgrecco, 2019-03-06:
While I do agree with @MatthieuDartiailh that visa32 or visa64 should be fine, I do think that having a deterministic discovery is something good to aim at. The offending part, I think is deduplicating using a `set` [here](https://github.com/pyvisa/pyvisa/blob/master/pyvisa/ctwrapper/highlevel.py#L78). I am submiting a PR for this.

Regarding choosing bitness, I would like to add some historical information that might be useful. 

In PyVISA 1.4 the path was required when creating a VisaLibrary object with hardcoded fallbacks for windows and linux. No fallback for OSX. See [here](https://github.com/pyvisa/pyvisa/blob/1.4/pyvisa/vpp43.py#L111). A large number of questions were related to problems finding the library across different platforms. This was 2012 and while 64 bits computers were already decades olds, home computers, operating systems, libraries and applications (including python) where still in the middle of the migration. A correct installation required to look at 3 different things: Operating System, Python Interpreter and library; and most users were not able to get this information consistently.

So one of the main aims of PyVISA 1.5 was to make this automatic and easier to debug. That is how the automatic detection was incorporated [here](https://github.com/pyvisa/pyvisa/blob/1.5/pyvisa/util.py#L108) and a function to collect hardware and software diagnostics information was developed [here](https://github.com/pyvisa/pyvisa/blob/1.5/pyvisa/util.py#L311) (this last function is the base of the current `info` subcommand). This released included several functions to find the bitness of libraries in different operating systems. 

In PyVISA 1.6.1 we included in the diagnostics output the results from several calls to the library. In other words, we open the library and get the vendor, impl. and spec. version by calling a library function [here](https://github.com/pyvisa/pyvisa/blob/1.6.2/pyvisa/ctwrapper/highlevel.py#L106). We found this to be a better way of assuring compatibility that bitness checking as it required to actually use the library.  Bitness checking was difficult to make it work accross all supported platforms, compilers, interpreters and OS.

Additionally the user can still use a configuration file or provide a libery path as an argument.

This is way PyVISA (ni backend) operates now: 
- gets all matching libraries 
- prepends a library provided by the user in a configuration file
- try to open them sequentially until one works
- inform you at every step via logging

MatthieuDartiailh, 2019-03-06:
If you have any idea about the resource issue @hgrecco please share them. I looked at NI forums briefly but could not find anything relevant.

hgrecco, 2019-03-07:
The grammar for a USB INSTR is: 
`USB[board]::manufacturer ID::model code::serial number[::USB interface number][::INSTR]`
with: 
- `[board]` defaulting to `0`
- `[::USB interface number]` defaulting to `0`

Therefore:
1. `USB0::0x0483::0x7450::SPD3X2R4938::INSTR0` is invalid
2. `USB0::0x0483::0x7450::SPD3XHBD2R4938::0::INSTR` is valid
3. `USB0::0x0483::0x7450::SPD3XHBD2R4938::INSTR` is valid and equivalent to `2`
4. `USB::0x0483::0x7450::SPD3XHBD2R4938::INSTR` is valid and equivalent to `2`

Please notice that in your post you sometimes reported as model code. `0x7450` and others `0x7540`, which correspond to different devices. Same for the serial number: is `SPD3X2R4938` or `SPD3XHBD2R4938`.

Please make sure that you are using the correct model and serial number and let us know.

HansJA, 2019-03-08:
2.	USB0::0x0483::0x7450::SPD3XHBD2R4938::0::INSTR is valid
3.	USB0::0x0483::0x7450::SPD3XHBD2R4938::INSTR is valid and equivalent to 2
I tried 
Ps1 = rm.open_resource(’USB0::0x0483::0x7450::SPD3XHBD2R4938::0::INSTR’)
and
Ps1 = rm.open_resource(’USB0::0x0483::0x7450::SPD3XHBD2R4938::INSTR’) 
both, and both failed as reported in my issue report.
Skickades från E-post för Windows 10

Från: Hans J Albertsson
Skickat: den 7 mars 2019 21:14
Till: pyvisa/pyvisa
Kopia: pyvisa/pyvisa; Author
Ämne: Re: [pyvisa/pyvisa] visa.ResourceManager('@ni') returns eithervisa64.dll or visa32.dll, choosing randomly (#398)

The longer serial number is correct, as is 0x7540.

Neither 2., 3 or 4 was accepted, and returned the error concerning location info was missing
Hans J. Albertsson
From my Nokia 8

hgrecco, 2019-03-08:
> Neither 2., 3 or 4 was accepted, and returned the error concerning location info was missing

This is fine, because they are equivalent. If one fails, they should all fail. The purpose of that part of the answer was to clarify "In the resourceinfo there’s a different string in the resource_name()! An extra 0::" and your first post where you said you used "'USB0::0x0483::0x7450::SPD3X2R4938::INSTR0'

In the most recent code, you said
> The longer serial number is correct, as is 0x7540.

but then in the same post your code uses the other model number:
> I tried Ps1 = rm.open_resource(’USB0::0x0483::0x7450::SPD3XHBD2R4938::0::INSTR’) and Ps1 = rm.open_resource(’USB0::0x0483::0x7450::SPD3XHBD2R4938::INSTR’)

Is this just a typo?

HansJA, 2019-03-08:
In a sense it's a typo, yes. I just copy/pasted your strings from your last
email.

The calls I actually tested used what was actually returned from the
list_resources calls, and I seem to remember that said ...::0x7540::.... I
can't check that now, I'm nowhere near my lab.

Hans J. Albertsson
From my Nokia 8

hgrecco, 2019-03-08:
I do not know if you copied and pasted from your python script or typed in. It would be important that you check again, because in your initial post there is a difference between the model number that you use to open the resource and the one that you posted that was given by `list_resources_info()`

HansJA, 2019-03-08:
Well You may be right.

Just forget my complaint! If it turns out to have actually occured, I’ll report it again.

My problem was I couldn’t copy/paste from the python window. So I probably kept misreading.

Sorry to have bothered you all, thanks for your patience.

Skickades från E-post för Windows 10

Från: Hernan Grecco
Skickat: den 8 mars 2019 11:34
Till: pyvisa/pyvisa
Kopia: Hans J Albertsson; Author
Ämne: Re: [pyvisa/pyvisa] visa.ResourceManager('@ni') returns eithervisa64.dll or visa32.dll, choosing randomly (#398)

I do not know if you copied and pasted from your python script or typed in. It would be important that you check again, because in your initial post there is a difference between the model number that you use to open the resource and the one that you posted that was given by list_resources_info()

MatthieuDartiailh, 2019-03-25:
Closed by #399

closed 2019-03-25 by MatthieuDartiailh