pyvisa/pyvisa #400: Making a device the talker

jondoesntgit, 2019-03-06:
How do you make devices the talker, and your interface the listener?

I'm working with a SRS844 square-wave lock-in amplifier. I'd like to get data off of the device using GPIB. On page 130 of [the manual](https://www.thinksrs.com/downloads/pdfs/manuals/SR844m.pdf) SRS gives these instructions for reading data off of the device in live time:

The transfer mode should be turned on (using `FAST1`) before storage is
started. Then use the `STRD` command (see below) to start data storage. After
sending the `STRD` command, immediately make the SR844 a talker and the
controlling interface a listener. Remember, the first transfer will occur with the
very first point.

I've been poking around the internet for a few hours trying to find out how to make the SR844 the talker.

They give some sample C code to accomplish this, but I don't know how to do it in python:

https://www.thinksrs.com/downloads/pdfs/manuals/SR844m.pdf

```c
// *******************************************************************
// ***** Fast Data Transfer mode test *****
printf("FastAcqData"); // Acquire and transfer data simultaneously
settimeout (11000); // Turn off timeout for SR844 or set the timeout longer
// than the total acquisition time. The timeout measures
// the time to complete a transfer,
// not the time since the most recent byte is received.
TxSr844("FAST1; STRD"); // Turn FAST mode data transfer ON,
// then start scan using the STRD start acquisition
// after 0.5s.
// The STRD command MUST be used if the data transfer
// is to be started by this program! Do NOT use STRT!
// Take data for 10 seconds and then stop.
// 10 seconds of data at 64 Hz sample rate has
// 640 points.
// Each point consists of X(2 bytes) and Y(2 bytes)
// for a total of 4*(64*10) bytes.
transmit ("MLA TALK 8", &status); // make sr844 the talker right away
j=0; // total bytes received
for (i=0; i<10; i++) { // get 64 points (256 bytes) 10 times
rarray ((char *)(FastBuf+128*i),64*4,&nCount,&status);
// rarray receives bytes from the interface and puts them into the
// FastBuf[] array. Each transfer moves 64 points (64*4 bytes).
// The actual number of bytes transferred is in nCount.
printf ("."); j+=nCount;
// update progress 
```

MatthieuDartiailh, 2019-03-07:
I never used that instrument but it seems that this particular mode of operation relies on fairly low level mechanism. My first question would be: are you sure you need to use that particular kind of transfer ? What is your application for continuous acquisition ?
Because if you need to get one point without stopping the acquisition SNAP seems to be a better solution, if you nee to get a full buffer use TRCA or TRCB.
If you really need that kind of acquisition I can look more in depth but I may need some time to do so.

jondoesntgit, 2019-03-07:
I’m wanting to collect continuous data from an experiment for several hours/days. I spent all day yesterday trying to get it to work with TRCB, but once the buffer fills up, it becomes nearly impossible to reliably obtain new data, as the buffer shifts as you try to read from it (in the middle of a read operation). 

I could do a SNAP operation, but I fear that if I can’t align my clock to the LIA clock then I may read some samples twice or miss some samples.

I agree it looks like a low-level function. I looked into it a little bit yesterday, but didn’t make much progress (I don’t know how to do low-levels GPIB... up until now your library has abstracted all of this away :D)

MatthieuDartiailh, 2019-03-07:
You can try to tinker with the following functions:
- `pass_control` from the ressource object.
- `buffer_read` from the visalib object (accessed by resource.visalib). This function requires a session. From my understanding you should open 'GPIB::INTFC' and get the session from the attribute of the same name.
Let me know how it goes.

hgrecco, 2019-03-08:
The important thing here is `transmit` and `rarray` which are function in the ieee488 library. (`TxSr844` is just using `write` + error checking + polling the status byte).

Transmit commands are defined in the library. 
- `MLA` stands for "My Listen Address" which makes the active board a listener
- `TALK <N>` tells that N is the address of the talker. the can be only one (talker). In the c code, 8 should be the address of the SR844. 

The equivalent in visa is `gpib_command`, so the equivalent pyvisa code should be `rm.visalib.gpib_command(gpib_intfc, "MLA TALK 8"`)`
where `gpib_intfc` is a session from 'GPIB::INTFC' (as @MatthieuDartiailh) said.

`rarray` should be mapped to `buffer_read`

jondoesntgit, 2019-03-08:
I'm at the lab right now working on this. Trying to understand exactly how to grab `gpib_intfc`...

jondoesntgit, 2019-03-08:
A stack trace....

```
>>> sess = rm.open_resource('GPIB::INTFC')
Traceback (most recent call last):
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 130, in get_session_class
    return cls._session_classes[(interface_type, resource_class)]
KeyError: (<InterfaceType.gpib: 1>, 'INTFC')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/pyvisa/resources/resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 192, in open
    cls = sessions.Session.get_session_class(parsed.interface_type_const, parsed.resource_class)
  File "/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 133, in get_session_class
    (interface_type, resource_class))
ValueError: No class registered for InterfaceType.gpib, INTFC
```

MatthieuDartiailh, 2019-03-08:
That is weird. Can you print `rm._resource_classes` ?  That class should be registered (the resource is defined in gpib.py).

hgrecco, 2019-03-08:
`pyvisa-py` does not have a `INTFC` resource class implemented yet.

jondoesntgit, 2019-03-08:
```
>>> rm._resource_classes
{(<InterfaceType.unknown: -1>, ''): <class 'pyvisa.resources.resource.Resource'>, (<InterfaceType.rsnrp: 33024>, 'INSTR'): <class 'pyvisa.resources.messagebased.MessageBasedResource'>, (<InterfaceType.gpib: 1>, 'INSTR'): <class 'pyvisa.resources.gpib.GPIBInstrument'>, (<InterfaceType.gpib: 1>, 'INTFC'): <class 'pyvisa.resources.gpib.GPIBInterface'>, (<InterfaceType.vxi: 2>, 'BACKPLANE'): <class 'pyvisa.resources.vxi.VXIBackplane'>, (<InterfaceType.vxi: 2>, 'INSTR'): <class 'pyvisa.resources.vxi.VXIInstrument'>, (<InterfaceType.vxi: 2>, 'MEMACC'): <class 'pyvisa.resources.vxi.VXIMemory'>, (<InterfaceType.tcpip: 6>, 'INSTR'): <class 'pyvisa.resources.tcpip.TCPIPInstrument'>, (<InterfaceType.tcpip: 6>, 'SOCKET'): <class 'pyvisa.resources.tcpip.TCPIPSocket'>, (<InterfaceType.asrl: 4>, 'INSTR'): <class 'pyvisa.resources.serial.SerialInstrument'>, (<InterfaceType.usb: 7>, 'INSTR'): <class 'pyvisa.resources.usb.USBInstrument'>, (<InterfaceType.usb: 7>, 'RAW'): <class 'pyvisa.resources.usb.USBRaw'>, (<InterfaceType.firewire: 9>, 'INSTR'): <class 'pyvisa.resources.firewire.FirewireInstrument'>, (<InterfaceType.pxi: 5>, 'INSTR'): <class 'pyvisa.resources.pxi.PXIInstrument'>, (<InterfaceType.pxi: 5>, 'MEMACC'): <class 'pyvisa.resources.pxi.PXIMemory'>}
```

MatthieuDartiailh, 2019-03-08:
Indeed I missed that you were using pyvisa-py. Can you access Ni-VISA ? It would be easier. Otherwise I can look for workarounds.

jondoesntgit, 2019-03-08:
I'm trying to get this to work on a Raspberry Pi, and I had to build visa from scratch...

I'm looking for the header library. Here is an output of `$ sudo find / -name "*visa*"`

```
find: ‘/run/user/1000/gvfs’: Permission denied
/home/pi/linux-5d63a4595d32a8505590d5fea5c4ec1ca79fd49d/arch/sparc/include/asm/visasm.h
/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/visa.py
/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/pyvisa-py
/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/__pycache__/visa.cpython-36.pyc
/home/pi/miniconda3/envs/py36/lib/python3.6/site-packages/pyvisa
/home/pi/miniconda3/envs/py36/bin/pyvisa-shell
/home/pi/miniconda3/envs/py36/bin/pyvisa-info
/home/pi/miniconda3/envs/py36/share/terminfo/v/visa50
/home/pi/miniconda3/pkgs/ncurses-6.1-h4f752ac_1/share/terminfo/v/visa50
```

MatthieuDartiailh, 2019-03-08:
I don't think you can install Ni-VISA on the Pi. I will try to look for workaround but I probably won't have a solution today.

jondoesntgit, 2019-03-08:
Thank you.

It seems like this may be a lot of effort on your part. I think I'll just settle for the `SNAP` option.

jondoesntgit, 2019-03-08:
Using OUTR (similar to snap), I can get about 24 samples per second. I can get up to 800 samples per second using TRCB, but I can't be collecting data while reading off the samples (which causes periodic holes in my data). 

If you ever find a way to resolve this, I'd really appreciate it. It would allow me to see frequencies faster than my current sampling rate of ~24 Hz.

hgrecco, 2019-03-08:
@jondoesntgit If you want to implement INTFC in pyvisa-py I think that you can copy the implementation for INSTR [here](https://github.com/pyvisa/pyvisa-py/blob/master/pyvisa-py/gpib.py#L68) and then basically write to the controller not the instrument. It will require other tweaks but it should give you a start.

closed 2020-07-23 by MatthieuDartiailh