pyvisa/pyvisa #401: VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.

ruggiamp, 2019-03-17:
Hello,
I am using pyvisa 1.9.1 within a Python 3.6 environment.
I have installed TekVisa and I am working with a Tektronix MDO3024.
I have connected the oscilloscope to the computer and I can detect it with the Tekvisa OpenChoice Instrument Manager.
When I am trying to list resources [ rm.list_resources() ], I get the following:

Traceback (most recent call last):

  File "<ipython-input-3-defa3779928e>", line 1, in <module>
    rm.list_resources()

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\highlevel.py", line 1637, in list_resources
    return self.visalib.list_resources(self.session, query)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 247, in list_resources
    raise e

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 243, in list_resources
    find_list, return_counter, instrument_description, err = self.find_resources(session, query)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\functions.py", line 485, in find_resources
    instrument_description)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)

VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.

I have no idea what is going on: could anyone help me?
Many Thanks
Ruggero

MatthieuDartiailh, 2019-03-17:
Could you post the output of `pyvisa-info` ? Also could you print the `session` attribute of the resource manager ?
Can you also try to directly open the resource since you know the address ?
I have encountered issues with TekVisa in the past but usually I could access instruments.

ruggiamp, 2019-03-17:
Thank you for your help!
I tried to open the source directly but I get the same error:

''
rm.open_resource("USB::0x0699::0x0408::C053707::INSTR")

2019-03-17 18:32:27,059 - pyvisa - DEBUG - viParseRsrcEx(492013408, 'USB::0x0699::0x0408::C053707::INSTR', 'c_ushort(0)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x000002281D693CC8>, <ctypes.c_char_Array_256 object at 0x000002281D693E48>, <ctypes.c_char_Array_256 object at 0x000002281D693DC8>) -> -1073807346
Traceback (most recent call last):

  File "<ipython-input-20-434b00b1ef02>", line 1, in <module>
    rm.open_resource("USB::0x0699::0x0408::C053707::INSTR")

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\highlevel.py", line 1703, in open_resource
    info = self.resource_info(resource_name, extended=True)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\highlevel.py", line 1663, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1383, in parse_resource_extended
    alias_if_exists)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)

VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
''

At the link there is the pyvisa-info I get from the command promt:
[pyvisa-info.pdf](https://github.com/pyvisa/pyvisa/files/2975531/pyvisa-info.pdf)
It seems there might be a problem with backends, isn't it?

I do not know exactly what do you mean with 'session attribute of the resource manager'

MatthieuDartiailh, 2019-03-17:
Indeed it does look like something goes wrong when we try to access to TekVISA. A short term solution might be to download NiVISA. 
To help me debug this could you do the following (if you know how) before switching to NiVISA:
- add print(rm.session) in your python script and post the result
- add `print(err)` in pyvisa/highlevel.py line 1539 and re-run your script and post the result. This file should be in your python install under site-packages.

ruggiamp, 2019-03-17:
Done.
This is what I get:
"
import visa

rm = visa.ResourceManager()
StatusCode.success

print(rm.session)
2627372896

rm.list_resources()
Traceback (most recent call last):

  File "<ipython-input-4-defa3779928e>", line 1, in <module>
    rm.list_resources()

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\highlevel.py", line 1637, in list_resources
    return self.visalib.list_resources(self.session, query)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 247, in list_resources
    raise e

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 243, in list_resources
    find_list, return_counter, instrument_description, err = self.find_resources(session, query)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\functions.py", line 485, in find_resources
    instrument_description)

  File "C:\Users\ruggero\Anaconda3\envs\QuantumLabControlPy36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)

VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
"

Is there anything else I can do?
Thank you again for your support

MatthieuDartiailh, 2019-03-17:
Stupid question, did you close Tektronik OpenChoice ? if not check if you get the same error once it is closed.

Thanks for the infos. What is weird is that the library pretend it managed to open the resource manager but later complain that the session id is wrong. They may be some conversion issue but it is surprising since I have never seen that kind of issue with Ni. If you can definitively try Ni or Keysight VISA.

ruggiamp, 2019-03-17:
Yeah, I also had the doubt that Tektronix OpenChoice could interfere, so I closed it before starting python...
I'll uninstall Tekvisa and try with NiVisa then!
Thank you

ruggiamp, 2019-03-17:
A naive question: I am using python on a different environment which is not the root one... Might that be  the problem?

MatthieuDartiailh, 2019-03-17:
No it should not be a problem. In particular it cannot affect the session id validity.

MatthieuDartiailh, 2019-05-31:
Did you fix your problem using Ni-VISA ?

jealcuna, 2020-05-21:
rm.my_instrument.close() at the end of your script. This works for me

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh

rebolloluis, 2021-10-07:
After reading a bit more of PyVISA's doc, it eventually mentions something along the lines I explain above at [An example](https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#an-example). Now I feel a little embarrassed. It might be useful to complement a bit more the first example on the doc, to make sure we can actually execute it without issues before going to the next pages 😅 .

MatthieuDartiailh, 2021-10-08:
Hi @rebolloluis,

I have seen dll from different vendor do all kind of things including redirect from 32 to 64 bits (in particular true 32 bits dll are expected to be in SysWOW64 not System32). Anyway we could do several things:
- when both visa32.dll and visa64.dll are present pick the one matching Python bitness that would make your case work out of the box
- improve the documenation
I am currently swamped with other things and answering issues is about as much I can do for PyVISA so if you can make PRs addressing the above points it would be great. I would do my best to review them in a timely manner.