pyvisa/pyvisa #402: GPIB::INTFC resource has no read() or write() methods

dirkenstein, 2019-03-23:
I need to be able to send a particular multi-instrument addressing command that will allow one remote instrument to send a response to two instruments simultaneously.
This comprises a single remote MTA and _two_ MLA addresses, for myself (the controller) and for another device. This is two synchronise two components in an instrument network.  

So the sequence is MTA8 MLA15 MLA30, where i am controller on 30.

GPIB allows this, although it's a pretty obscure edge case for usage it's something a current system i am re-implementing in Python already does.

I can send the command string by opening GPIB0::INTFC and calling send_command()

However, i cannot read or write on the resource opened this way. I believe using the appropriate device resource (I have GPIB0::8::INSTR open) causes system 15 to be unaddressed automatically before the response is sent to both myself and instrument 15.

I believe the underlying VISA implementation should allow this however.

Figuring out how to give the controller read and write functions is a little beyond me given the source code uses a lot of python features i am not deeply familiar with.

Is there a read functoin which leaves the addressing explicitly under control of the user?

MatthieuDartiailh, 2019-03-23:
Something similar actually came up recently in https://github.com/pyvisa/pyvisa/issues/400. 

I am not familiar with that kind of procedure, but you can have a look at the `buffer_read` method of the visa library object (which you can access from the resource object as follow: `resource.visalib`). The first argument (session) can be accessed from the resource: `resource.session`. Please let us if you manage it to work, it may interesting to expand pyvisa to support this use case more naturally.

dirkenstein, 2019-03-24:
Hi, I saw the discussion in issue #400 and tried the suggested buffer_read method which works perfectly, thanks.

`resource.send_command([MTA+ 8, MLA + 15, MLA + 30])`

followed by:

`resource.visalib.buffer_read(resource.session, 3)`

 reads the exact 3 characters from the instrument on GPIB 8 that trigger the other instrument on GPIB 15 whilst not unaddressing  it.

where: 
`resource = rm.open('GPIB0::INTFC')`

So this is possible. The interface is a little clunky but works.

MatthieuDartiailh, 2019-03-24:
I am happy to know you found a working solution @dirkenstein .

I agree that the interface could be improved. What would you propose ? Would exposing `buffer_read` and `buffer_write` on the GPIB interface resource be sufficient ?

Also would you be able to contribute a documentation page on 'advanced techniques' using GPIB bus ? I do not have the experience to write it myself but I believe it would be of great interest to some users.