pyvisa/pyvisa #404: Setting pyvisa-py as default backend in pyvisarc

jondoesntgit, 2019-03-28:
I am working on a system with both NI VISA and pyvisa-py (with linux-gpib) installed. NI-VISA doesn't support USB GPIB controllers, but linux-gpib does. I'd like to use `@py` as the default backend, but I haven't figured out how to do this using the `.pyvisarc` file.

```
$ python -m visa info

python -m visa info
Machine Details:
   Platform ID:    Linux-3.10.0-957.10.1.el7.x86_64-x86_64-with-centos-7.6.1810-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/daq/anaconda3/bin/python
   Version:        3.7.1
   Compiler:       GCC 7.3.0
   Bits:           64bit
   Build:          Dec 14 2018 19:28:38 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: /usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 16777216
         Spec. Version: 5244672
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.2.1')
```

And...

```
>>> import visa
>>> visa.ResourceManager('@py')
<ResourceManager(<VisaLibrary('unset')>)>
```

MatthieuDartiailh, 2019-03-28:
I believe it is not currently possible. I believe you can use the `PYVISA_LIBRARY` environment variable to achieve the same effect. Currently the uses of the pyvisarc file are very limited (I think the only thing it can be used for is the path towards the VISA library). 
Adding the possibility to specify the backend as part of his file would make sense. If you have time to work on this I would be happy to review a PR in that direction.

jondoesntgit, 2019-03-28:
`export PYVISA_LIBRARY='@py'` works for me. Thanks!

closed 2019-03-28 by jondoesntgit