pyvisa/pyvisa #405: highlevel.open_resource return type too abstract

gasparka, 2019-04-01:
Documented return type is `pyvisa.resources.Resource`, which is missing popular functions like `query`, `write` etc.. This leads IDEs to report mistakes in correct code:
[image]
Error: Unresolved attribute reference 'query' for class 'Resource' 

Current workaround is to annotate the type manually:
```Python
from pyvisa.resources import TCPIPInstrument
inst: TCPIPInstrument = rm.open_resource("TCPIP::192.168.1.245")
```
This also gives you the appropriate IntelliSense. Now, it would be nice if PyVisa would handle this by default, but looks like the class hierarchy is quite complex. Ideas?

MatthieuDartiailh, 2019-04-01:
The issue here is that depending on the resource name you can get very different results (including PXI interfaces that do not have read/write/query). I am not sure we can do much better. Manually annotating because you know what kind of resource to expect seems reasonable to me.

gasparka, 2019-04-02:
I agree. Thanks.

closed 2019-04-02 by gasparka