pyvisa/pyvisa #406: Windows Server 2003: GPIB devices not listed

rampDD, 2019-04-04:
Hello, 

I'm trying to implement some measurements using pyvisa to connect to the test equipment.
But I'm neither able to list the devices (list_resources) nor to connect to them (open_resource).

What I've tried is following using visa32.dll:

```
import visa
rm = visa.ResourceManager()
rm.list_resources('?*')
Output: ()
```

And using pyvisa-py with the same result:

```
import visa
rm = visa.ResourceManager('@py')
rm.list_resources('?*')
Output: ()
```

The NI Measurement and Automation Explorer (MAX?) shows 3 devices:
- Keithley Switch Matrix (Address: 23)
- Keithley Source Measure Unit (Address: 24)
- Probestation (Address: 22)

What suprises  me is the fact that there is still a package / software is missing for pyvisa-py GPIB despite gpib-ctypes is installed.

Without '@py' it is directly using visa32.dll and because it is the same backend as NI MAX uses, I would  expect the same output.

Visa Installation Status:
```
C:\Python34\Scripts>python -m visa info
Machine Details:
   Platform ID:    Windows-2003Server-5.2.3790-SP2
   Processor:      x86 Family 6 Model 63 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python34\python.exe
   Version:        3.4.1
   Compiler:       MSC v.1600 32 bit (Intel)
   Bits:           32bit
   Build:          May 18 2014 10:38:22 (#v3.4.1:c0e311e010fc)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: Tektronix
         Impl. Version: 3146496
         Spec. Version: 3145728
   py:
      Version: 0.3.1
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      TCPIP SOCKET: Available
      TCPIP INSTR: Available
```

Pip list:
```
C:\Python34\Scripts>pip.exe list
gpib-ctypes (0.3.0)
pip (1.5.6)
pyserial (3.4)
pyusb (1.0.2)
PyVISA (1.9.1)
PyVISA-py (0.3.1)
setuptools (2.1)
```

So my questions are:
1. Why is it not listing devices (with visa32.dll) when NI MAX shows all devices?
2. What is missing to use gpib visa pyvisa-py ?

Please note that this is on a Windows server 2003 machine (I know it's old but i can't change it).
That is also the reason why I did not attach it to similar reported issues of linux systems.

Thanks in advance and Kind Regards,

MatthieuDartiailh, 2019-04-04:
Hi,

Looking at the output of `pyvisa-info`, I noticed that the detected VISA library is from Tektronik. Are you sure it is the same one as used by NI-Max (I would expect NI) ? NI-Max allows to define the notion of primary and secondary VISA per resource, and here I am wondering if pyvisa may not be picking a different library compared to NI-Max. If you can I would suggest uninstalling the Tektronik VISA.

As for pyvisa-py gpib-ctypes is automatically used only in the development version at the moment. You can install it using `pip install htpps://github.com/pyvisa/pyvisa-py/tarball/master`. Otherwise you have to manually enable it (please refer to gpib-ctypes documentation).

rampDD, 2019-04-05:
Enabling gpib_ctypes as default does not change the result:
```
from gpib_ctypes import make_default_gpib
make_default_gpib()

import visa
rm = visa.ResourceManager("@py")
print(rm.list_resources('?*'))
Output: ()
```

I'll report back when I had a look on the options I've to change the visa32.dll.
It is the only visa32.dll on the entire C drive, so doubt that MAX uses another dll.

MatthieuDartiailh, 2020-09-07:
Closing for lack of activity. Feel free to reopen if necessary.

closed 2020-09-07 by MatthieuDartiailh