pyvisa/pyvisa #407: How to cite PyVISA?

lucask07, 2019-04-10:
I would like to cite PyVISA in an academic article. How to do this? 

My current bibtex entry is:

> @misc{pyvisa,
> 	Author = {{PyVISA Authors}},
> 	Title = {{P}y{VISA} Documentation},
> 	Url = {https://pyvisa.readthedocs.io/en/stable/},
> 	Year = {2018},
> 	Bdsk-Url-1 = {https://pyvisa.readthedocs.io/en/stable/}
> 	}

If there is a preferred citation could this be added to the documentation?

MatthieuDartiailh, 2019-04-10:
I don't think there is any paper you can site. @hgrecco could you confirm that ?
I would use the short url https://pyvisa.readthedocs.io (the one you specified is currently broken it appears). Otherwise this looks ok.

hgrecco, 2019-04-11:
There is no pyvisa publication

vstadnytskyi, 2019-06-27:
One can use https://zenodo.org where you will need to make a copy of the version of pyVISA library that was used in your work. You will get a Digital Object Identifier (DOI) which you can cite.

vstadnytskyi, 2019-06-27:
> I would like to cite PyVISA in an academic article. How to do this?
>
> My current bibtex entry is:
>
>
> If there is a preferred citation could this be added to the documentation?

This citation is umbiguous because the content your are citing will change and will not reflect what was used in your work. Hence, it will become time-dependent.