pyvisa/pyvisa #410: USB connection - Keithley works, SRS Lock-in timeout

Rromotpd, 2019-04-29:
I want to control both devices: Keithley multimeter and SRS Lock-in with Raspberry Pi. The issue is that Keithley works perfect - I can write commands / read data from device, but no matter what I do, my SRS Lock-in timeout in every write command.

```
>>> import visa
>>> visa.log_to_screen()
>>> rm=visa.ResourceManager("@py")
2019-04-29 15:41:01,053 - pyvisa - DEBUG - SerialSession was correctly imported.
2019-04-29 15:41:01,165 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2019-04-29 15:41:01,191 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2019-04-29 15:41:01,204 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2019-04-29 15:41:01,207 - pyvisa - DEBUG - Created library wrapper for unset
2019-04-29 15:41:01,212 - pyvisa - DEBUG - Created ResourceManager with session 6482890
>>> rm.list_resources()
('ASRL/dev/ttyAMA0::INSTR', 'USB0::46342::8192::003398::0::INSTR', 'USB0::1510::8464::8008202::0::INSTR')
############################# KEITHLEY MULTIMETER ####### 
>>> k=rm.open_resource('USB0::1510::8464::8008202::0::INSTR')
2019-04-29 15:41:23,342 - pyvisa - DEBUG - USB0::1510::8464::8008202::0::INSTR - opening ...
2019-04-29 15:41:23,631 - pyvisa - DEBUG - USB0::1510::8464::8008202::0::INSTR - is open with session 3820438
>>> k.query("*IDN?")
2019-04-29 15:41:51,899 - pyvisa - DEBUG - USB0::1510::8464::8008202::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
'KEITHLEY INSTRUMENTS INC.,MODEL 2110,8008202,02.02-02-01\n\x00\x00\x00'
######################################################
############################# SRS LOCK-IN ####### 
>>> s=rm.open_resource('USB0::46342::8192::003398::0::INSTR')
2019-04-29 15:42:20,745 - pyvisa - DEBUG - USB0::46342::8192::003398::0::INSTR - opening ...
2019-04-29 15:42:21,038 - pyvisa - DEBUG - USB0::46342::8192::003398::0::INSTR - is open with session 2920480
>>> s.query("*IDN?")
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 225, in write
    return self.usb_send_ep.write(data)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 824, in bulk_write
    timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 563, in query
    self.write(message)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 223, in write
    count = self.write_raw(message.encode(enco))
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 201, in write_raw
    return self.visalib.write(self.session, message)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 368, in write
    ret = self.sessions[session].write(data)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/usb.py", line 126, in write
    count = self.interface.write(data)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 314, in write
    bytes_sent += raw_write(data)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 227, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
######################################################
```

Any suggestions how can I establish connection? So far I've tried:
1. Change read_termination and write_termination for: \r \n \r\n, None for Lock-in, 
2. Switch USB cables,
3. Reset Lock-in to default settings

Nothing works, I'm clueless..

Connection issues are only on Raspberry - on PC with Windows and NI-VISA drivers installed, my scripts for SRS Lock-in work as they should. But it is so weird that Keithley works perfect on both Raspberry and PC.

MatthieuDartiailh, 2019-04-29:
This is tough one. We know that some instruments do not implement properly the USB-TMC protocol. One thing you can try is testing https://github.com/pyvisa/pyvisa-py/pull/186 (after fixing the typo mentioned in the review). Another option you have is testing python-usbtmc (which implements a similar quirk work-around). 
If you find any solution please let me know, I really wish we can improve pyvisa-py.

Rromotpd, 2019-05-02:
I'm more newbie than a programmer - unfortunately I don't know how ca`n I install version you provided. I installed python-usbtmc tho, and timeout is still there.

```
>>> import usbtmc
>>> d1=usbtmc.Instrument("USB0::1510::8464::8008202::INSTR")
>>> d2=usbtmc.Instrument('USB0::46342::8192::003398::INSTR')
>>> d1.ask("*IDN?")
'KEITHLEY INSTRUMENTS INC.,MODEL 2110,8008202,02.02-02-01'
>>> d2.ask("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.5/site-packages/usbtmc/usbtmc.py", line 629, in ask
    self.write(message, encoding)
  File "/home/pi/.local/lib/python3.5/site-packages/usbtmc/usbtmc.py", line 609, in write
    self.write_raw(str(message).encode(encoding))
  File "/home/pi/.local/lib/python3.5/site-packages/usbtmc/usbtmc.py", line 492, in write_raw
    self.bulk_out_ep.write(req, timeout=int(self.timeout*1000))
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 824, in bulk_write
    timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

```

Rromotpd, 2019-05-02:
I have connected SRS Lock-in with RS-232 to USB adapter, which I use with Keithley 2400 Source Meter and it works flawlessly..

```
>>> import visa
>>> rm=visa.ResourceManager('@py')
>>> rm.list_resources()
('ASRL/dev/ttyUSB0::INSTR', 'ASRL/dev/ttyAMA0::INSTR', 'USB0::46342::8192::003398::0::INSTR')
>>> d1=rm.open_resource('ASRL/dev/ttyUSB0::INSTR')
>>> d1.write_termination = '\r\n'
>>> d1.read_termination = '\r\n'
>>> d1.baud_rate = 9600
>>> d1.query("*IDN?")
'Stanford_Research_Systems,SR865A,003398,v1.51'
>>> d1.query("SNAPD?")
'2.0302819426e-09,4.9405222136e-09,5.3414237478e-09,67.66003418'

```

MatthieuDartiailh, 2019-05-02:
Good you found a workaround. Could you nonetheless test the following just in case ?
`pip uninstall pyvisa-py` followed by `pip install https://github.com/joedoesntgit/pyvisa-py/tarball/master` (this will install the code from the PR I mentioned earlier, but the typo has been fixed now).
I will try to test this on my end too (but those lock-in are used a lot in my lab it may take some time).

MatthieuDartiailh, 2019-05-02:
By curiosity was the Keithley you mentioned in your first message using an adapter or USBTMC ?

Rromotpd, 2019-05-02:
Thanks for the solution. I reinstalled the library but still got timeout on write command.
I checked Keithley 2110 and Keithley 2100 multimeters with USB cables - they're working good with both PYVISA and USBTMC libraries on my RPi. Unfortunately I have only one lock-in in my company so I can't really check if there's problem only with this one (but, again, it works on Windows).

I will try to solve this problem if I will have some free time, for now I will just use adapter or do the measurements on Windows' PC

MatthieuDartiailh, 2019-05-02:
If you want to dive deeper you can have a look at wireshark to capture the usb packets. It may be interesting to compare the Keithley messages and the lock-in messages.

MatthieuDartiailh, 2019-05-02:
And thanks for testing !