pyvisa/pyvisa #412: Timeout expired before operation completed

juniorcl, 2019-05-20:
Hi. I use PyVISA since 2017.
I use it to control the Lock-in (model SR830). 

But since friday I am having problems to control it.
When I type the command `inst.query('SNAP?')`, I get the error below:

> Traceback (most recent call last):
>   File "<pyshell#9>", line 1, in <module>
>     inst.query('SNAP?')
>   File "C:\Users\User\AppData\Local\Programs\Python\Python36\lib\site-packages\pyvisa\resources\messagebased.py", line 569, in query
>     return self.read()
>   File "C:\Users\User\AppData\Local\Programs\Python\Python36\lib\site-packages\pyvisa\resources\messagebased.py", line 413, in read
>     message = self._read_raw().decode(enco)
>   File "C:\Users\User\AppData\Local\Programs\Python\Python36\lib\site-packages\pyvisa\resources\messagebased.py", line 386, in _read_raw
>     chunk, status = self.visalib.read(self.session, size)
>   File "C:\Users\User\AppData\Local\Programs\Python\Python36\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1584, in read
>     ret = library.viRead(session, buffer, count, byref(return_count))
>   File "C:\Users\User\AppData\Local\Programs\Python\Python36\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
>     raise errors.VisaIOError(ret_value)
> pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation 
> 

Therefore I tried to reinstall Python and the NIVISA driver, however nothing worked.
The command` inst.query('*IDN?')` works well.

Thank you!

MatthieuDartiailh, 2019-05-20:
This sounds weird. Since you are on windows I would test the same command using NI-MAX (or the Keysight equivalent).
Also according to the documentation (https://www.thinksrs.com/downloads/pdfs/manuals/SR830m.pdf) you are supposed to pass the index of the measurement you want. Did you update the firmware of the instrument recently ? It may have become stricter in that respect.

juniorcl, 2019-05-21:
> This sounds weird. Since you are on windows I would test the same command using NI-MAX (or the Keysight equivalent).
> Also according to the documentation [...]

I didn't update the firmware and it's happening on two different computers and lock-ins.

I installed old versions of the NI-VISA (http://www.ni.com/pt-br/support/downloads/drivers/download.ni-visa.html#305862) and NI-488.2 (http://www.ni.com/pt-br/support/downloads/drivers/download.ni-488-2.html#305442), but it didn't work either.

MatthieuDartiailh, 2019-05-21:
Can your reproduce using Ni-MAX ?

juniorcl, 2019-05-22:
I solved the problem, but I do not know how.

I just installed the version 15.0 and 19.0 of the NI-VISA and NI-488.2 in different computers (now using NI-MAX to send the commands).

At first, the command `'SNAP?'` didn't work, however fortunately `'SNAP?1,2'` worked.
Afterwards, the same command worked using pyvisa on all the other devices.

I really don't know what was the problem, as I had used the same commands before and they were not working (the program of the lab always use the `'SNAP?1,2,3,4'`).

Finally everything works again.

Thanks very much again for your help.

closed 2019-05-22 by juniorcl

MatthieuDartiailh, 2019-05-22:
Happy it is working again !