pyvisa/pyvisa #414: List of open Resources

ruggiamp, 2019-05-28:
Hi,

I would like to know how to retrieve a list of just the resources that are already open.
Or, maybe, how do I ask a resource if it is open or not.

Many thanks
Ruggero

MatthieuDartiailh, 2019-05-28:
There is no way to do that in 1.9.1. If you are willing to work on the master branch you can access the open resources from the resource manager (attribute `_created_resources`).

Could you describe your use case ? I am wondering if we should expose a Public API to do that.

This feature is expected to be released in the next release (in July a priori)

ruggiamp, 2019-05-28:
Thanks for the promt answer!
I am dealing with many devices at the same time and I might open and close them in different moments.
I just wanted to have the possibility to know in realtime what is the situation: how many and which devices are open. 
I ended up dealing with that by locally keeping track of each resource. It works but it would be better if there was a method already implemented in the ResourceManager I think.

Have a nice day!
Ruggero

MatthieuDartiailh, 2019-05-28:
I could add a public method returning a snapshot of the resources opened at the time of the call. `list_opened_resources` for example returning a list based on the WeakSet the resource manager keeps for book keeping purposes. WDYT ?

ruggiamp, 2019-05-28:
It seems good!
I think it would be great if the output was just a list of the open devices, without any additional information, so to be easily readable...

MatthieuDartiailh, 2019-05-28:
Do you mean a list of resource names ? I feel it would be more flexible to return the actual resource objects. Then people can do `[r.resource_name for r in rm.list_opened_resources]`. (I need to check if `resource_name` is publicly accessible).

ruggiamp, 2019-05-28:
Yep, I meant a list of resource names. I didn't know that possibility...

MatthieuDartiailh, 2019-07-23:
Addressed by #415

closed 2019-07-23 by MatthieuDartiailh