pyvisa/pyvisa #417: Issues Writing to Keithley 2611 Sourcemeter with Visa

rjserpe18, 2019-06-04:
I have a Keithley 2611 Sourcemeter connected to my computer via GPIB to USB. Visa behaves as I would expect when I ask it to list its resources. It provides the correct address for the source meter, and lets me set it as a new resource. 

I am also able to query the Keithley for its IDN, which it provides correctly. 

However, when I try and run the example code from the pyVisa documentation, "A More Complex Example", I encounter problems. 

Pasted below is the code from that example: 

>>> keithley = rm.open_resource("GPIB::12")
>>> keithley.write("*rst; status:preset; *cls")

>>> interval_in_ms = 500
>>> number_of_readings = 10
>>> keithley.write("status:measurement:enable 512; *sre 1")
>>> keithley.write("sample:count %d" % number_of_readings)
>>> keithley.write("trigger:source bus")
>>> keithley.write("trigger:delay %f" % (interval_in_ms / 1000.0))
>>> keithley.write("trace:points %d" % number_of_readings)
>>> keithley.write("trace:feed sense1; feed:control next")

>>> keithley.write("initiate")
>>> keithley.assert_trigger()
>>> keithley.wait_for_srq()

When I run this portion of the code, the editor gets hung up on the last line (wait_for_srq()), giving me the following error: 

Traceback (most recent call last):
  File "/Users/rex/PycharmProjects/EQEMeasurement/Multimeter.py", line 28, in <module>
    keithley.wait_for_srq()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyvisa/resources/gpib.py", line 111, in wait_for_srq
    self.wait_on_event(constants.VI_EVENT_SERVICE_REQ, adjusted_timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyvisa/resources/resource.py", line 342, in wait_on_event
    event_type, context, ret = self.visalib.wait_on_event(self.session, in_event_type, timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyvisa/ctwrapper/functions.py", line 1848, in wait_on_event
    byref(out_event_type), byref(out_context))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pyvisa/ctwrapper/highlevel.py", line 188, in _return_handler
    raise errors.VisaIOError(ret_value)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

The Keithley also displays "ERROR -285" on its readout when I try and run any of these write commands. It doesn't throw the timeout error until I ask it to wait for srq. 

Any idea what might be causing this timeout?

MatthieuDartiailh, 2019-06-04:
That example has been written for a Keithley 2000 not a 2611. Did you check in the manual that the same commands are supported ? Does the Keithley complains on the front panel before the srq ?

rjserpe18, 2019-06-04:
> That example has been written for a Keithley 2000 not a 2611. Did you check in the manual that the same commands are supported ? Does the Keithley complains on the front panel before the srq ?

I have not checked the manual- I'll do that.

In the mean time, yes. I went through line by line, and the Keithley complaint on the front error appears at this line: 

>>>keithley.write('*rst; status:preset; *cls')

closed 2019-07-23 by MatthieuDartiailh