pyvisa/pyvisa #418: Trying to run an manual example in Rohde & schwarz RTB2004, nothing happens

tarciszera, 2019-06-06:
Hello guys, i'm trying to do get some x/y information from the osciloscope (time and voltage). For this I've tried to run an [example code] from the manual of the equipment, nothing happened.

So I found this [issue] and tried to just run the code, it worked somehow but I'm just getting the voltage information. I've tried to add the ["EXP:WAV:INCX ON"] command to get time and voltage information. But nothing changed :(

Another thing is when I ask something, ["CHAN1:WAV1:DATA:VAL? 0,10"] for example it is getting timeout problems, with a lot of commands from the manual :(

Could you guys help me somehow? To understand what I'm doing wrong, or what I have to study to understand it properly, I'm really lost u.u
**\* The "CHAN1:DATA?" command is working, the variations are not.**

[example code]:https://i.imgur.com/ox8VM2F.png
["EXP:WAV:INCX ON"]:https://i.imgur.com/3unZDRS.png
["CHAN1:WAV1:DATA:VAL? 0,10"]:https://i.imgur.com/GkIQl8R.png
[issue]:https://github.com/pyvisa/pyvisa/issues/306

Here the code I'm using (From the other [issue]):

``` python
# RTO/RTE Oscilloscope example for Python using PyVISA

import visa
import matplotlib.pyplot as plt
#import VISAresourceExtentions

# -----------------------------------------------------------
# Initialization
# -----------------------------------------------------------
rm = visa.ResourceManager()
RTB = rm.open_resource('USB0::0x0AAD::0x01D6::107205::0::INSTR')

# The file VISAresourceExtentions.py must be in the same folder as this file
# It contains PyVISA Visa.Resource extension functions for and 2 new exception types

# try block to catch any InstrumentErrorException()

RTB.write_termination = ''
# Some instruments require LF at the end of each command. In that case, use:
# scope.write_termination = '\n'
 #   RTB.ext_clear_status()  # Clear instrument io buffers and status
print(RTB.query('*IDN?'))  # Query the Identification string
#print(RTB.query('ACQ:CURR?'))  # NOTHING HAPPENS
RTB.write('*RST;*CLS')  # Reset the instrument, clear the Error queue
RTB.write("EXP:WAV:INCX ON") # NOTHING HAPPENS

 #   RTB.ext_error_checking()  # Error Checking after Initialization block
# -----------------------------------------------------------
# Basic Settings:
# ----------------------------------------------------------
RTB.write('TIM:SCAL 0.02')
RTB.write('CHAN1:RANG 10.0')  # Horizontal range 5V (0.5V/div)
RTB.write('CHAN1:OFFS 0.0')  # Offset 0
RTB.write('CHAN1:STAT ON')  # Switch Channel 1 ON
 #   RTB.ext_error_checking()  # Error Checking after Basic Settings block
# -----------------------------------------------------------
# Trigger Settings:
# -----------------------------------------------------------
RTB.write('TRIG:A:MODE AUTO')  # Trigger Auto mode in case of no signal is applied
RTB.write('TRIG:A:TYPE EDGE;:TRIG:A:EDGE:SLOP POS')  # Trigger type Edge Positive
RTB.write('TRIG:A:SOUR CH1')  # Trigger source CH1
RTB.write('TRIG:A:LEV1 0.05')  # Trigger level 0.05V
RTB.query('*OPC?')  # Using *OPC? query waits until all the instrument settings are finished
 #   RTB.ext_error_checking()  # Error Checking after Trigger Settings block

# -----------------------------------------------------------
# SyncPoint 'SettingsApplied' - all the settings were applied
# -----------------------------------------------------------
# Arming the RTB
# -----------------------------------------------------------
 
RTB.timeout = 2000  # Acquisition timeout in milliseconds - set it higher than the acquisition time
RTB.write('SING')
# -----------------------------------------------------------
# DUT_Generate_Signal() - in our case we use Probe compensation signal
# where the trigger event (positive edge) is reoccurring
# -----------------------------------------------------------
print('Waiting for the acquisition to finish... ')
RTB.query('*OPC?')  # Using *OPC? query waits until the instrument finished the acquisition
 #   RTB.ext_error_checking()  # Error Checking after the acquisition is finished
# -----------------------------------------------------------
# SyncPoint 'AcquisitionFinished' - the results are ready
# -----------------------------------------------------------
# Fetching the waveform in ASCII format
# -----------------------------------------------------------
print('Fetching waveform in ASCII format... ')
waveformASCII = RTB.query_ascii_values('FORM ASC;:CHAN1:DATA?')
plt.plot(waveformASCII)
print( 'ASCII data samples read: {}'.format(len(waveformASCII)))
 #   RTB.ext_error_checking()  # Error Checking after the data transfer
# -----------------------------------------------------------
# Fetching the trace in Binary format
# Transfer of traces in binary format is faster.
# The waveformBIN data and waveformASC data are however the same.
# -----------------------------------------------------------
print('Fetching waveform in binary format... ')
waveformBIN = RTB.query_binary_values('FORM:BORD LSBF;:FORM REAL;:CHAN1:DATA?')
print( 'Binary data samples read: {}'.format(len(waveformBIN)))
RTB.close()
  #  RTB.ext_error_checking()  # Error Checking at the end of the program

#except VISAresourceExtentions.InstrumentErrorException, e:
    # Catching instrument error exception and showing its content
   # print( 'Instrument error(s) occurred:\n' + e.message

```

MatthieuDartiailh, 2019-06-06:
I have no experience with that instrument but I will try to have a look sometimes next week. Ping me If I forget.

tarciszera, 2019-06-07:
oookay, anyway, thank you very much for the attention :)

MatthieuDartiailh, 2019-06-12:
Looking at the instrument documentation (https://scdn.rohde-schwarz.com/ur/pws/dl_downloads/pdm/cl_manuals/user_manual/1333_1611_01/RTB_UserManual_en_08.pdf) on page 282 you have an example of reading data. In particular it appears that you cannot get the x axis directly as data you have to parse the header to get the information you need.
Regarding your other points:
- "EXP:WAV:INCX ON": is not supported by your instrument (as a general rule of thumb do not expect commands running on one equipment to run completely unmodified on a different and unrelated hardware). All the supported commands are listed starting on page 554.
-  "CHAN1:WAV1:DATA:VAL? 0,10" is not listed in the manual either.

Hope this helps

tarciszera, 2019-06-13:
So in the end I was using the wrong manual? 🤦 thank you a lot! I will try this one. :)

closed 2019-06-13 by tarciszera

carvalhoatx, 2019-08-16:
Tarcis, voce resolveu, Ainda trabalhando com data acquisition remote control?

carvalhoatx, 2019-08-19:
Hello MatthieuDartiailh  and folks. I have four 4 working and the MATH function operation working too (sum between CHannel 1 and Channel 4). Im trying to read the Measurement for all channel (Amplitude, Frequecy and Period). When I use the command " MEASurement<m>:ARES?" it worked nice, but ony to read the measurements set from Channel one. So, im trying to read another and MATH result too. Do you how to do it?  Kind regards,

MatthieuDartiailh, 2019-08-20:
"MEASurement:ARES?" return all the values of the measurements enabled through "MEAS<n> ON" (you can have up to 8, so you cannot get all the parameters from all the traces). I am not sure you can achieve you have in mind.

carvalhoatx, 2019-08-20:
MatthieuDartiailh thank you for the info. It really worked perfect now. The sequence used was:
MEAS1:SOUR C1W1
MEAS1:MAIN AMPL
MEAS1:ADDitional FREQ, On   # to add new MEAS.
MEAS1 ON

MEAS1:RES:ACT?
MEAS1:ARES?

MatthieuDartiailh, 2019-08-20:
I missed MEAS:ADD. Happy you found a solution.