pyvisa/pyvisa #419: The interface type is valid, but the specified interface number is not configured.  VISA error code -1073807195 (0xBFFF00A5), ErrorInterfaceNumberNotConfigured.

eronenergy, 2019-06-07:
I am trying to connect instruments in my lab with my python code .i am able to connect with my USB port devices  but having issue with GPIBO port.

I am getting the below error.

The interface type is valid, but the specified interface number is not configured.  VISA error code -1073807195 (0xBFFF00A5), ErrorInterfaceNumberNotConfigured.

I have attached screenshot 

[image]

[image]

If anyone can guide . what could be this issue??. I am struggling since long time

MatthieuDartiailh, 2019-06-09:
Do you see your instrument in NI-Max/Keysight connection expert ? Do you get the same result using `'GPIB::5::INSTR'` ? That will let VISA pick the proper board address.

You can also have a look at the following stack overflow thread: https://stackoverflow.com/questions/28663308/python-pyvisa-gpib-connection-issue

eronenergy, 2019-06-10:
Hi
@MatthieuDartiailh 

I have both the NI-MAX/Keysight connection expert installed on my pc . I can see that i am able to detect  the device connected in Keysight  connection expert but unable to detect  in NI_MAX.

I have attached the screenshots

my GPIB devices are only two that are mentioned below  as other are getting connected which are on USB and RS 232 serial cable

1-Keysight6813B
2-Agilent6060B

Also i have followed the link mentioned by you in the above comment . I could see that my paths are already set for visa library and also followed third answer and uninstalled NI VISA. 

Now, only devices with RS 232 Serial cable are only running and the usb once has also stopped and getting below error
ValueError: PyUSB does not seem to be properly installed.
Please refer to PyUSB documentation and
install a suitable backend like
libusb 0.1, libusb 1.0, libusbx,

[image]

[image]

eronenergy, 2019-06-10:
Hi
@MatthieuDartiailh

I have both the NI-MAX/Keysight connection expert installed on my pc . I can see that i am able to detect the device connected in Keysight connection expert but unable to detect in NI_MAX.

I have attached the screenshots

[image]

[image]

my GPIB devices are only two that are mentioned below as other are getting connected which are on USB and RS 232 serial cable

1-Keysight6813B
2-Agilent6060B

Also i have followed the link mentioned by you in the above comment . I could see that my paths are already set for visa library and also followed third answer and uninstalled NI VISA.

Now, only devices with RS 232 Serial cable are only running and the usb once has also stopped and getting below error
ValueError: PyUSB does not seem to be properly installed.
Please refer to PyUSB documentation and
install a suitable backend like
libusb 0.1, libusb 1.0, libusbx,

I think uninstalling NI VISA was not fruitful

Please can you guide me what direction  should I take so that i can connect all my GPIB Devices

MatthieuDartiailh, 2019-06-11:
First as you can see from the Keysight screenshot the board number of your GPIB board is 1 not 0. So as I mentioned before I would suggest to not specify the board number in the address or to specify the correct one.

Could you post the output of `pyvisa-info` ? It seems that the Keysight implementation is not properly detected. You could try to also uninstall Keysight connection expert and then re-install it. My guess is that NI-VISA was your primary visa and when you uninstalled it may have removed the dll from system32. What kind of GPIB adpater are you using ? USB adapters can usually be used only with the library from the manufacturer.

eronenergy, 2019-06-11:
Hi 
@MatthieuDartiailh 
First of all very thank you for guiding us with such quick responses 

I have tried using GPIB Port without the board number but there was no change in the error but there is no error with USB Port Instruments.

Below is the error getting with GPIB Ports:
The interface type is valid, but the specified interface number is not configured. VISA error code -1073807195 (0xBFFF00A5), ErrorInterfaceNumberNotConfigured.

The output for the following command "pyvisa-info" is mentioned below with screenshot

C:\ERON_LAB_TEST>pyvisa-info

Traceback (most recent call last):

  File "c:\users\my computer\appdata\local\programs\python\python37-32\lib\runpy.py", line 193, in _run_module_as_main

    "__main__", mod_spec)

  File "c:\users\my computer\appdata\local\programs\python\python37-32\lib\runpy.py", line 85, in _run_code

    exec(code, run_globals)

  File "C:\Users\My Computer\AppData\Local\Programs\Python\Python37-32\Scripts\pyvisa-info.exe\__main__.py", line 5, in <module>

ImportError: cannot import name 'visa_info' from 'visa' (c:\users\my computer\appdata\local\programs\python\python37-32\lib\site-packages\visa\__init__.py)

[image]

and now we have reinstalled NI and has set NI as primary and keysight as secondary visa and we are using keysight cable for connection to device which is a USB to GPIB converter . Below is the image 

[image]

Please can you let us know what you inspect from all this scenario and what step should we take to get ahead and connect to GPIB devices with the python code.

MatthieuDartiailh, 2019-06-11:
First of all it seems something is wrong with your install. Pyvisa does not install a visa package only a visa module which is used to execute `pyvisa-info`. First try to see what is that visa package and to remove it so that we properly get the output of `pyvisa-info`.

Recently we have seen apparently bad behavior of Python-32bits with Keysight (see https://github.com/pyvisa/pyvisa/issues/337 at the end). Since you have to use Keysight for the adapter to work I would suggest to switch to a 64 bits version of Python and try again. Also make sure Keysight is the preferred visa for GPIB resources.

eronenergy, 2019-06-12:
Hi 
@MatthieuDartiailh 

Thanks a lot for taking follow ups 

We have uninstalled the visa package.

Now below is the  response for the command  pyvisa-info

C:\ERON_LAB_TEST>pyvisa-info

Machine Details:

   Platform ID:    Windows-8.1-6.3.9600-SP0

   Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

 

Python:

   Implementation: CPython

   Executable:     c:\users\my computer\appdata\local\programs\python\python37-32\python.exe

   Version:        3.7.2

   Compiler:       MSC v.1916 32 bit (Intel)

   Bits:           32bit

   Build:          Dec 23 2018 22:20:52 (#tags/v3.7.2:9a3ffc0492)

   Unicode:        UCS4

 

PyVISA Version: 1.10.0.dev0

 

Backends:

   ni:

      Version: 1.10.0.dev0 (bundled with PyVISA)

      #1: C:\windows\system32\visa32.dll:

         found by: auto

         bitness: 32

         Vendor: National Instruments

         Impl. Version: 19922944

         Spec. Version: 5244928

   py:

      Version: 0.3.1

      ASRL INSTR: Available via PySerial (3.4)

      **USB INSTR:

         PyUSB does not seem to be properly installed.

         Please refer to PyUSB documentation and

         install a suitable backend like

         libusb 0.1, libusb 1.0, libusbx,

         libusb-win32 or OpenUSB.

         No backend available**

      **USB RAW:

         PyUSB does not seem to be properly installed.

         Please refer to PyUSB documentation and

         install a suitable backend like

         libusb 0.1, libusb 1.0, libusbx,

         libusb-win32 or OpenUSB.

         No backend available**

      TCPIP INSTR: Available

      TCPIP SOCKET: Available

      GPIB INSTR:

         Please install linux-gpib to use this resource type.

         No module named 'gpib' 

Also the error is still same when the code is executed for GPIB ports:

PFA Screenshot:
[image]

Also I would like to ask as that my NI-MAX  shows my USB connected instruments and RS-232 and both of them are responding in the code also  where as GPIB port instruments are neither displayed in NI_MAX nor running in my code . So can you confirm on this that does  NI-MAX even connect with GPIB Ports instruments connected with the Keysight 82357b cables. If yes What could be the issue that my GPIB instruments are not detected by NI-MAX.

[image]

Also thanks once again for helping us with the issue . we were stuck since long time now i could feel some direction now.

MatthieuDartiailh, 2019-06-12:
As I mentioned in my previous post Keysight cables cannot be used with Ni-MAX (basically the required magic happens inside the Keysight driver, NI would work for a NI adapter).

From my point of the safest path to follow is:
- re-install Keysight library
- define Keysight as the primary VISA for GPIB resources (you can do it from NI-MAX)
- switch to a 64 bits version of Python (you can try again your 32 bits but as mentioned I suspect there is something funny when using a 32 Python with Keysight)

eronenergy, 2019-06-18:
Hi 
@MatthieuDartiailh 
Thanks for such supports

We are now trying the python 64 bits and also trying to avail Ni adapters in a day or two.So that we can even troubleshoot and test with Ni adapters for GPIB resoures also we will be following your pattern for installation Keysight library.
we will update you once we are done with the complete steps.

eronenergy, 2019-06-19:
Hi
@MatthieuDartiailh 

We have tried Ni adapter now with Ni visa only installed in the machine and we are getting the below error displayed on Ni-Max.

[image]

eronenergy, 2019-06-19:
Hi
@MatthieuDartiailh

We have anyhow arranged and  tried using  Ni adapter for connecting instruments  as suggested by you now with Ni visa only installed in the machine and we are getting the below error displayed on Ni-Max.

[image]

Below is the cable we are using:

Part Number: 778927-01

Also we have now switched to 64-bit python version but dedicatedly running machine on Ni visa and trying to connect the Ac source with the Ni adapter.

Please guide what could be the cause of this error and what ahead of this we should try.

MatthieuDartiailh, 2019-06-19:
If National Instrument does not recognize its own cable, there is not much I can suggest except contacting them. Sorry.

You can still give the Keysight cable a shot (with Keysight library) we use some of them in the lab I work in and they work mostly fine.

eronenergy, 2019-06-20:
Hi
@MatthieuDartiailh can u just make us sure according to your experience basis that does keysight instruments supports the Ni adapter using Ni-visa and Ni-488 and also would like to know whether this  error raised on the NI-Max that the GPIB device is not genuine is pointing to the Ni -cable or the keysight device we are connect Ni -cable   

FYI:
currently we are using keysight 6813b instrument with Ni-adapter.

Thanks for such instant supports.

MatthieuDartiailh, 2019-06-20:
The adapter behavior should not depend on the instrument it is connected to. Using a Keysight instrument and a NI cable should work. What I find surprising is that NI-MAX complains about the cable but appear to find the instrument. Can you connect and talk to that instrument, even though the cable is believed to not be genuine ?

eronenergy, 2019-06-26:
Hi @MatthieuDartiailh 

Glad to inform you that finally we are able to connect with the GPIB devices now.
So much thanks for your continuous support.
We have connected the instruments with keysight GPIB cables . Also switched to previous version of  keysight i/o library suite and two version previous of Py-visa and Now no Ni-max  is installed in the  system.

@MatthieuDartiailh thanks once again. we will definitely reach you  again if facing some kind of trouble.

NK

MatthieuDartiailh, 2019-06-26:
Just to know are you using a 32 or a 64 bit python interpreter ?

eronenergy, 2019-06-26:
We are using 64 bit python as suggested by you.

Also i  would need little help. i Have some basic test cases written that sets voltage and frequency to the equipments. we are able to accomplish those test in GPIB and USB based instrument but my test is failing in setting the temperature to chamber (testequity1000)  . I am connecting this with COM PORTS and logs shows that it is connected with test equity 100 but it is not able to set temperature at the chamber.

We are getting the below errors

this error for latest version------
raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: '\x01\x03\x00d\x00\x01ÅÕ'

 this one handles int and str for our emulation and ints for Python 3.x
TypeError: 'str' object cannot be interpreted as an integer------------------------serial2.7

Also i would like to ask you does keysight i/o suite also shows com port instruments 

FYI:
We are running the same test case on python 2.7 with same equipments  with that it is running absolutely fine but with python 3.7 we are facing this issue .

Below are the screenshots:
[image]

[image]

MatthieuDartiailh, 2019-06-26:
Keysight VISA supports serial instruments (which implement the VISA protocol), so you do not need to use pyserial. As for your pyserial errors:
- pyserial cannot send unicode object (the default string representation under Python 3) but only bytes. You should encode your string to bytes (`'aaa'.encode('ascii')`) before writing to the instrument.
- for your second error I cannot figure it out without more code. But as I mentioned you should be able to use Keysight directly.

eronenergy, 2019-07-04:
Hi 
@MatthieuDartiailh 
hope you are fine and doing well.

We are still struggling with making connections with the chamber(test equity 1007) . We are able to make connections with the ports like open port and close port but i am not able to communicate with the device.

[image]

In the output you can see we are getting open connection at com 23 port  as we are using rs232 to usb connector  (triplite connector) for connecting to the computer. Also we are  getting [b' '] in output. can you please guide us what the output says and what can you interpret from this.

MatthieuDartiailh, 2019-07-06:
According to the documentation of the controller of your chamber (https://www.testequity.com/UserFiles/documents/pdfs/F4-Temperature-Controller-User-Manual.pdf) it uses a specific protocol: MODBUS (see page 20). I have no knowledge of this protocol and I cannot provide further assistance on this topic. Note that TestEquity apparently sells ethernet to serial converters that enable the use of VISA to communicate with the controller. However the manual does not go into much detail so you may find it useful to contact TestEquity directly to learn more about it.

MatthieuDartiailh, 2019-07-23:
Closing since the issue reported has been resolved and the last question is out of scope.

closed 2019-07-23 by MatthieuDartiailh