pyvisa/pyvisa #420: Trying to acquire continuous data from Opsens. Stuck for two weeks

iamyevesky, 2019-06-07:
I have been working on how to control my Opsens PicoM device remotely and collect data from it but I have multiple errors that I do not understand why they are happening. If someone could help me out that would be great. I have spent almost two weeks on this.

Here is my complete code:

`# -*- coding: utf-8 -*-
"""
:Created on Tue Jun  4 16:35:26 2019
:Author:   Sena Yevenyo.
:Program:  Reads data from Opsens-PicoM Sensor.
:Return:   Recorded temperatures in list [time, temp] format
           as data1.
:Return:   Recorded temperature in list [temp] format in delay*ntimes seconds
           as data0.
"""

"""
:Log Details:

Machine Details:
   Platform ID:    Windows-10-10.0.14393-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda3\python.exe
   Version:        3.6.3
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct 15 2017 03:27:45 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
      #2: C:\windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
"""
ntimes = 100
delay = 0.02
import pyvisa, time
from pyvisa import constants

"""
:IDE:   Anaconda Spyder Python 3.6
readme: Always .open() and .close() opsens ResourceManager object
        after every query.
        Restart console kernel (Ctrl+.) and reboot Opsens-PicoM to avoid
        running errors.
        Read Opsens Serial Comm. User Manual and pyVisa documentation on  
        ResourceManager class, subclass SerialInstrument 
        to initialize Opsens PicoM.
"""
 
rm = pyvisa.highlevel.ResourceManager()
opsensVisaAddress = 'ASRL3::INSTR'
opsens = rm.open_resource(opsensVisaAddress)
opsens.CR = '\r'
opsens.LF='\n'
opsens.stop_bits = constants.StopBits.one
opsens.parity = constants.Parity.none
opsens.flow_control = constants.VI_ASRL_FLOW_NONE
opsens.data_bits = 8
opsens.baud_rate = 9600

opsens.open()
print(opsens.query("system:idn?\r"))
opsens.close()

opsens.open()
opsens.query("memory:delete:all\r")
opsens.close()

"""
:First Solution
:Status: Not working
         Raises Err-202
"""
#rawData0=''
#opsens.open()
#opsens.query("memory:acquire:start "+str(ntimes)+"\r")
#time.sleep((delay*ntimes)+0.001)
#opsens.close()
#
#opsens.open()
#for x in range(ntimes):
rawData0+=opsens.query("memory1:download? "+str(ntimes)+",0")
time.sleep(delay+0.0001)
#opsens.close()
"""
:Second Solution
:Status: Not working
         Raises VisaIOError
"""
#opsens.open()
#opsens.write("measure.start "+str(ntimes)+"\r")
#time.sleep(ntimes*delay)
#rawData0 = opsens.read("measure.start "+str(ntimes)+"\r")
#opsens.query("measure:stop\r")
#opsens.close()
"""
:Third Solution
:Status: Functions but not optimally
         Raises Err-201, Err-170, Err-140
         Times may not be corresponding to temp value
         
"""
rawData0=''
opsens.open()
for i in range(ntimes):
    #20000 is arbitrary. Must be between 1-INFINITE   
    rawData0+=opsens.query("measure:start "+str(ntimes)+"\r")
    time.sleep((delay)+0.0001)
opsens.close()

data0 = rawData0.split('\n')
removeData = ['\4','\4','CH1','Err -170','Err -201','Err -140', "Err -202",""]
data1 = []

for x in range(len(data0)):
    time=x*float(delay)
    temp = "?"
    if data0[x] not in removeData:
        temp=data0[x]
    data1.append([time,temp])

for y in range(len(removeData)):
    while removeData[y] in data0:
        data0.remove(removeData[y])`

MatthieuDartiailh, 2019-06-09:
I am not familiar with that instrument so my suggestions may be off, and I have a couple of questions based on your code.

One thing you can to simplify your code is set `write_termination` to `\r` so that you do not have to add it manually yourself each time. Also those two lines do not do anything:
```python
opsens.CR = '\r'
opsens.LF='\n'
```
For the same command you use sometimes `write` sometimes `query` (`'measure:start'`, that you sometimes seem to misspell `'measure.start'` !). Those are not equivalent and if the instrument send an answer you should read it. 

In you first solution you close and reopen the resource in between the acquisition and the retrieval of the results. Why ? this seems useless and the instrument may decide to discard the data.

In your second solution you use write and misspelled `'measure:start'`.

In you can provide me with the meaning of the error code it could also help me in providing more relevant suggestions.

iamyevesky, 2019-06-10:
Attached here is the Serial Communications User Manual.
[IMP0019 SCPI Serial comm (SCBG-TMM) rev1.7.pdf](https://github.com/pyvisa/pyvisa/files/3270508/IMP0019.SCPI.Serial.comm.SCBG-TMM.rev1.7.pdf)

Thanks for the tip on the `write_termination` and setting that to `\r`
`measure.start` was a mistake and that gives an error. It is supposed to be `measure:start`. I deleted that part of the code. I rewrote it again to give much more clarity to how I approached the problem but it seems I made a typo with it.

About the `opsens.open()` and `opsens.close()` I used to get errors when I send keep running the same code multiple times if I leave them open and close them at the end (as expected). I actually got this tip from a person facing a similar issue #236 from a comment with suggested this workaround

>The behavior I detailed above (only a single `read()` is successful on a Tektronix TBS 1052B-EDU oscilloscope) has also been reported with Keysight devices and a Tektronix TBS1202B-EDU scope; see https://github.com/pyvisa/pyvisa-py/issues/71

> Although the work-around described above (closing and re-opening the communications session) works for most commands, it does not work when trying to transfer entire waveforms from the oscilloscope.

_Originally posted by @mligare in https://github.com/pyvisa/pyvisa/issues/236#issuecomment-313140398_

This is the reason why I close and opened the resources to avoid running into issues and keep it open only during data collection as in solution 2 and 3 in the original code

I used write and query sometimes for the same commands because query is the same as the combination of write and read in one line of code, but I thought there might be a delay due to sampling of data so I write it first, delay using a buffer `time.sleep(delay)` and then read from the machine.

I used puTTy (URL: https://www.chiark.greenend.org.uk/~sgtatham/putty/) to read continuous data from it recently and it worked out fine, but then I just cannot use Python to replicate it.

MatthieuDartiailh, 2019-06-11:
Since you are using National Instrument VISA implementation and not the pyvisa-py backend, I would say that if you need to systematically close and re-open the connection something else is wrong.

Note that when using write and read separately, read does not take as argument the command. You should just do `opsens.read()`. You can also achieve the same delayed reading using the `delay` argument of `query` or setting the `query_delay` attribute on `opsens`.

Based on my understanding of the manual, I would try the following:
```python
rm = pyvisa.highlevel.ResourceManager()

with rm.open_resource('ASRL3::INSTR') as opsens:
    opsens.write_termination = '\r'
    # Message ends with that sequence by default pyvisa would stop at '\n' 
    # but since most message use '\n' in between part of the answer you would only read 
    # a fraction of the answer which causes issues later on
    opsens.read_termination = '\4\n'
    opsens.stop_bits = constants.StopBits.one
    opsens.parity = constants.Parity.none
    opsens.flow_control = constants.VI_ASRL_FLOW_NONE
    opsens.data_bits = 8
    opsens.baud_rate = 9600
    print(opsens.query("system:idn?"))

    meas = []
    # Request the system to measure 100 points and read the first line of the answer
    opsens.query('measure:start 100')
    # Set the termination to \n to break at the end of each line
    with opsens.read_termination_context('\n'):
        while True:
            # Read a single line
            msg = opsens.read()
            # If we get the end marker stop 
            if msg == '\4':
                break
            # Otherwise split the answer (you probably need to do some additional conversion)
            meas.append(msg.split('\t'))

```

Let me know if you get it to work.

iamyevesky, 2019-10-14:
I tried that but that did not work unfortunately.

I called the manufacturer and this is the code that was recommended

`import serial`
`ser = serial.Serial("COM3",9600, timeout=10) #10 second recording data range`
`unicodestring = "measure:start "+str(ntimes)+"\n"`
`ser.write(unicodestring.encode("ascii"))`
`rawData0 = ser.read(10)`
`rawData0 = rawData0.decode("ascii")`
`data0 = rawData0.split('\n')`

closed 2019-10-14 by iamyevesky