pyvisa/pyvisa #421: ValueError: Please install linux-gpib to use this resource type. No module named 'gpib'

Rozakos, 2019-06-12:
I am trying write some commands to Keithley 2000 multimeter, but i am getting that specific error, any ideas?

Traceback (most recent call last): File "C:/PyTests/Keithley.py", line 4, in <module> keithley = rm.open_resource("GPIB::12") File "C:\PyTests\venv\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource res.open(access_mode, open_timeout) File "C:\PyTests\venv\lib\site-packages\pyvisa\resources\resource.py", line 209, in open self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout) File "C:\PyTests\venv\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource return self.visalib.open(self.session, resource_name, access_mode, open_timeout) File "C:\PyTests\venv\lib\site-packages\pyvisa-py\highlevel.py", line 194, in open sess = cls(session, resource_name, parsed, open_timeout) File "C:\PyTests\venv\lib\site-packages\pyvisa-py\sessions.py", line 170, in __init__ raise ValueError(msg)

MatthieuDartiailh, 2019-06-12:
Can you please post the output of `pyvisa-info` ? And the script you use ?

Note that your resource name is wrong you want `'GPIB::12::INSTR'` (or but it is far less likely `'GPIB::12::INTFC'`). 

Since you are on Windows you need to install Ni-VISA or Keysight VISA and either use it directly (using the ni backend of pyvisa) or as you are doing using pyvisa-py (but pyvisa-py need a gpib library installed which basically means you have to install one of the above mentioned library).

Rozakos, 2019-06-13:
First of all thank you for your support, installing Ni-Visa seemed to have fixed the problem.
Also using either  rm.open_resource("GPIB::16") or rm.open_resource("GPIB::16::INSTR") will do the trick.

I ran on another problem though. Even though i managed to write some basic SCPI commands to the multimeter, when i am trying to create a function i get an error.

Working code with error:

import visa

rm = visa.ResourceManager()
print(rm.list_resources())
inst2000 = rm.open_resource("GPIB::16")
print(rm.session)

def res():
    print((inst2000.query('*IDN?')))

res()

Error Message : 

Exception ignored in: <function Resource.__del__ at 0x02F418A0>
Traceback (most recent call last):
  File "C:\PyTests\venv\lib\site-packages\pyvisa\resources\resource.py", line 108, in __del__
  File "C:\PyTests\venv\lib\site-packages\pyvisa\resources\resource.py", line 243, in close
  File "C:\PyTests\venv\lib\site-packages\pyvisa\resources\resource.py", line 235, in before_close
  File "C:\PyTests\venv\lib\site-packages\pyvisa\resources\resource.py", line 252, in __switch_events_off
  File "C:\PyTests\venv\lib\site-packages\pyvisa\resources\resource.py", line 309, in disable_event
  File "C:\PyTests\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 410, in disable_event
  File "C:\PyTests\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 188, in _return_handler
pyvisa.errors.VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.

Working Code without error : 

import visa

rm = visa.ResourceManager()
print(rm.list_resources())
inst2000 = rm.open_resource("GPIB::16")
print((inst2000.query('*IDN?')))
print(rm.session)

C:\PyTests\venv\Scripts\python.exe C:/PyTests/Keithley.py
('TCPIP0::192.168.100.200::inst0::INSTR', 'ASRL1::INSTR', 'ASRL3::INSTR', 'GPIB0::16::INSTR')
KEITHLEY INSTRUMENTS INC.,MODEL 2000,1187823,A20  /A02  

7747080

Again THANK YOU for your help, i was stuck for a day and a half!

MatthieuDartiailh, 2019-06-13:
This errors is not really related to the fact that you use a function. What happens is that to be able to close a resource the resource manager need to be alive. At the end of your script, both objects are deleted in a somewhat random order. If the resource manager is deleted first the error occurs. The next release of PyVISA will make this impossible (you can test by installing the current master from Github). 
The best practice here is to use a context manager for your resource which will make sure it is properly closed (and to pass the resource to your function as an argument rather than relying on a global).
```
def read_idn(res):
    return res.query('*IDN?')

rm = visa.ResourceManager()
with rm.open_resource("GPIB::12::INSTR") as res:
    read_idn(res)
```

closed 2019-06-17 by MatthieuDartiailh