pyvisa/pyvisa #422: VisaIOError: VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.

Apcha, 2019-06-17:
Hello,

I try to connect my Power Analysor from Rohde&Schwarz,HMC8015 with my computer. First I tried to read the self identification from my device with the following code lines (I use python VISA):

rm = visa.ResourceManager()
name = rm.list_resources()
print(name)

Power_Analysor = rm.open_resource('ASRL3::INSTR')
print(Power_Analysor.query('*IDN?'))
print(Power_Analysor)

which work at first, and shows me: ('ASRL3::INSTR',)
Rohde&Schwarz,HMC8015,035620622,01.403
SerialInstrument at ASRL3::INSTR

 But as soon as I run the code a second time, it shows me: VisaIOError: VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.

If I disconnect my device (USB) and reconnect it, the code works again the first time I run it and then it shows me the error as I run it again. I don't know what to do. I tried everything, checked that no other programms uses the device etc.... Does anyone knows what is happening ?

MatthieuDartiailh, 2019-06-17:
One possible origin of  your problem is that you do not close explicitly the resource at the end of your script and that automatic clean up during shutdown is not perfect (a patch to be included in the next release should make it better). In any case, it is better to ensure that the resource is properly closed by using a context manager:
```python 
rm = visa.ResourceManager()
name = rm.list_resources()
print(name)

with rm.open_resource('ASRL3::INSTR') as Power_Analysor:
    print(Power_Analysor.query('*IDN?'))
    print(Power_Analysor)
```

Apcha, 2019-06-17:
It's working ! Thanks a lot !

MatthieuDartiailh, 2019-06-17:
You are welcome.

closed 2019-06-17 by MatthieuDartiailh

juggernaut545, 2022-11-12:
I was having the same error and was hoping for help on how to resolve it.

I have a lock-in amplifier and Lakeshore Model 335 Temperature Controller connected.
Here is a screenshot of the only COM port that is in use according to device manager. 
[image]

This temperature controller is what I am trying to connect to using Pyvisa. Here is the basic code that I am running: 
```import pyvisa
from lakeshore import Model335
from lakeshore.model_335 import *

rm = pyvisa.ResourceManager()
print(rm.list_resources())

m1 = rm.open_resource('GPIB0::8::INSTR')
m2 = rm.open_resource('ASRL3::INSTR')
```

Here is my output again:
```
('ASRL3::INSTR', 'GPIB0::8::INSTR')
Traceback (most recent call last):

  File ~\Documents\temperaturetestcode.py:16 in <module>
    m2 = rm.open_resource('ASRL3::INSTR')

  File ~\anaconda3\lib\site-packages\pyvisa\highlevel.py:3284 in open_resource
    res.open(access_mode, open_timeout)

  File ~\anaconda3\lib\site-packages\pyvisa\resources\resource.py:278 in open
    self.session, status = self._resource_manager.open_bare_resource(

  File ~\anaconda3\lib\site-packages\pyvisa\highlevel.py:3209 in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py:1851 in open
    ret = library.viOpen(

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py:226 in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore

  File ~\anaconda3\lib\site-packages\pyvisa\highlevel.py:251 in handle_return_value
    raise errors.VisaIOError(rv)

VisaIOError: VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.
```

I have tried looking online but the only answers I could find were those that said to make sure nothing else was connected to the serial port. In device manager, it does not show anything else connected to COM port 3 or any other COM ports for that matter. 
I as well attempted to do what was suggested in the first answer, specifically: 
```rm = visa.ResourceManager()
name = rm.list_resources()
print(name)

with rm.open_resource('ASRL3::INSTR') as Power_Analysor:
    print(Power_Analysor.query('*IDN?'))
    print(Power_Analysor)```

However, I still receive the same error when I use this. The 'ASRL3::INSTR' is the same address that Pyvisa detected, so I just kept the Power_Analysor name. 

When I attempt to change the COM port in device manager, to, for example, COM port 12 (I chose one at random), I receive another error. 
Here is what device manager lists:
![image](https://user-images.githubusercontent.com/29962836/201446105-6634fd58-2a6e-4808-9278-a15639715649.png)

Here is what the new code is: 
```import pyvisa
from lakeshore import Model335
from lakeshore.model_335 import *

rm = pyvisa.ResourceManager()
print(rm.list_resources())

m1 = rm.open_resource('GPIB0::8::INSTR')
m2 = rm.open_resource('ASRL12::INSTR')

my_instrument = Model335(57600)
```

Here is what it outputs:
```
('ASRL3::INSTR', 'ASRL12::INSTR', 'GPIB0::8::INSTR')
Traceback (most recent call last):

  File ~\Documents\temperaturetestcode.py:18 in <module>
    my_instrument = Model335(57600)

  File ~\anaconda3\lib\site-packages\lakeshore\model_335.py:329 in __init__
    TemperatureController.__init__(self, serial_number, com_port, baud_rate, timeout, ip_address,

  File ~\anaconda3\lib\site-packages\lakeshore\temperature_controllers.py:281 in __init__
    GenericInstrument.__init__(self, serial_number, com_port, baud_rate, 7, 1, serial.PARITY_ODD,

  File ~\anaconda3\lib\site-packages\lakeshore\generic_instrument.py:77 in __init__
    self.connect_usb(serial_number, com_port, baud_rate, data_bits, stop_bits, parity,

  File ~\anaconda3\lib\site-packages\lakeshore\generic_instrument.py:189 in connect_usb
    self.device_serial = serial.Serial(port.device,

  File ~\anaconda3\lib\site-packages\serial\serialwin32.py:33 in __init__
    super(Serial, self).__init__(*args, **kwargs)

  File ~\anaconda3\lib\site-packages\serial\serialutil.py:244 in __init__
    self.open()

  File ~\anaconda3\lib\site-packages\serial\serialwin32.py:64 in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))

SerialException: could not open port 'COM12': PermissionError(13, 'Access is denied.', None, 5)
```

The ```my_instrument = Model335(57600)``` line is a call to the Model335 lakeshore class, with 57600 as the baud rate listed in the manual. 

Here is my pyvisa-info output in case it is needed. 
[image]

I am at a loss as to how to proceed. Please let me know if you need any further information. I appreciate your time.

Thank you in advance.

MatthieuDartiailh, 2022-11-12:
When you changed the port number it appears the call to pyvisa succeeded but the call to your class failed when trying to open the connection which makes sense it is the second one. If you remove one of the two calls opening a connection you should be fine.

juggernaut545, 2022-11-16:
Thank you very much for the help. I was able to resolve my issue by doing what you suggested and reinstalling the drivers. 
I greatly appreciate it.