pyvisa/pyvisa #423: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Apcha, 2019-06-17:
Hello,

I try to connect my Power Analyser Rohde&Schwarz,HMC8015 ('ASRL3::INSTR') to my computer, and to read any data that my device can show with python VISA. I have a lot of problems with the line, which allows to read my device data. My code is: 

```
rm = visa.ResourceManager()
name = rm.list_resources()

#using with allows to close explicitly the resource at the end of the script
with rm.open_resource('ASRL3::INSTR') as Power_Analyser:
		
	Power_Analyser.values_format.is_binary = True
	Power_Analyser.values_format.datatype = 'B'
	Power_Analyser.values_format.is_big_endian = False
	Power_Analyser.values_format.container = bytearray

	Power_Analyser.timeout = 25000 #2,5 seconds

	Power_Analyser.write_termination = '\n'
	
	Data = Power_Analyser.query_ascii_values('P?',datatype='s')[0]
	print(Data)

        #write the Data to a file on my PC
        PCfilePath = 'C:\\Users\\ApollineChappard\\Documents\\Python Scripts\\a.txt'
        newFile = open(PCfilePath, "wb")
        newFile.write(Data)
        newFile.close()
```
When I runt it, it shows me:
`VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.`

no matter how big the timeout is set.  I'm guessing that the problem comes from the syntax in 
`Power_Analyser.query_ascii_values('P?',datatype='s')[0] ` but I can't figure out what's the correct syntax.

 I checked on my device manual: [https://scdn.rohde-schwarz.com/ur/pws/dl_downloads/dl_common_library/dl_manuals/gb_1/h/hmc80115/HMC8015_SCPImanual_en_01.pdf](url)
but nothing seems to work and nothing is clearly explained for python VISA and I don't have any experience with that. Does anyone know how to fix the problem ?

MatthieuDartiailh, 2019-06-17:
What are you trying to achieve ? The P? command is not listed in the manual you link, so it does not surprise me it does not work. 

Based on your other issue, it seems you can write and read to the instrument just fine (since it answers to *IDN?). So you need to figure what command you need to use. After start with simply reading the answer in text format (using read or query), once you uderstand the format of the answer you can use read_ascii_values, read_binary_values to parse them in an automatic manner. You can have a look at https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html for more details.

Apcha, 2019-06-24:
I just took the first variable to read in the variables list in my device manual so P is Active power P (Watt). It is listed at page  28 (VIEW:NUMeric:PAGE<n>:CELL<m>:FUNCtion?). And I try to apply this in python.

It's true I can connect my instrument fine, but as you said I can't find the rigth command to read it. Okay I will try with read.

Apcha, 2019-06-24:
I just took the first variable to read in the variables list in my device manual so P is Active power P (Watt). It is listed at page 28 (VIEW:NUMeric:PAGE:CELL:FUNCtion?). And I try to apply this in python.
It's true I can connect my instrument fine, but as you said I can't find the rigth command to read it. Okay i will try that with read.

Apcha, 2019-06-24:
So in the  pyVISA Manual it said that i can read raws with 

```
>>> inst.write('CURV?')
>>> data = inst.read_raw()
```
But what should i write instead CURV? for my instrument ?

MatthieuDartiailh, 2019-06-24:
Your manual states that P is a possible response when you query `VIEW:NUMeric:PAGE:CELL:FUNCtion?` not that it is a valid query. The answer you get to `CHAN:MEAS:DATA?` is simply a list of comma separated values corresponding to what your instrument measured. You can use `read_ascii_values` or `query_ascii_values` to get directly a list or a numpy array of values. You can also request your instrument to return the data in binary format using `CHAN:MEAS:FORM BIN` in which case you should use `read_binary_values`.

I will close since it seems you can communicate with your instrument just fine. Feel free to re-open if you encounter any new issue.

closed 2019-06-24 by MatthieuDartiailh