pyvisa/pyvisa #425: HP33120A - function generator RS232

tarciszera, 2019-06-19:
Hello pyVisa guys,

I'm trying to communicate with a HP33120A equipment through RS232 interface using a serial to USB converter cable for PC, using a female female adapter as the [Manual] says.

**Cable:**

[image]

**F/F adapter:**

[image]

I've installed the cable driver, so it's been recognized by the system (windows 10), it's been found by the resource mananger (rm.resources_info) as resource_name = 'ASRL10::INSTR' (just the cable itself).

The problem here is when I connect to the instrument and try to communicate, nothing happens, nothing change in the instrument output, like frequency for example, and nothing is responsed, triggering the timeout.

What I really wanna know is if **I have to install some driver for instrument?**, if there's some **special configuration to control instruments through RS232?** (I checked the instrument configuration and it's in default, like in the next image) at least I configurate the baudrate to be the same. 

[image]

The possible handShake problem (next image) and the adapter not working problem can't be checked right now, but in the next week, so don't care about it, the objective of this issue is to be sure that the problem is not with NI VISA libraries, instrument driver or pyVisa configuration.

[image]

Here is the code I'm using:

```python

import visa     # pyvisa

rm = visa.ResourceManager()                 # create the resource manager

print(rm.list_resources_info())             # to get the resource name

resource_name = 'ASRL10::INSTR'             # resource name

hp33120a = rm.open_resource(resource_name)  # Open a resource
hp33120a.baudrate = 9600                    # Set the default baudrate of the instrument

hp33120a.timeout =10000                     # huge timeout just to be sure
print(hp33120a)                             # print the resource name basicaly

print(hp33120a.query('*IDN?'))              # wait for some information and it triggers the timeout

# Closing the resource and the manager
hp33120a.close()
rm.close()

```

**PS: I'm already remotely controlling instruments through USB**

[Manual]:https://www.keysight.com/upload/cmc_upload/All/6C0633120A_USERSGUIDE_ENGLISH.pdf

MatthieuDartiailh, 2019-06-19:
Hi,
Could you please post the output of `pyvisa-info` (mostly for reference here) ? 

You do not need a special driver for serial instrument.

Since you are running windows, do you have National Instrument/Keysight VISA implementation installed ? Those come with NI-MAX/Keysight Connection Expert that you can use to quickly debug your connection.

Comparing your instrument setting with PyVISA default value it appears that you need to set the stop_bits:
```python
from pyvisa.constants import StopBits
hp33210.stop_bits = StopBits.two
```

Hope this helps

tarciszera, 2019-06-19:
Hi,
thank you very much for the response,
> do you have National Instrument/Keysight VISA implementation installed ? 

I don't know for sure, I just installed big packages from NI VISA, maybe it comes together.
> NI-MAX/Keysight Connection Expert

this is a software?
> Comparing your instrument setting with PyVISA default value it appears that you need to set the stop_bits:

I will try to do it!

In monday (because of holydays) I will test again with the stop bits and will check the keysight visa implementation and then give you some feedback.

Thank you veeery much, It helped a lot!

MatthieuDartiailh, 2019-06-19:
Yes the big install of NI-VISA comes with NI-MAX (which a program to explore the VISA resources connected to your computer).
Let me know if you manage to fix your issue.

tarciszera, 2019-06-24:
Hello,

- Installed the **[IO Libraries Suite]**, just to be sure, but I think it was already installed.
- I've set the two bits, but nothing changed in the results 
- Used the NI MAX to try a communication, nothing happened.
[image]
[image]
[image]

I think it's kind of wierd that the connected flag have invalidy property value, don't know if it's related to the problem. Observing those results if "everything" is set up rightly in software, I assume that probably the problem is in the cable (?).

I don't know what is pyvisa-info sorry.

I will give up this time, bacause I'm running out of time in the project. :(

Anyway, I didn't manage to find the problem but I really appreciate your help, 
Thank you very much!

Have a nice week!

[IO Libraries Suite]:https://www.keysight.com/main/software.jspx?cc=AT&lc=ger&id=2175637&pageMode=CV

MatthieuDartiailh, 2019-06-24:
I am sorry to hear that you could not fix your problem in time. I will close since this is unrelated to pyvisa itself but is really a configuration/hardware issue. Feel free to re-open if you come back to this.

closed 2019-07-10 by MatthieuDartiailh