pyvisa/pyvisa #426: Not able to get response from device over TCPIP pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

kk20198615, 2019-06-26:
Hi All,

I am trying to set attenuation on JDSU MAP chassis. when i set attenuation i do not get responce and my query is like hung.

I have pasted responce for both getattenuation and set attenuation with debug. I have tried different pyvisa version i have same issue.
my code snippet
===========

    def setVOAattenuation(self, device, chassis, slot, port, attenuationVal):
        try:
            id = self.inst.query("*IDN?")
            print(id)

            if id.__contains__(" M A P") or id.__contains__("MAP-200") :
                self.inst.query("OUTP:ATT %d,%d,%d,%f" % (chassis, slot, port, attenuationVal))
            else:
                self.inst.query("INP:ATT %f" % attenuationVal)
            return
        except Exception as ex:
            print("Unable to communicate with the device %s" % ex)

    """ Get Voa attenuation Set on the VOA port """

    def getVOAattenuation(self, device, chassis, slot, port):
        try:
            id = self.inst.query("*IDN?")
            print(id)
            if id.__contains__(" M A P") or id.__contains__("MAP-200"):
                att = self.inst.query("OUTP:ATT? %d,%d,%d" % (chassis, slot, port))
            else:
                att = self.inst.query("INP:ATT?")
            return att

voa.getVOAattenuation(1,1,2,1)
2019-06-26 12:56:28,901 - pyvisa - DEBUG - viWrite(1L, '*IDN?\r\n', 7, 'c_ulong(7L)') -> 0
2019-06-26 12:56:28,901 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
JDSU,MAP-200,22902,5.0.10
2019-06-26 12:56:29,691 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054133C8>, 20480, 'c_ulong(26L)') -> 0
2019-06-26 12:56:30,002 - pyvisa - DEBUG - viWrite(1L, 'OUTP:ATT? 1,2,1\r\n', 17, 'c_ulong(17L)') -> 0
2019-06-26 12:56:30,002 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:30,273 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054131C8>, 20480, 'c_ulong(6L)') -> 0
Out[4]: u'2.000\n'
voa.setVOAattenuation(1,1,2,1,0)
2019-06-26 12:56:45,844 - pyvisa - DEBUG - viWrite(1L, '*IDN?\r\n', 7, 'c_ulong(7L)') -> 0
2019-06-26 12:56:45,844 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
JDSU,MAP-200,22902,5.0.10
2019-06-26 12:56:46,114 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054131C8>, 20480, 'c_ulong(26L)') -> 0
2019-06-26 12:56:46,394 - pyvisa - DEBUG - viWrite(1L, 'OUTP:ATT 1,2,1,0.000000\r\n', 25, 'c_ulong(25L)') -> 0
2019-06-26 12:56:46,394 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:46,711 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054133C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:46,711 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:47,045 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054131C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:47,045 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:47,365 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054133C8>, 20480, 'c_ulong(20480L)') -> 1073676294

2019-06-26 12:56:47,365 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:47,809 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054131C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:47,811 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:48,135 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054133C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:48,135 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:48,446 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054131C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:48,446 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:48,756 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054133C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:48,756 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:49,056 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054131C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:49,056 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:49,378 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054133C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:49,378 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:49,677 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054131C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:49,677 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-06-26 12:56:49,997 - pyvisa - DEBUG - viRead(1L, <ctypes.c_char_Array_20480 object at 0x00000000054133C8>, 20480, 'c_ulong(20480L)') -> 1073676294
2019-06-26 12:56:49,997 - pyvisa - DEBUG - TCPIP0::10.100.59.51::inst0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

MatthieuDartiailh, 2019-06-26:
Most instruments do not answer to a set operation, only to queries. You can check the instrument manual to be sure but that seems normal. If you want to check the operation succeeded, you can either check for errors (different instruments use different mechanisms: status byte, error queue read using SYST:ERR?), or read the values using the same command you use in your get function.

Feel free to re-open if you need to.

closed 2019-06-26 by MatthieuDartiailh

kk20198615, 2019-06-26:
Set operation is success on device. Attenuation is updated on device. but i expect to return sucess and not hung.

I have not see issue when i used GPIB intstead of Ethernet

kk20198615, 2019-06-26:
> Most instruments do not answer to a set operation, only to queries. You can check the instrument manual to be sure but that seems normal. If you want to check the operation succeeded, you can either [...]

Set operation is success on device. Attenuation is updated on device. but i expect to return sucess and not hung.

I have not see issue when i used GPIB intstead of Ethernet. if you give me clue will be helpfull

MatthieuDartiailh, 2019-06-26:
What does your instrument answer in the case of GPIB ? That difference is surprising.
I looked up the manual of your instrument but could not find a complete one. If you have one in pdf could you share it by email so that I can have a look. 
You should be able to simply use `write` instead of `query` for setting the value and get the expected result.