pyvisa/pyvisa #431: HiSLIP - VI_ERROR_CONN_LOST when transfering large binary data sets

bjaraujo, 2019-06-28:
First reported here:
https://github.com/pyvisa/pyvisa-py/issues/58

In binary transfers, pyvisa doesn't read the header before it reads rest of the content and this may cause problems when data is larger than 20*1024 bytes specially in HiSLIP protocol. In HiSLIP when it reads partial messages VISA produces the error VI_ERROR_CONN_LOST.

I don't think increasing chunk_size is a good fix as then there is always a number above which it might fail.

So my fix is to read the header first and then determine how many bytes to read from visa. I have tested for IEEE format and it works. Then I also changed the HP format for consistency and it may have similar problem (I haven't tested for HP).

https://github.com/pyvisa/pyvisa/pull/430

MatthieuDartiailh, 2019-06-28:
The issue may be a bit more complex. If you look at the code for `read_bytes` that your solution use you will see it uses chunk_size which means that even in your new code the read is done in multiple chunks. I am not sure what to make of it yet, but it would be worth digging a bit more, I think.
If you can, could you time the transfer time of large block size for different chunk sizes. I have a feeling that using larger chunk sizes for large amount of data would make sense and that exposing `chunk_size` as a keyword argument in `read/query_binary_values` would still make sense.

MatthieuDartiailh, 2019-07-23:
closed by #435

closed 2019-07-23 by MatthieuDartiailh