pyvisa/pyvisa #432: VisaIOError: VI_ERROR_INV_SETUP (-1073807302): Unable to start operation because setup is invalid (usually due to attributes being set to an inconsistent state).

Rocknpools, 2019-06-28:
I'm trying to use PyVISA for my project. This is the output when I type "python -m visa info"

```
Machine Details:
   Platform ID:    Darwin-18.6.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /anaconda3/bin/python
   Version:        3.6.8
   Compiler:       GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)
   Bits:           64bit
   Build:          Dec 29 2018 19:04:46 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 32, 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
```

I'm following the getting started tutorial with this:
```
import visa
rm = visa.ResourceManager()
rm.list_resources()
```
and the output is :
```
('USB0::0x0957::0x0118::US56070644::INSTR', 'ASRL1::INSTR', 'ASRL2::INSTR')
```

So this mean that there is an instrument (Keysight  N5227A PNA) connected to the usb port. Then if  I type:

```
my_instrument = rm.open_resource('GPIB0::14::INSTR')
print(my_instrument.query('*IDN?'))`
```

I  get this error:

VisaIOError: VI_ERROR_INV_SETUP (-1073807302): Unable to start operation because setup is invalid (usually due to attributes being set to an inconsistent state).

MatthieuDartiailh, 2019-06-28:
I am not sure to follow you process here. Why after asking VISA to list the available resources do you try to access a completely different one ? Since you are trying to connect to the instrument connected in USB you should do the following:
```python
my_instrument = rm.open_resource('USB0::0x0957::0x0118::US56070644::INSTR')
print(my_instrument.query('*IDN?'))
```

Rocknpools, 2019-06-28:
sorry it's a copy/paste error i originally did what you said

```
my_instrument = rm.open_resource('USB0::0x0957::0x0118::US56070644::INSTR')
print(my_instrument.query('*IDN?'))
```

MatthieuDartiailh, 2019-06-28:
Does the error occur when you open the resource or when you try to talk to it ? Also can you talk to your instrument NI-MAX ? That would be the first thing to get to work since it is often easier to debug when the configuration is wrong.

Rocknpools, 2019-07-01:
The error occur while I try to talk with the instrument (everything is fine when I open it)
What to you mean exactly with "talk to your instrument NI-MAX"? 

This is what I see when I open the NI-MAX app

[image]

I can also open the VISA test panel

[image]

I guess those are the attributes the error is talking about ("attributes being set to an inconsistent state")

MatthieuDartiailh, 2019-07-01:
Under the input/output tab of the VISA test panel you can write and read messages to/from the instrument. Can you check if this works ?

Rocknpools, 2019-07-01:
Same error if I query the instrument from the test panel

[image]

MatthieuDartiailh, 2019-07-01:
This does not look right. I have never seen that kind of error before. I tried to give a look to the manual but could not find anything directly related (but the manual is huge so I may have missed something). At this stage, I would suggest you contact Keysight directly, they are more likely to be able to help than I am. In the meantime, I would suggest you try to communicate over GPIB or TCPIP.  I have used those option succesfully in the past on a PNA (cannot remember the exact model). You can also try to install Keysight VISA implementation, there may be some corner cases they handle better for their instruments.

Rocknpools, 2019-07-01:
Can i "force" pyvisa to use Keysigth VISA library? (despite some possible incompatibility)

MatthieuDartiailh, 2019-07-01:
Usually when multiple VISA library are installed on a 64 bits system, you can specify which one to use for each resource type. To do so, in Connection Expert (Keysight equivalent of NI-MAX), you will see a gear at the top right corner, click it and go to tools and click VISA conflict manager. Then you will be able to select the primary VISA library to use.

bjaraujo, 2019-07-02:
It might be some driver issue in host or device.

Rocknpools, 2019-07-02:
> Usually when multiple VISA library are installed on a 64 bits system, you can specify which one to use for each resource type. To do so, in Connection Expert (Keysight equivalent of NI-MAX), you [...]

But will pyvisa work wwith the Keysigth equivalent of NI-MAX?

MatthieuDartiailh, 2019-07-02:
PyVISA works with the implementation of VISA by Keysight yes (PyVISA does not actually go through Ni-MAX simply through the VISA library that comes with it)

MatthieuDartiailh, 2020-09-07:
Closing for lack of activity. Feel free to reopen if necessary.

closed 2020-09-07 by MatthieuDartiailh

laurenwilliams22, 2020-10-14:
I'm having the same issue now. Were you able to get it working with Keysight?

MatthieuDartiailh, 2020-10-14:
Please open a new issue with the output of pyvisa-info and as many details as you can provide (method of installation of Python and Keysight VISA in particular).