pyvisa/pyvisa #433: GPIB

GitHub191536, 2019-06-28:
Hi,
on the picture is my code with error.
[image]

This is !python -m visa info:
!python -m visa info
Machine Details:
   Platform ID:    Windows-10-10.0.17134-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\HP\Anaconda3\python.exe
   Version:        3.7.3
   Compiler:       MSC v.1915 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 27 2019 17:13:21 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
      #2: C:\windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17827072
         Spec. Version: 5244928
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      **USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and 
         install a suitable backend like 
         libusb 0.1, libusb 1.0, libusbx, 
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and 
         install a suitable backend like 
         libusb 0.1, libusb 1.0, libusbx, 
         libusb-win32 or OpenUSB.
         No backend available**
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      **GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'**
I think that I should install **linux-gpib** and **PyUSB** correctly. The problem is that I have no clue how to install **linux-gpib** and why **PyUSB** does not seem to be properly instaled? I am new to PyVISA, so please write only easy to understand advices (if I should install something without !pip ... please tell me which file executes installation; give me proper directory of installation; ... ). 
Thanks a lot!

MatthieuDartiailh, 2019-06-28:
You do not need to install linux-gpib and pyUSB since you have installed the VISA implementation by National Instrument. The easiest thing to do to debug your issue is to use NI-MAX which is a programmed installed by National Instrument when you install VISA. It should be in your Start menu. There you should see the instrument connected to your computer. And find the proper address

I suspect that the board id is incorrect for example it is possible that the address is not GPIB0::22::INSTR but GPIB1::22::INSTR. From Pyvisa you can also do:
```python
rm = visa.ResourceManager()
print(rm.list_resources())
```
which will display the instruments connected to your computer.

Hope this helps.

MatthieuDartiailh, 2020-09-07:
Closing for lack of activity. Please note that some USB-GPIB adapter appear with a board index of 1 not 0 as is the default. Feel free to reopen if necessary.

closed 2020-09-07 by MatthieuDartiailh