pyvisa/pyvisa #436: Default pyvisa query list_resources.

bjaraujo, 2019-07-02:
Why is the default query something like query=u"?*::INSTR" in list_resources?
I don't think this is obvious and it filters out other devices like raw socket devices.

MatthieuDartiailh, 2019-07-06:
@hgrecco may know the reason behind the choice for that default.

hgrecco, 2019-07-06:
Most newcomers are trying to connect to `INSTR` devices and we thought that it was better to put the burden in most experienced users. This was also the behavior of `get_instruments_list`, the equivalent function of the original PyVISA 

By the way, this is also the choice (or at least used to be) in the LabVIEW VISA Resource Name Control and in some implementations for other languages.

MatthieuDartiailh, 2019-07-06:
Also we clearly state in the documentation that this is how things works by default and how to change it: https://pyvisa.readthedocs.io/en/latest/introduction/communication.html

closed 2019-07-22 by bjaraujo