pyvisa/pyvisa #438: not able to communicate with testequtity1007

eronergy, 2019-07-08:


eronergy, 2019-07-08:
We are 
[image]
In the output you can see we are getting open connection at com 23 port as we are using rs232 to usb connector (triplite connector) for connecting to the computer. Also we are getting [b' '] in output. can you please guide us what the output says and what can you interpret from this.

MatthieuDartiailh, 2019-07-08:
This question is a duplicate of the last one asked at the end of #419. And I cannot provide more guidance than what I already offered in that issue. I copy my answer below. Since that instrument does not use VISA to communicate on the serial port, this is outside the scope of PyVISA. I am closing the issue but feel free to re-open if you deem it necessary.

According to the documentation of the controller of your chamber (https://www.testequity.com/UserFiles/documents/pdfs/F4-Temperature-Controller-User-Manual.pdf) it uses a specific protocol: MODBUS (see page 20). I have no knowledge of this protocol and I cannot provide further assistance on this topic. Note that TestEquity apparently sells ethernet to serial converters that enable the use of VISA to communicate with the controller. However the manual does not go into much detail so you may find it useful to contact TestEquity directly to learn more about it.

closed 2019-07-08 by MatthieuDartiailh