pyvisa/pyvisa #439: ValueError: Could not find hash sign (#) indicating the start of the block.

ApollineChappard, 2019-07-08:
Hello,

I try to connect my Power Analyzer from Rohde&Schwarz,HMC8015 (it's me again sorry lol) and read values that it's measuring. It's working fine with the following command of Python VISA:

`Power_Analyzer.write("CHANnel1:MEASurement:DATA?")
Data = `Power_Analyzer.read_raw()` 

but now I would to try with the same code just with the different command:
`values = Power_Analyser.query_binary_values('CHANnel1:MEASurement:DATA?', datatype='d', is_big_endian=True)`

but it shows me the error : `ValueError: Could not find hash sign (#) indicating the start of the block.`

What can I do to fix it ? And is there a big difference between the 2 commands ? 

Here is my code : 
```
import os
import time
import visa
import binascii
import struct
import csv
import numbers

#reading self-identification from power analyser
rm = visa.ResourceManager()
name = rm.list_resources()
print(name)

bytes is str

#using with allows to close explicitly the resource at the end of the script
with rm.open_resource('ASRL3::INSTR') as Power_Analyser:
        
     print(Power_Analyser.query('*IDN?'))
     print(Power_Analyser.query('SYSTem:DEVice?    '))#Queries the instrument type.

     Power_Analyser.timeout = 2500 #2,5 seconds
     Power_Analyser.write_termination = '\n'

    #read the variables of the instrument
#    Power_Analyser.write("CHANnel1:MEASurement:DATA?")
    
    
     header_fmt='empty'
     expect_termination=False
     values = Power_Analyser.query_binary_values('CHANnel1:MEASurement:DATA?')
     print(values)
```

MatthieuDartiailh, 2019-07-08:
Did you check in what format the instrument is returning the data ? My guess is that your instrument is returning data as ASCII and not in binary form hence the error. Either use `query_ascii_values` or if you need a fast data transfer ask the instrument to send data as binary using `CHANnel<n>:MEASurement:FORMat {BINary | ASCii}` as mentioned in the documentation.

ApollineChappard, 2019-07-08:
Okay it's working with `query_ascii_values` ! I though my instrument would return bytes since  it returned bytes with `Power_Analyser.read_raw()` ... Thanks a lot again 👍 🥇

MatthieuDartiailh, 2019-07-08:
`read_raw` always return a byte string (not a unicode string like `read` does), so it is not a good test. As I mentioned you can use the command `CHANnel<n>:MEASurement:FORMat?` to check the format of the answer and for fast transfer it is usually better to use a binary format. 

I am closing since this is fixed feel free to reopen if the need arises.

closed 2019-07-08 by MatthieuDartiailh

aamirlore, 2023-01-10:
I am currently working with an optical switch instrument by JGR optics and I am facing the same issue.
There's an SCPI command to query the information of the module ```MODule#:INFO? ``` which return a ```'1Ax3,SM,9\xb5m\n'``` and here we have bytearray value in the end which is ```\xb5m\n```. In my python program I am using ```cmd = self.inst.query_ascii_values(f'MODule{module}:INFO?')``` which throws a UniCodeDecodeError ```[undefined]UnicodeDecodeError:'ascii' codec can't decode byte 0xb5 in position 9: ordinal not in range(128)```
and when I use ```cmd = self.inst.binary_ascii_values(f'MODule{module}:INFO?')``` it throws a valueError ```cmd = self.inst.query_ascii_values(f'MODule{module}:INFO?')``` 
Could you please help me to understand what exactly has to be done to get or convert it into a normal string? I am beginner  using pyvisa implementation.