pyvisa/pyvisa #440: 'Device' object has no attribute 'serial_number' , Could not open VISA library:, AttributeError: 'Device' object has no attribute 'serial_number'

ioabro17, 2019-07-10:
Ok my machine status

python -m visa info
Machine Details:
   Platform ID:    Linux-3.10.0-957.21.3.el7.x86_64-x86_64-with-centos-7.6.1810-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.5
   Compiler:       GCC 4.8.5 20150623 (Red Hat 4.8.5-36)
   Bits:           64bit
   Build:          Jun 20 2019 20:27:34 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0b1). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b1). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

From the visa shell I get:

python -m visa shell

Welcome to the VISA shell. Type help or ? to list commands.

(visa) list
'Device' object has no attribute 'serial_number'

When I run the python script:

import visa
visa.log_to_screen()
rm = visa.ResourceManager()
print(rm.list_resources())

Output is:
2019-07-10 13:31:08,228 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2019-07-10 13:31:08,229 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2019-07-10 13:31:08,234 - pyvisa - DEBUG - No user defined library files
2019-07-10 13:31:08,631 - pyvisa - DEBUG - Automatically found library files: []
2019-07-10 13:31:08,632 - pyvisa - DEBUG - Did not find NI binary
2019-07-10 13:31:08,652 - pyvisa - DEBUG - SerialSession was correctly imported.
2019-07-10 13:31:08,704 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2019-07-10 13:31:08,729 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2019-07-10 13:31:08,731 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2019-07-10 13:31:08,732 - pyvisa - DEBUG - pyvisa-py is available.
2019-07-10 13:31:08,732 - pyvisa - DEBUG - Created library wrapper for unset
2019-07-10 13:31:08,733 - pyvisa - DEBUG - Created ResourceManager with session 3854464
Traceback (most recent call last):
  File "pyvisatest.py", line 8, in <module>
    print(rm.list_resources())
  File "/usr/lib64/python2.7/site-packages/pyvisa/highlevel.py", line 1637, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/lib64/python2.7/site-packages/pyvisa-py/highlevel.py", line 326, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/lib64/python2.7/site-packages/pyvisa-py/usb.py", line 227, in list_resources
    serial = dev.serial_number
AttributeError: 'Device' object has no attribute 'serial_number'

If i pass '@py' in the ResourceManager I get:
2019-07-10 13:43:33,616 - pyvisa - DEBUG - SerialSession was correctly imported.
2019-07-10 13:43:33,667 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2019-07-10 13:43:33,694 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2019-07-10 13:43:33,696 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2019-07-10 13:43:33,697 - pyvisa - DEBUG - Created library wrapper for unset
2019-07-10 13:43:33,697 - pyvisa - DEBUG - Created ResourceManager with session 6657516
Traceback (most recent call last):
  File "pyvisatest.py", line 8, in <module>
    print(rm.list_resources())
  File "/usr/lib64/python2.7/site-packages/pyvisa/highlevel.py", line 1637, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/lib64/python2.7/site-packages/pyvisa-py/highlevel.py", line 326, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/lib64/python2.7/site-packages/pyvisa-py/usb.py", line 227, in list_resources
    serial = dev.serial_number
AttributeError: 'Device' object has no attribute 'serial_number'

If i pass '@ni' in the ResourceManager I get:
2019-07-10 13:44:23,634 - pyvisa - DEBUG - No user defined library files
2019-07-10 13:44:24,064 - pyvisa - DEBUG - Automatically found library files: []
2019-07-10 13:44:24,065 - pyvisa - DEBUG - Could not open VISA wrapper <class 'pyvisa.ctwrapper.highlevel.NIVisaLibrary'>: 
Could not open VISA library:

Traceback (most recent call last):
  File "pyvisatest.py", line 7, in <module>
    rm = visa.ResourceManager('@ni')
  File "/usr/lib64/python2.7/site-packages/pyvisa/highlevel.py", line 1526, in __new__
    visa_library = open_visa_library(visa_library)
  File "/usr/lib64/python2.7/site-packages/pyvisa/highlevel.py", line 1498, in open_visa_library
    return cls(argument)
  File "/usr/lib64/python2.7/site-packages/pyvisa/highlevel.py", line 97, in __new__
    raise OSError('Could not open VISA library:\n' + '\n'.join(errs))
OSError: Could not open VISA library:

I have installed Keysight IO suite and from the suite i can actually connect with the instrument, what is wrong with the python code?

MatthieuDartiailh, 2019-07-10:
Not sure this is the solution but usually USB devices require a particular udev rule under linux. A user that could not list USB instrument fixed it by adding the following, however it did not get such a hard error:

My issue is solved, by adding a udev rule.
ATTRS{idVendor}=="f4ed", ENV{ID_MM_DEVICE_IGNORE}="1" ATTRS{idVendor}=="f4ed", MODE="0660", GROUP="adm"
First line is to tell ModemManager to ignore this device. (Ubuntu). Second is to make the scope part of "adm" and to change the privileges to non-root.

Let me know if this helps.

MatthieuDartiailh, 2019-07-10:
This link might help: http://ask.xmodulo.com/change-usb-device-permission-linux.html

ioabro17, 2019-07-12:
Thank you for replying. I tried your sollution and it didn't work. I installed again the NI-VISA drivers using this:
http://www.ni.com/product-documentation/54754/en/

Now it works but it dosn't find the USB. I only get 
(u'ASRL1::INSTR', u'ASRL2::INSTR')
 Working on that now. So till now I recomend anyone who experiences this problems so far to install NI-VISA drivers with this method. Keysight IO is not enough.

closed 2020-09-07 by MatthieuDartiailh