pyvisa/pyvisa #442: Unable to locate visa even though it is installed.

lucamedeir, 2019-07-15:
Hello,

Before I explain the issue, I should say that my system is CentOS7 and I installed the keysight VISA.
When I called visa.ResourceManager(), it says it could not locate a visa binary (you should not install the pyvisa-py to replicate the bug). I tried passing the location on the parameter to no avail. 
However if one creates the .pyvisarc as specified on the documention, it works. 
In summary, the issue is that passing the location on the ResourceManager() does not work.

Best Regards,
Lucas Medeiros

MatthieuDartiailh, 2019-07-16:
Thanks for reporting. Could you post the output of the `pyvisa-info` command, the test script you used when passing directly the library and the path reported for the library when you pass the value using .pyvisarc ?

MatthieuDartiailh, 2019-07-16:
Comparing 1.9.1 and master, I believe that passing the path should work on master. Could you check ?

MatthieuDartiailh, 2019-07-16:
I do not have access to a linux box to test this but if you have time you can try to play with the code in https://github.com/pyvisa/pyvisa/blob/master/pyvisa/ctwrapper/cthelper.py that handles the detection of the library. Let me know if you find anything.

TraceBivens, 2019-07-31:
I'm having a similar issue on a Win10 box, cannot for the life of me figure out how to get ResourceManager() to see the NI backend. I've checked and NI MAX works fine, and have attempted passing the path directly under 1.9.1, master, and 1.8 with no success.

MatthieuDartiailh, 2019-08-01:
As mentioned 1.9.1 suffers from a bug that prevents passing the path from working properly. Try to use master (a new release is planned next week).

TraceBivens, 2019-08-02:
Yep, tried that.  Code and errors follow:

>>> import visa as v
>>> rm = v.ResourceManager("C:\Windows\SysWOW64\visa32.dll")
Traceback (most recent call last):
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 135, in _init
    lib = Library(self.library_path)
  File "C:\Users\Streak\Miniconda3\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\highlevel.py", line 1551, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\highlevel.py", line 1523, in open_visa_library
    return cls(argument)
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\highlevel.py", line 109, in __new__
    obj._init()
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 137, in _init
    raise errors.LibraryError.from_exception(exc, self.library_path)
pyvisa.errors.LibraryError: Error while accessing C:\Windows\SysWOW64 isa32.dll: [WinError 126] The specified module could not be found
>>>

TraceBivens, 2019-08-02:
Just noticed something: in the path listed on the next to last line of the traceback the characters \v are missing, and in the anaconda prompt window there’s a boxed question mark where they should be. Perhaps something somewhere thinks that \v is an escaped character?

On Aug 1, 2019, at 9:36 PM, Robert Bivens <robertbivens@my.unt.edu<mailto:robertbivens@my.unt.edu>> wrote:

Yep, tried that.  Code and errors follow:

>>> import visa as v
>>> rm = v.ResourceManager("C:\Windows\SysWOW64\visa32.dll")
Traceback (most recent call last):
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 135, in _init
    lib = Library(self.library_path)
  File "C:\Users\Streak\Miniconda3\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\highlevel.py", line 1551, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\highlevel.py", line 1523, in open_visa_library
    return cls(argument)
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\highlevel.py", line 109, in __new__
    obj._init()
  File "C:\Users\Streak\Miniconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 137, in _init
    raise errors.LibraryError.from_exception(exc, self.library_path)
pyvisa.errors.LibraryError: Error while accessing C:\Windows\SysWOW64 isa32.dll: [WinError 126] The specified module could not be found
>>>

MatthieuDartiailh, 2019-08-02:
You need to either escape the \ or use a r string (r"C:\Windows\SysWOW64\visa32.dll") so that Python does not interpret \v as a special character. Also save if you are using a 32-bit Python install, you probably want to point to the visa.dll under System32 rather than SysWOW64.

lucamedeir, 2019-08-02:
@MatthieuDartiailh Sorry for taking too long to respond. We were not able to reproduce the mistake given that we do not have access anymore to the computer with linux used before. I thank you for your help nonetheless.

closed 2019-08-02 by lucamedeir