pyvisa/pyvisa #443: PyVisa no module named "GPIB" found

killertoilet, 2019-07-16:
I am trying to connect to an E3631A power supply using an 82357B USB-GPIB connector. running the list_resource command returns nothing. From Keysight's IO libraries I can send commands such as *IDN? and get an expected response. If I try to connect using open_resource, I get the following errors. 

  File "test.py", line 5, in <module>
    rm.open_resource('GPIB0::2::INSTR')
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa-py\highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa-py\sessions.py", line 170, in __init__
    raise ValueError(msg)
ValueError: Please install linux-gpib to use this resource type.
No module named 'gpib'

Here is the info from pyvisa:

Python:
   Implementation: CPython
   Executable:     C:\Users\person\AppData\Local\Programs\Python\Python37\python.exe
   Version:        3.7.4
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Jul  8 2019 20:34:20 (#tags/v3.7.4:e09359112e)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

Why does IO libraries see it jsut fine but pyvisa cannot? I tried installing the linux-gpib as it asked and it says it cannot find it.

MatthieuDartiailh, 2019-07-16:
It seems that Pyvisa does not detect the visa library part of Keysight, which is weird. Can you find a visa.dll under Windows\System32 ? A possible option for you is to find the visa.dll file under the Keyisght folder and pass the path to this file when creating the resource maanger (or using a .pyvisarc file since passing the path directly seems to be problematic cf https://github.com/pyvisa/pyvisa/issues/442).
Do not worry about linux-gpib it does not make sense on windows. Another option is to install (using pip) gpib-ctypes but since pyvisa does not find the visa.dll I am not sure gpib-ctypes will find the gpib library.

killertoilet, 2019-07-16:
So I have the following dlls under Windows/System32:
visaext.dll
visa32.dll
visa64.dll
visaConfMgr.dll
visaUtilities.dll

I have also tried running resource manager with the following paths
C:\Program Files\IVI Foundation\VISA\Win64\ktvisa\ktbin\ktvisa32.dll
C:\Program Files\IVI Foundation\VISA\Win64\ktvisa\ktbin\visa32.dll
C:\Program Files\IVI Foundation\VISA\Win64\ktvisa\ktbin\ktvisaext.dll
C:\Program Files\IVI Foundation\VISA\Win64\ktvisa\ktbin\visaext.dll

not sure if these paths make sense.

MatthieuDartiailh, 2019-07-16:
Can you try running the following and post the output ? Your issue is weird because those dll should be detected by Python.  

```python
from visa import log_to_screen
from pyvisa.ctwrapper.highlevel import NIVisaLibrary
log_to_screen()
NIVisaLibrary.get_library_paths()
```

killertoilet, 2019-07-16:
C:\Users\person\Documents\N1092x Clock Input Failure>python PowerSwitching.py
2019-07-16 14:12:05,765 - pyvisa - DEBUG - No user defined library files
2019-07-16 14:12:05,769 - pyvisa - DEBUG - Automatically found library files: []
2019-07-16 14:12:05,770 - pyvisa - DEBUG - No user defined library files
2019-07-16 14:12:05,774 - pyvisa - DEBUG - Automatically found library files: []
2019-07-16 14:12:05,774 - pyvisa - DEBUG - Did not find NI binary
2019-07-16 14:12:05,779 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2019-07-16 14:12:05,795 - pyvisa - DEBUG - USBSession and USBRawSession were not imported No backend available.
2019-07-16 14:12:05,799 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2019-07-16 14:12:05,800 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2019-07-16 14:12:05,801 - pyvisa - DEBUG - pyvisa-py is available.
2019-07-16 14:12:05,801 - pyvisa - DEBUG - Created library wrapper for C:\Program Files\IVI Foundation\VISA\Win64\Bin\
2019-07-16 14:12:05,801 - pyvisa - DEBUG - Created ResourceManager with session 6939383
2019-07-16 14:12:05,801 - pyvisa - DEBUG - GPIB0::3::INSTR - opening ...
Traceback (most recent call last):
  File "PowerSwitching.py", line 34, in <module>
    supply = ConnectToInstrument(address)
  File "PowerSwitching.py", line 7, in ConnectToInstrument
    myInstrument = rm.open_resource(visaAddress)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa\highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa\resources\resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa\highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa-py\highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "C:\Users\person\AppData\Local\Programs\Python\Python37\lib\site-packages\pyvisa-py\sessions.py", line 170, in __init__
    raise ValueError(msg)
ValueError: Please install linux-gpib to use this resource type.
No module named 'gpib'
2019-07-16 14:12:05,809 - pyvisa - DEBUG - GPIB0::3::INSTR - closing
2019-07-16 14:12:05,812 - pyvisa - DEBUG - Closing ResourceManager (session: 6939383)

MatthieuDartiailh, 2019-07-16:
Can you also post your script ?

killertoilet, 2019-07-16:
import visa
from visa import log_to_screen
from pyvisa.ctwrapper.highlevel import NIVisaLibrary

def ConnectToInstrument(visaAddress):
	 rm = visa.ResourceManager('C:\\Program Files\\IVI Foundation\\VISA\\Win64\\Bin\\')
	 #rm = visa.ResourceManager()
	 myInstrument = rm.open_resource(visaAddress)
	 print(myInstrument.query('*IDN?'))
	 return myInstrument
	 
def QueryInstrument(instrument, query):
	 instrument.write(query)
	 return instrument.read()

def GetSystErr(instrument):
	 return QueryInstrument(instrument, ':SYST:ERR?')

def ClearSystErr(instrument):
	 loop = True
	 while loop:
			err = GetSystErr(instrument)  
			if ('No error' in err):
				 loop = False
			else:
				 print(err)

log_to_screen()
NIVisaLibrary.get_library_paths()

address = 'GPIB0::3::INSTR'

supply = ConnectToInstrument(address)

killertoilet, 2019-07-16:
I have tried a bunch of different paths in the resource manager field, that just happens to be the latest attempt.

MatthieuDartiailh, 2019-07-16:
Can you please try to install the master branch ? It won't fix the detection of the dll that fails, but it will let you specify properly the path to it. There is a bug in 1.9.1 that prevents it.
To install master you can do the following:
```
pip uninstall pyvisa
pip install https://github.com/pyvisa/pyvisa/tarball/master
```

killertoilet, 2019-07-16:
I used 'C:\\Program Files\\IVI Foundation\\VISA\\Win64\\ktvisa\\ktbin\\visa32.dll' as the path.

Here is the output from the log to screen:
2019-07-16 15:09:06,849 - pyvisa - DEBUG - Automatically found library files: [None, None, None, None, None]
2019-07-16 15:09:06,852 - pyvisa - DEBUG - No user defined library files
2019-07-16 15:09:06,971 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2019-07-16 15:09:06,972 - pyvisa - DEBUG - Created library wrapper for C:\Program Files\IVI Foundation\VISA\Win64\ktvisa\ktbin\visa32.dll
2019-07-16 15:09:08,542 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x0000022364E423C8>',) -> 0
2019-07-16 15:09:08,542 - pyvisa - DEBUG - Created ResourceManager with session 1
2019-07-16 15:09:08,560 - pyvisa - DEBUG - viParseRsrcEx(1, 'GPIB0::3::INSTR', 'c_ushort(1)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x0000022364E42448>, <ctypes.c_char_Array_256 object at 0x0000022364E424C8>, <ctypes.c_char_Array_256 object at 0x0000022364E42548>) -> 0
2019-07-16 15:09:08,561 - pyvisa - DEBUG - GPIB0::3::INSTR - opening ...
2019-07-16 15:09:08,623 - pyvisa - DEBUG - viOpen(1, 'GPIB0::3::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x0000022364E424C8>') -> 0
2019-07-16 15:09:08,623 - pyvisa - DEBUG - GPIB0::3::INSTR - is open with session 2
2019-07-16 15:09:08,624 - pyvisa - DEBUG - viWrite(2, b'*IDN?\r\n', 7, 'c_ulong(7)') -> 0
2019-07-16 15:09:08,624 - pyvisa - DEBUG - GPIB0::3::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2019-07-16 15:09:08,770 - pyvisa - DEBUG - viRead(2, <ctypes.c_char_Array_20480 object at 0x0000022364E424C8>, 20480, 'c_ulong(37)') -> 0
HEWLETT-PACKARD,E3631A,0,2.1-5.0-1.0

2019-07-16 15:09:08,776 - pyvisa - DEBUG - GPIB0::3::INSTR - closing
2019-07-16 15:09:08,776 - pyvisa - DEBUG - viDisableEvent(2, 1073709055, 65535) -> 0
2019-07-16 15:09:08,776 - pyvisa - DEBUG - viDiscardEvents(2, 1073709055, 65535) -> 1073676292
2019-07-16 15:09:08,777 - pyvisa - DEBUG - viClose(2,) -> 0
2019-07-16 15:09:08,777 - pyvisa - DEBUG - GPIB0::3::INSTR - is closed
2019-07-16 15:09:08,777 - pyvisa - DEBUG - Closing ResourceManager (session: 1)
2019-07-16 15:09:08,786 - pyvisa - DEBUG - viClose(1,) -> 0

killertoilet, 2019-07-16:
I got the *IDN? response! nice, so that seems to be working. Thank you very much, at least now I can get somehwere.

MatthieuDartiailh, 2019-07-16:
Happy you got it to work ! I am not sure why the dll is not detected, I have never something similar. I am planning a release in the coming weeks with this patch so you should be able to use a released version soon.

closed 2019-07-16 by killertoilet