pyvisa/pyvisa #446: ResourceManager.open_resource() does not time out when trying to connect to TCP resource

canol, 2019-08-16:
Hello, I am passing `open_timeout` parameter when connecting to a non-connected device's resource name like this:

`resource_manager.open_resource("TCPIP0::10.0.20.20::inst0::INSTR", open_timeout=100)`

so I expect it to throw timeout error after just 100 milliseconds but it waits for about 5 seconds until a VisaIOError is thrown.

PyVisa version: 1.9.1
Visa backend: NI
Python version: 3.6 64-bit

What might be the problem?

MatthieuDartiailh, 2019-08-16:
I can reproduce. However it seems to be a bad documentation issue on PyVisa side since the VISA specification says:
```
If the accessMode parameter requests a lock, then this parameter specifies the absolute time period (in milliseconds) that the resource waits to get unlocked before this operation returns an error.
```
Hence in the absence of any lock, this timeout does not have any effect. We need to fix PyVISA documentation and possibly pyvisa-py implementation to conform to this (but this is lower priority).

MatthieuDartiailh, 2019-08-18:
@canol do you want to make a PR for the documentation ?

canol, 2019-08-19:
@MatthieuDartiailh Ok, I'll do that tomorrow. By the way I briefly checked out the docs for viOpen function here:

http://zone.ni.com/reference/en-XX/help/370131S-01/ni-visa/viopen/

which doesn't mention the access mode. So, I won't edit the comments of ctwrapper/functions.py that calls viOpen, but change the comments everywhere else (In ResourceManager, VisaLibraryBase etc.)?

MatthieuDartiailh, 2019-08-19:
Interesting, NI may have gotten its documentation wrong. For reference, the VISA specification can be found here: http://www.ivifoundation.org/downloads/Class%20Specifications/vpp43_2017-10-17.pdf
I will post on NI forum to get to the bottom of this.

canol, 2019-08-19:
This is my first ever PR, let me know if I did something wrong or nonideal :)

MatthieuDartiailh, 2019-08-19:
NI forum ticket: https://forums.ni.com/t5/Instrument-Control-GPIB-Serial/viOpen-open-timeout-documentation/td-p/3961430

MatthieuDartiailh, 2019-09-18:
Closed by #451

closed 2019-09-18 by MatthieuDartiailh

MatthieuDartiailh, 2019-09-25:
Closed by #451