pyvisa/pyvisa #447: Conflict name with visa-python

MatthieuDartiailh, 2019-08-17:
The visa-python project (https://github.com/visa-sdk/visa-python) which gives access to the visa payment processing API defines in a very natural a visa package.

This can cause issues such as https://github.com/pyvisa/pyvisa/issues/392. I managed to find the root cause simply because I saw the issue happening on a coworker system.

As a consequence I believe we should deprecate the use of the visa module and migrate the command line tools to a different module (since we nowadays provide entry points to access those we can easily move them around).

@hgrecco @tivek @thliebig Do you have an opinion on this matter ?

In the meantime I will prepare a PR implementing the above changes and adding an entry to the FAQ about this issue.

changhexuefei, 2019-08-19:
As you might guess, there is a visa.py file in my project. I haven't noticed this problem, so after I modified this file, there is no 'AttributeError: module 'visa' has no Attribute 'ResourceManager'' error like this, thank you

MatthieuDartiailh, 2019-09-18:
Closed by #448

closed 2019-09-18 by MatthieuDartiailh