pyvisa/pyvisa #449: device.read() gives msgid assert error

torquil, 2019-08-18:
Hi!

I'm trying to read a response from a Siglent SDS1104X-E oscilloscope using python-visa on Debian Linux Unstable. I'm getting the following error when running the following python lines

oscilloscope.write('*IDN?')
response = oscilloscope.read()

Traceback (most recent call last):
  File "./get_trace_usb.py", line 28, in <module>
    response = oscilloscope.read()
  File "/usr/lib/python3/dist-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "/usr/lib/python3/dist-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python3/dist-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/usr/lib/python3/dist-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/usr/lib/python3/dist-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/usr/lib/python3/dist-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/usr/lib/python3/dist-packages/pyvisa-py/protocols/usbtmc.py", line 340, in read
    response = BulkInMessage.from_bytes(resp)
  File "/usr/lib/python3/dist-packages/pyvisa-py/protocols/usbtmc.py", line 110, in from_bytes
    assert msgid == MsgID.dev_dep_msg_in
AssertionError

Inserting print()-functions in the from_bytes() function to check the two variables that are compared by the assert():

print('msgid:', msgid)
print('MsgID.dev_dep_msg_in:', MsgID.dev_dep_msg_in)

gives the following output before the error message is printed:

msgid: 46
MsgID.dev_dep_msg_in: MsgID.request_dev_dep_msg_in

so the MsgID.dev_dep_msg_in is not even a number.

An ordinary oscilloscope.query('*IDN?') command works fine.

I'm using python-visa 1.9.1-1 togetyer with python 3.7.3-1.

Best regards,
Torquil Sørensen

MatthieuDartiailh, 2019-08-18:
This is really weird given that query is exactly this (cf https://github.com/pyvisa/pyvisa/blob/master/pyvisa/resources/messagebased.py#L583). Could you post the output of `pyvisa-info` ? I am curious to know what version of pyvisa-py you are running.

torquil, 2019-08-18:
I created two minimal test scripts. Apart from some minimal code to get access to the oscilloscope, the first scripts runs only query('*IDN?'). The second script runs only write('*IDN?') and then read().

Both work fine. The problem occurs if I run one piece of code after the other:

# Code A
oscilloscope.query('*IDN?')
# Code B
oscilloscope.write('*IDN?')
oscilloscope.read()

This generates the error message above at the line corresponding to read() in Code B. If I invert the order of the two parts, the Code B works fine, but then the error occurs at the query() in Code A. 

I don't seem to have a pyvisa-info command, but the Debian package management system says:

python3-pyvisa    1.9.1-1
python3-pyvisa-py 0.3.1-2 (this is the backend)

The oscilloscope is a Siglent SDS1104X-E with firmware version 8.1.6.1.26

- Torquil

MatthieuDartiailh, 2019-08-18:
Since that release we had a couple of changes made that may fix your issue (Not sure if Siglent does the same as Rigol but some of the fixes were to address the fact that Rigol does not implement USBTMC in a very rigorous manner it seems).
Since you are using your distribution Python, I am not sure how to do that cleanly but could you test against the master branch of pyvisa-py ?

torquil, 2019-08-19:
Thanks! With the development versoin of pyvisa and pyvisa-py it now works. I'm getting a warning message which is related to the Rigol problem.

```
/home/torquil/.local/lib/python3.7/site-packages/PyVISA_py-0.4.0.dev0-py3.7.egg/pyvisa-py/protocols/usbtmc.py:119: UserWarning: Unexpected MsgID format. Consider updating the device's firmware. See https://github.com/pyvisa/pyvisa-py/issues/20
  warnings.warn('Unexpected MsgID format. Consider updating the device\'s firmware. See https://github.com/pyvisa/pyvisa-py/issues/20')
```

It recommends to check for a newer firmware version, but I'm using the newest one. Do you think it is likely that Siglent will fix this and release a newer firmware if I report it and refer to this issue on GitHub for detailed info?

Btw, the pyvisa documentation shows how to install the development version, but when I follow the pyvisa-py docs to install the pyvisa-py development version, it only tells me how to install the pyvisa dev-version, at the bottom of this page:
`
https://pyvisa-py.readthedocs.io/en/latest/installation.html#using-the-development-version`
'

Thanks again! It is nice that pyvisa/pyvisa-py has a workaround for an "upstream problem".

- Torquil

closed 2019-08-19 by torquil

MatthieuDartiailh, 2019-08-19:
Siglent might fix it but I doubt it (Rigol issue is known for some time but was not adressed, mostly because most VISA providers actually cover it). Feel free to silence the warning.