pyvisa/pyvisa #450: Not getting complete data with read_raw()

torquil, 2019-08-19:
Hi! I'm using the development version of pyvisa and pyvisa-py together with a Siglent SDS1104X-E oscilloscope via USB on Debian Linux Unstable. I'm trying to retrieve the channel 1 waveform from the scope. When I used lxi-tools via ethernet this worked fine, but it caused my scope to freeze up quite often.

The command to request the waveform is:

oscilloscope.write('C1:WF? ALL')

which uses the SCPI command that worked fine with lxi-tools.

However, when I subsequently invoke read_raw(), I'm not getting the complete data. I only get around 60-70 bytes. By running read_raw() many times, I seem to get a sensible amount of data corresponding to the number of displayed points, but I'm not sure if the data is completely correct, because I'm having some trouble extracting variable values from the second or later segments.

When running many times manually, I have to manuall test each returned segment for the correct termination characters for the Siglent, which is b'\n\n'

So my question is: why is read_raw() stopping early, and what can I do to make it receive all the data?

The scope firmware is 8.1.6.1.26.

Best regards,
Torquil Sørensen

MatthieuDartiailh, 2019-08-20:
I am assuming you are trying to retrieve data that are transferred in binary format. If that is indeed the case I would suggest you to have a look at https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html
The basic issue is that `read_raw` will break on any termchar it encounters (not sure that visa may not break in that particular case even on single \n), which means you need to know how many bytes to expect to know for how long to read. `read_binary_values` takes care of that for you for the most common format and can be used for inspiration if you have something different (but if that is the case please file an issue to add that format to pyvisa).

torquil, 2019-08-20:
I am still convinced that something is wrong when retrieving data from the scope, whenever more than just a few characters are involved in the response, e.g. query('*IDN?') works fine. But perhaps it is best to focus on a simpler example to show the problem than the waveform retrieval. One SCPI-command that the scope supports is "TEMPLATE?". It will return a long text describing the format of the response to the "C1:WF? ALL" command. Using lxi-tools over ethernet I have no problems receiving the file.

I'm running the following line of code to work with the problem:

```
resources = visa.ResourceManager('@py')
oscilloscope = resources.open_resource(identifier, write_termination='\n', read_termination = '\n', query_delay=1.0)

print('\nSCPI command: TEMPLATE?')
template = oscilloscope.query('TEMPLATE?')
print('\ntemplate:\n', repr(template))
print('\nlen(template):', len(template))
print('\ntype(template):', type(template))

more1 = oscilloscope.read()
print('\nmore1:\n', repr(more1))
print('\nlen(nmore1):', len(more1))
print('\ntype(nmore1):', type(more1))

more2 = oscilloscope.read()
print('\nmore2:\n', repr(more2))
print('\nlen(nmore2):', len(more2))
print('\ntype(nmore2):', type(more2))

oscilloscope.close()
resources.close()
```

The "template" variable that is outputted is as follows:

'TMPL "\r\n/00\r\n\t000000              SDS1000X_E:  TEMPL             8 66 111\r\n;\r\n; Explanation of the formats of waveforms and their descriptorsiglent Digital Oscilloscopes,\r\n;     Software Release 5.1.3.8, 17/04/13.\r\n;\r\n; A descriptor and/or a waveform consists of one or sever'

This is truncated in the middle of a word when I compare with the correct output gotten from lxi-tools/ethernet. When I subsequently run read()-commands, I get more text, but unfortunately, some bytes are missing in between each segment, so I cannot even patch together all the data after multiple runs of read(). I have tried different read_termination values, but it doesn't make much of a difference.

The problem with running "TEMPLATE?" is similar to my problems when trying to get the waveform data through "C1: WF? ALL" as well as "C1: WF? DAT2" using binary methods. Also, the same problem occurs with the screendump function "SCDP".

An interesting thing is the warning from pyvisa that the read response does not end in the termination character. From the output from the program above:

/home/torquil/.local/lib/python3.7/site-packages/pyvisa/resources/messagebased.py:601: UserWarning: read string doesn't end with termination characters
  return self.read()

template:
 'TMPL "\r\n/00\r\n\t000000              SDS1000X_E:  TEMPL             8 66 111\r\n;\r\n; Explanation of the formats of waveforms and their descriptorsiglent Digital Oscilloscopes,\r\n;     Software Release 5.1.3.8, 17/04/13.\r\n;\r\n; A descriptor and/or a waveform consists of one or sever'

The warning "read string doesn't end with termination characters" occurs after each of the invokations of read().

Anyway, I decidde to run with visa.log_to_screen(), and got a bunch of suspicious (to me at least!) messages containing: "UnknownAttribute: Unknown attribute 1073676344 (0x3fff0038 - VI_ATTR_TERMCHAR_EN)". Here is the debug log:

```
2019-08-20 20:25:55,043 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2019-08-20 20:25:55,054 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2019-08-20 20:25:55,056 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2019-08-20 20:25:55,057 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib_ctypes'.
2019-08-20 20:25:55,057 - pyvisa - DEBUG - Created library wrapper for unset
2019-08-20 20:25:55,057 - pyvisa - DEBUG - Created ResourceManager with session 4994085
2019-08-20 20:25:55,057 - pyvisa - DEBUG - USB0::62700::60984::SDSMMEBX3R2053::0::INSTR - opening ...
2019-08-20 20:25:55,449 - pyvisa - ERROR - Unknown attribute 1073676310 (0x3fff0016 - VI_ATTR_SEND_END_EN)
Traceback (most recent call last):
  File "/home/torquil/.local/lib/python3.7/site-packages/PyVISA_py-0.4.0.dev0-py3.7.egg/pyvisa-py/sessions.py", line 425, in set_attribute
    return self._set_attribute(attribute, attribute_state)
  File "/home/torquil/.local/lib/python3.7/site-packages/PyVISA_py-0.4.0.dev0-py3.7.egg/pyvisa-py/usb.py", line 173, in _set_attribute
    raise UnknownAttribute(attribute)
pyvisa-py.sessions.UnknownAttribute: Unknown attribute 1073676310 (0x3fff0016 - VI_ATTR_SEND_END_EN)
2019-08-20 20:25:55,451 - pyvisa - ERROR - Unknown attribute 1073676312 (0x3fff0018 - VI_ATTR_TERMCHAR)
Traceback (most recent call last):
  File "/home/torquil/.local/lib/python3.7/site-packages/PyVISA_py-0.4.0.dev0-py3.7.egg/pyvisa-py/sessions.py", line 425, in set_attribute
    return self._set_attribute(attribute, attribute_state)
  File "/home/torquil/.local/lib/python3.7/site-packages/PyVISA_py-0.4.0.dev0-py3.7.egg/pyvisa-py/usb.py", line 173, in _set_attribute
    raise UnknownAttribute(attribute)
pyvisa-py.sessions.UnknownAttribute: Unknown attribute 1073676312 (0x3fff0018 - VI_ATTR_TERMCHAR)
[... 85 lines cut ...]
2019-08-20 20:25:56,473 - pyvisa - DEBUG - USB0::62700::60984::SDSMMEBX3R2053::0::INSTR - closing

SCPI command: TEMPLATE?

template:
 'TMPL "\r\n/00\r\n\t000000              SDS1000X_E:  TEMPL             8 66 111\r\n;\r\n; Explanation of the formarms and their descriptors on the\r\n; siglent Digital s,\r\n;     Software Release 5.1.3.8, 17/04/13.\r\n;\r\n; A descriptor an [... cut ...] r\n;\r\n;   <variable type> = string        up to 1name\r\n;                                   terminated byte\r\n;                       byte        08-bit silue\r\n;                       word        16-bit signe\r\n;                       long        32-bit signed\n;                      float        32-bit IEEE flovalue\r\n;\t\t\t\t\t\t\t\t\t with the format shown below\r\n;                                   31  30 .. 23   22 ... 0   bit po'

len(nmore2): 843

type(nmore2): <class 'str'>
```

MatthieuDartiailh, 2019-08-20:
I forgot you were using pyvisa-py on usb...
Since you instrument supports LXI, I assume you could connect to it over TCPIP rather than USB, no ? If that is the case can you run the same test over ethernet (I trust pyvisa-py more over ethernet than TCPIP). I would just like to know where we are having the issue (inside pyvisa or inside the USB backend).
I will try to have a look at the USB backend for the attribute error but I do not know it well so it may take some time.

torquil, 2019-08-20:
Well, I'm having even more problems with the TEMPLATE? command when using TCPIP. Simple commands such as *IDN? and TDIV? work. But whenever I send TEMPLATE? strange things happen. It all seems undeterministic. With visa.log_to_screen(), the complete correct response from the TEMPLATE? command can be seen in the log, but an error occurs so that it is not stored in the assignment variable "template". Other times it never appears in the log.

Here is the script:

```
import visa
visa.log_to_screen()

identifier = 'TCPIP::10.0.0.2::INSTR'
resources = visa.ResourceManager('@py')
oscilloscope = resources.open_resource(identifier, write_termination='\n', read_termination='\n')

oscilloscope.query('*IDN?')
oscilloscope.query('TDIV?')
oscilloscope.query('TEMPLATE?')

oscilloscope.close()
resources.close()
```

Here is an example of the visa.log_to_screen() output where the correct response is buried in the log (and below it another example where it is not):

```
2019-08-20 23:11:00,131 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2019-08-20 23:11:00,142 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2019-08-20 23:11:00,144 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2019-08-20 23:11:00,144 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib_ctypes'.
2019-08-20 23:11:00,145 - pyvisa - DEBUG - Created library wrapper for unset
2019-08-20 23:11:00,145 - pyvisa - DEBUG - Created ResourceManager with session 9324734
2019-08-20 23:11:00,145 - pyvisa - DEBUG - TCPIP::10.0.0.2::INSTR - opening ...
2019-08-20 23:11:00,145 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (10.0.0.2, 111)
2019-08-20 23:11:00,145 - pyvisa - DEBUG - Make call <PortMapperVersion.get_port: 3>, (395183, 1, 6, 0), <bound method PortMapperPacker.pack_mapping of <pyvisa-py.protocols.rpc.PortMapperPacker object at 0x7fcda598b850>>, <bound method Unpacker.unpack_uint of <pyvisa-py.protocols.rpc.PortMapperUnpacker object at 0x7fcda5927310>>
2019-08-20 23:11:00,146 - pyvisa - DEBUG - Sending record through <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.1', 42008), raddr=('10.0.0.2', 111)>: b'\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x01\x86\xa0\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x06\x00\x00\x00\x00'
2019-08-20 23:11:00,146 - pyvisa - DEBUG - Received record through <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.1', 42008), raddr=('10.0.0.2', 111)>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\xbe')
2019-08-20 23:11:00,146 - pyvisa - DEBUG - RawTCPClient: closing socket
2019-08-20 23:11:00,146 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (10.0.0.2, 958)
2019-08-20 23:11:00,147 - pyvisa - DEBUG - Make call 10, (353261987, 0, 10000, 'inst0'), <bound method Vxi11Packer.pack_create_link_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer object at 0x7fcda59e5590>>, <bound method Vxi11Unpacker.unpack_create_link_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker object at 0x7fcda5934f50>>
2019-08-20 23:11:00,147 - pyvisa - DEBUG - Sending record through <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.1', 42228), raddr=('10.0.0.2', 958)>: b"\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x15\x0eY\xa3\x00\x00\x00\x00\x00\x00'\x10\x00\x00\x00\x05inst0\x00\x00\x00"
2019-08-20 23:11:00,148 - pyvisa - DEBUG - Received record through <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.1', 42228), raddr=('10.0.0.2', 958)>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00')
2019-08-20 23:11:00,148 - pyvisa - DEBUG - TCPIP::10.0.0.2::INSTR - is open with session 7731179
2019-08-20 23:11:00,148 - pyvisa - DEBUG - Make call 11, (0, 2000, 10000, 136, b'*IDN?\n'), <bound method Vxi11Packer.pack_device_write_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer object at 0x7fcda59e5590>>, <bound method Vxi11Unpacker.unpack_device_write_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker object at 0x7fcda5934f50>>
2019-08-20 23:11:00,148 - pyvisa - DEBUG - Sending record through <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.1', 42228), raddr=('10.0.0.2', 958)>: b"\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\xd0\x00\x00'\x10\x00\x00\x00\x88\x00\x00\x00\x06*IDN?\n\x00\x00"
2019-08-20 23:11:00,149 - pyvisa - DEBUG - Received record through <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.1', 42228), raddr=('10.0.0.2', 958)>: bytearray(b'\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06')
[... cut ...]
2019-08-20 23:11:48,830 - pyvisa - DEBUG - Received record through <socket.socket fd=9, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.1', 42232), raddr=('10.0.0.2', 958)>: bytearray(b'\x00\x00\x00\x08\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2019-08-20 23:11:48,830 - pyvisa - DEBUG - RawTCPClient: closing socket
2019-08-20 23:11:48,831 - pyvisa - DEBUG - TCPIP::10.0.0.2::INSTR - is closed
2019-08-20 23:11:48,835 - pyvisa - DEBUG - Closing ResourceManager (session: 3963646)
```

The TEMPLATE? command works reliably when sent using lxi-tools over ethernet.

MatthieuDartiailh, 2019-08-20:
That is quite weird. First since you define `read_termination` as `\n` the query should return a partial result like the USB one. So I need to understand why it is not so, but I may need some time. Second it looks like your instrument may be slow to start answering (basically you should never see a hard socket timeout). 

Do you get in a more reliable manner the first case if you increase the timeout ?

MatthieuDartiailh, 2019-08-20:
Also note that VISA defines the termination character as a single character and hence having a redundant character does not really make sense... I will try to have a look at your instrument manual.

MatthieuDartiailh, 2019-08-20:
Do you have a programming manual ? The only manual I found does not have any remote command described ?

torquil, 2019-08-20:
This is the one I have been looking at:
[https://tinyurl.com/y2f22vjz](url)

It mentions on page 11 that it is applicable to the SDS1000X-E series of scopes, which includes my SDS1104X-E.

MatthieuDartiailh, 2019-08-21:
Currently siglent website (access easier through your link or through google) appears to have some certificate issue. I will look at it more later.

MatthieuDartiailh, 2019-08-28:
I looked at the documentation (that does not help that much). I need to look more at the issue you found in the TCPIP backend (easier to investigate than the USB one) but I am swamped in other stuff and that will take time.

bjaraujo, 2019-08-30:
It seems the pyvisa is terminating in the middle of the response as the response has several termination characters. 

TMPL "\r\n \r\n"\n

Maybe it should ignore the termination characters inside the quoted response?