pyvisa/pyvisa #452: Typo in pyvisa/resources/messagebased.py

torquil, 2019-08-20:
I think I found a bug in the current version of pyvisa/resources/messagebased.py:

On line 493, in read_binary_values(), it should say "header_fmt", not "header". Otherwise pyvisa crashes when passing header_fmt='empty' to read_binary_values().

Best regards,
Torquil Sørensen

MatthieuDartiailh, 2019-08-20:
Indeed ! Does correcting this allow you to use this method ? (Just want to make there not anotjer issue hidden, instruments using this protocol are rare so it is not easily tested).

torquil, 2019-08-20:
Yes, however there is another small issue there: The if-code block that is run when using format_fmt='empy' also needs to have data_length = data_points. Otherwise, line 499 fail because data_length is not set. And on line 465 in the description of header_fmt perhaps it should say something line: If 'empty' is specified, a value for data_points must be provided.

MatthieuDartiailh, 2019-08-20:
Good catch. However I believe something like this would be more general:
```python
        block = self._read_raw(chunk_size)

        if header_fmt == 'ieee':
            offset, data_length = util.parse_ieee_block_header(block)

        elif header_fmt == 'hp':
            offset, data_length = util.parse_hp_block_header(block,
                                                             is_big_endian)
        elif header_fmt == 'empty':
            offset = 0
            data_length = 0
        else:
            raise ValueError("Invalid header format. Valid options are 'ieee',"
                             " 'empty', 'hp'")

        # Allow to support instrument such as the Keithley 2000 that do not
        # report the length of the block
        data_length = data_length or data_points*struct.calcsize(datatype)
```
Once you confirm it works, I will make a PR and release a bugfix release ASAP.

Also technically you don't have to provide data_points, but it is unsafe to do so. The warning is there for this reason and should now be properly triggered.

MatthieuDartiailh, 2019-08-20:
I will also fix the warning message (noticed it was not accurate).

commit 20a2d260 (2019-08-28): Merge #454

454: Fix binary read for empty header r=MatthieuDartiailh a=MatthieuDartiailh

Closes #452
PR #454 (2019-08-28): Fix binary read for empty header
Closes #452
closed 2019-08-28 by bors[bot]