pyvisa/pyvisa #453: Unable to establish connection with Old USB Oscilloscope DS1M12 (Meilhaus Electronic)

attel-malagate, 2019-08-20:
I have been working with several instruments using NI and R&S VISA implementations via the pyvisa module. So far all of them respond to the easy to the usual code to connect via USB and i can control them with the usual SCPI commands:
```
RM = visa.ResourceManager('@py')
Resources= RM.list_resources()
ResourcesList= RM.list_resources_info()
```
However, this old pocket oscilloscope DS1M12 does not show up, i just get an empty tuple () and an empty list [] when doing:
`print(RM.list_resources())
print(RM.list_resources_info())`
respectively.
Nevertheless, the drivers for the device work correctly as i can run without any problem the provided datalogger and other software from the manufacturer. It read data as intended, however it is not detected using pyvisa.

Could anyone please provide some guidance on how to approach the connection with this device still using pyvisa?

MatthieuDartiailh, 2019-08-20:
Could you post the output of `pyvisa-info` ? I need to know on what OS you are running, because the pyvisa-py backend over USB requires some manual configuration. Depending on the OS. 
You mentioned NI and R&S VISA, do you have a particular need to use pyvisa-py as a backend ? I am asking because if you have access to those library you can use the `@ni` backend that is more stable.

attel-malagate, 2019-08-20:
Sure. Thanks for the fast reply BTW!

> Machine Details:
>    Platform ID:    Windows-10-10.0.18362-SP0
>    Processor:      Intel64 Family 6 Model 78 Stepping 3, GenuineIntel
> 
> Python:
>    Implementation: CPython
>    Executable:     c:\programdata\anaconda3\python.exe
>    Version:        3.7.3
>    Compiler:       MSC v.1915 64 bit (AMD64)
>    Bits:           64bit
>    Build:          Apr 24 2019 15:29:51 (#default)
>    Unicode:        UCS4
> 
> PyVISA Version: 1.9.1
> 
> Backends:
>    ni:
>       Version: 1.9.1 (bundled with PyVISA)
>       #1: C:\WINDOWS\system32\visa64.dll:
>          found by: auto
>          bitness: 64
>          Vendor: National Instruments
>          Impl. Version: 19922944
>          Spec. Version: 5244928
>       #2: C:\WINDOWS\system32\visa32.dll:
>          found by: auto
>          bitness: 64
>          Vendor: National Instruments
>          Impl. Version: 19922944
>          Spec. Version: 5244928
>    py:
>       Version: 0.3.1
>       ASRL INSTR:
>          Please install PySerial (>=3.0) to use this resource type.
>          No module named 'serial'
>       USB INSTR:
>          PyUSB does not seem to be properly installed.
>          Please refer to PyUSB documentation and
>          install a suitable backend like
>          libusb 0.1, libusb 1.0, libusbx,
>          libusb-win32 or OpenUSB.
>          No backend available
>       USB RAW:
>          PyUSB does not seem to be properly installed.
>          Please refer to PyUSB documentation and
>          install a suitable backend like
>          libusb 0.1, libusb 1.0, libusbx,
>          libusb-win32 or OpenUSB.
>          No backend available
>       TCPIP INSTR: Available
>       TCPIP SOCKET: Available
>       GPIB INSTR:
>          Please install linux-gpib to use this resource type.
>          No module named 'gpib'

attel-malagate, 2019-08-20:
In addition this is how the device manager is seeing the oscilloscope
[image]

attel-malagate, 2019-08-20:
> Could you post the output of `pyvisa-info` ? I need to know on what OS you are running, because the pyvisa-py backend over USB requires some manual configuration. Depending on the OS.
> You mentioned [...]

I do not need pyvisa-py in particular, just been using it as trial and error after reading some other issues here on github. I have tried '@ni' and the same empty tuple and list are returned

MatthieuDartiailh, 2019-08-20:
You are using pyvisa-py but without pyusb it cannot interact with a USB device. Since you do have NI installed I would suggest you just use that.

attel-malagate, 2019-08-20:
> 
> 
> You are using pyvisa-py but without pyusb it cannot interact with a USB device. Since you do have NI installed I would suggest you just use that.

I have been using NI and R&S for my work with other instruments. All working fine!. 
I installed pyvisa-py after trying to connect to this USB oscilloscope in order to follow one solution in another post and be able to use the '@py' backend. However it was not working either before the installation. I just uninstalled it again and things are just the same, no detection of the usb oscilloscope

MatthieuDartiailh, 2019-08-20:
Looking at the manufacturer website, it does not look like this instrument uses the VISA protocol (if it does not show up in NI-MAX I doubt it does). So pyvisa cannot help. However you may be able to access the oscilloscope or data logger interface by using ctypes to call the dll (looking at the manufacturer example code in C may help). 

I am closing since this is outside the scope of pyvisa

MatthieuDartiailh, 2019-08-20:
I am assuming this the instrument you are talking about https://www.connectiveperipherals.com/products/ds1m12.html. If not please provide a link to the manual and feel free to reopen.

closed 2019-08-31 by MatthieuDartiailh