pyvisa/pyvisa #455: Reduce bias towards National Instruments

MatthieuDartiailh, 2019-08-21:
Currently PyVISA suggests heavily that it works only with National Instruments implementation of VISA. This is not the case (PyVISA is known to work with NI, Keysight, Tek, R&S ?) and is regularly tested (by me at least) with NI and Keysight.

As a consequence it would be desirable to make PyVISA code and documentation less biased towards NI and more balanced. This was already suggested in #211 and it makes more and more sense.

Ideally the `ni` backend would be replaced with an `ivi` backend (`ni` would be available for backward compatibility). Any volunteer to work on that would be welcome.

DSTHub, 2019-08-23:
I work with R&S and Tek VISA, PyVISA works perfectly. 

[image]

Absolutely agree that  name 'ni' backend is not correct for different implementation of VISA, 'ivi' backend would be better as the most general and common definition of that type VISA's backend.

In addition, regarding to IVI foundation website the VISA specification are now being maintained by the IVI Foundation. 

I am ready to work on it.

MatthieuDartiailh, 2019-08-23:
If you want to start working on this feel free to do so.

hgrecco, 2019-08-23:
Sadly, not all libraries work the same and the user might want to select a specific one. While providing the dll path is an option, this is not cross platform (that is why pyvisa has the machinery to autodected libraries). So I think we can make this `ni` -> `ivi` suggestion, also a way to allow vendor selection.

DSTHub, 2019-08-23:
In addition, many functions were originally written as a wrapper specifically for NI-VISA.

I think you need to carefully describe this point in the documentation. 

Namely: IVI-VISA is not only NI-VISA or even any other VISA implementation, it depends on VISA, that you have installed in your system.

I've made some fix, but not sure is it enough. #456

MatthieuDartiailh, 2019-08-23:
@hgrecco are suggesting to allow things like `@ni`, `@keysight`, `@tek`, `@r&s` and have pyvisa try to find a matching library? That might be tough since VISA already handle the notion of primary internally (at least on windows) so I am not sure how reliable that would be. (Also from memory you always get a single visa.dll under System32 and the others remain under the manufacturer folder and are not as easy to discover.)
I will review the PR this week-end.

MatthieuDartiailh, 2019-09-18:
Addressed in #456

closed 2019-09-18 by MatthieuDartiailh

MatthieuDartiailh, 2019-09-25:
Closed by #456