pyvisa/pyvisa #457: python crash due to ibutil.c: conf_lock_board

bserra, 2019-08-28:
I have been using pyvisa to communicate with GPIB instruments with the help of a raspberry.
Recently I configured another raspberry to control one single instrument through GPIB using the NI GPIB to usb adapter. 
I changed the gpib.conf file and communication seems to be working at first.
Then, randomly, I get the following error message:

```
python: ibutil.c:382: conf_python: ibutil.c:382: conf_lock_board: Assertion `conf->has_lock == 0' failed. 
Aborted       
```                                                                                      
And the program crash. When I try to reconnect everything is working until I stumble into this error again.
I have no idea what causes it and what I could do to at least catch it with a try... except.

Any idea of what I could do to handle this?

bserra, 2019-08-29:
Thanks for the suggestion!

MatthieuDartiailh, 2019-09-18:
@bserra any follow up to this ?

bserra, 2019-09-19:
Yes, I contacted the mailing list of linux-gpib. It is due to me doing concurrent access: two threads were trying to access the same gpib ressource. Did not encounter the issue since I did a correction.
https://sourceforge.net/p/linux-gpib/bugs/71/

MatthieuDartiailh, 2019-09-25:
Sounds good I will close then. Feel free to reopen if necessary.

closed 2019-09-25 by MatthieuDartiailh