pyvisa/pyvisa #458: Raspberry Pi 4 VL805 bulk transfer problem

charkster, 2019-09-04:
I am having a problem with query_binary_values() and read_raw() when downloading waveform or csv data from my Rigol DS1054Z. These commands run just fine on my RPi3, RPi2 and PC, but with the latest RPi4 (and the VL805) I am seeing errors.

Here is what I see from query_binary_values():

```
  File "test_print_screen.py", line 6, in <module>
    scope.write_screen_capture()
  File "/home/pi/python_code/rigol/rigol_ds1054z.py", line 120, in write_screen_capture
    raw_data_list = self.oscilloscope.query_binary_values(':DISP:DATA? ON,OFF,PNG')
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 703, in query_binary_values
    data_points, chunk_size)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 481, in read_binary_values
    block = self._read_raw(chunk_size)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 387, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 338, in read
    response = BulkInMessage.from_bytes(resp)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 108, in from_bytes
    assert msgid == MsgID.dev_dep_msg_in
AssertionError
```

Here is what I see from read_raw():

Traceback (most recent call last):
  File "test_print_screen.py", line 6, in <module>
    scope.write_screen_capture()
  File "/home/pi/python_code/rigol/rigol_ds1054z.py", line 124, in write_screen_capture
    raw_data = self.oscilloscope.read_raw()[11:] # strip off first 11 bytes
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 364, in read_raw
    return bytes(self._read_raw(size))
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 387, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 345, in read
    ret = self.sessions[session].read(count)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line 111, in read
    usb.USBError)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py", line 477, in _read
    current = reader()
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line 105, in <lambda>
    return self._read(lambda: self.interface.read(count),
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 338, in read
    response = BulkInMessage.from_bytes(resp)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 108, in from_bytes
    assert msgid == MsgID.dev_dep_msg_in
AssertionError

If I wait one second after sending my write(':DISP:DATA? ON,OFF,PNG') command, I don't get any errors, but I only download 41bytes of data (when expecting 30KB).

Is this an issue with the VL805?
I suspect this is not a pyvisa issue, but a problem with bulk transfers with the VL805.

MatthieuDartiailh, 2019-09-04:
You can try running using pyvisa-py master. Rigol instruments are known to implement the USBTMC protocol in a nor fully rigorous manner and the master can circumvent that. Let me know if you get it to work.

charkster, 2019-09-04:
Hi Matthieu,

Thank you for the quick response. I installed the master version of
pyvisa-py and things actually got worse. Still had the issue on the RPi4,
but RPi3 broke also.
Was I supposed to upgrade to the master version of pyvisa too?

I will just keep using my older raspberry pi 3 as pyvisa-py is working just
fine with it.

Thank you for this great software, it saves me a lot of time.
-Chark Stanford

MatthieuDartiailh, 2019-09-04:
Could you post the traceback you get on your raspberry 3 ?

charkster, 2019-09-05:
Hi Matthieu,

Here is the visa.log_to_screen() traceback for the successful case (RPi3):

2019-09-04 22:17:50,002 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - reading 20480 bytes (last
status <StatusCode.success_max_count_read: 1073676294>)
Scope information: RIGOL TECHNOLOGIES,DS1104Z,DS1ZA192107675,00.04.04.SP3
2019-09-04 22:17:53,022 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - reading 20480 bytes (last
status <StatusCode.success_max_count_read: 1073676294>)
Wrote screen capture to filename "rigol_2019-09-04_22-17-53.png"
2019-09-04 22:17:58,126 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - closing
2019-09-04 22:17:58,128 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - is closed

Here is my python script:

class rigol_ds1054z:

# Constructor
def __init__(self, debug=False):
resources = visa.ResourceManager('@py')
# insert your device here
# resources.list_resources() will show you the USB resource to put below
self.oscilloscope =
resources.open_resource('USB0::6833::1230::DS1ZA192107675::0::INSTR')
self.debug = debug
if (debug):
visa.log_to_screen()

def print_info(self):
self.oscilloscope.write('*IDN?')
fullreading = self.oscilloscope.read_raw()
readinglines = fullreading.splitlines()
print("Scope information: " + readinglines[0])
* time.sleep(2)*

 def write_screen_capture(self, filename=''):
self.oscilloscope.write(':DISP:DATA? ON,OFF,PNG')
time.sleep(1)
raw_data = self.oscilloscope.read_raw()[11:] # strip off first 11 bytes
# save image file
if (filename == ''):
filename = "rigol_" + datetime.datetime.now().strftime("%Y-%m-%d_%H-%M-%S")
+".png"
fid = open(filename, 'wb')
fid.write(raw_data)
fid.close()
print ("Wrote screen capture to filename " + '\"' + filename + '\"')
* time.sleep(5)*

from rigol_ds1054z import rigol_ds1054z
scope = rigol_ds1054z(debug=True)
scope.print_info()
scope.write_screen_capture()

-Chark

charkster, 2019-09-05:
Hi Matthieu,

Here is the traceback for the unsuccessful RPi4 case (same script as the
RPi3).

2019-09-04 22:36:18,442 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - reading 20480 bytes (last
status <StatusCode.success_max_count_read: 1073676294>)
Scope information: RIGOL TECHNOLOGIES,DS1104Z,DS1ZA192107675,00.04.04.S
2019-09-04 22:36:20,459 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - reading 20480 bytes (last
status <StatusCode.success_max_count_read: 1073676294>)
Traceback (most recent call last):
  File "test_print_screen.py", line 6, in <module>
    scope.write_screen_capture()
  File "/home/pi/python_code/rigol/rigol_ds1054z.py", line 122, in
write_screen_capture
    raw_data_list = self.oscilloscope.query_binary_values(':DISP:DATA?
ON,OFF,PNG')
  File
"/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py",
line 715, in query_binary_values
    data_points, chunk_size)
  File
"/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py",
line 493, in read_binary_values
    block = self._read_raw(chunk_size)
  File
"/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py",
line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 345,
in read
    ret = self.sessions[session].read(count)
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 111, in
read
    usb.USBError)
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/sessions.py", line 477,
in _read
    current = reader()
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 105, in
<lambda>
    return self._read(lambda: self.interface.read(count),
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py",
line 338, in read
    response = BulkInMessage.from_bytes(resp)
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py",
line 108, in from_bytes
    assert msgid == MsgID.dev_dep_msg_in
AssertionError
2019-09-04 22:36:21,313 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - closing
2019-09-04 22:36:21,316 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - is closed

-Chark

MatthieuDartiailh, 2019-09-05:
Could you also post the traceback you get on pyvisa-py master ? I would prefer to avoid breaking your code with the new release.

charkster, 2019-09-06:
Hi Matthieu,

Here is the traceback on my good RPi3 after upgrading pyvisa-py to
0.4.0.dev0.

2019-09-05 19:36:04,574 - pyvisa - ERROR - Unknown attribute 1073676310
(0x3fff0016 - VI_ATTR_SEND_END_EN)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py",
line 377, in get_attribute
    return self._get_attribute(attribute)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line
160, in _get_attribute
    raise UnknownAttribute(attribute)
UnknownAttribute: Unknown attribute 1073676310 (0x3fff0016 -
VI_ATTR_SEND_END_EN)
2019-09-05 19:36:04,581 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - reading 20480 bytes (last
status <StatusCode.success_max_count_read: 1073676294>)
2019-09-05 19:36:04,583 - pyvisa - ERROR - Unknown attribute 1073676312
(0x3fff0018 - VI_ATTR_TERMCHAR)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py",
line 377, in get_attribute
    return self._get_attribute(attribute)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line
160, in _get_attribute
    raise UnknownAttribute(attribute)
UnknownAttribute: Unknown attribute 1073676312 (0x3fff0018 -
VI_ATTR_TERMCHAR)
2019-09-05 19:36:04,584 - pyvisa - ERROR - Unknown attribute 1073676344
(0x3fff0038 - VI_ATTR_TERMCHAR_EN)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py",
line 377, in get_attribute
    return self._get_attribute(attribute)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line
160, in _get_attribute
    raise UnknownAttribute(attribute)
UnknownAttribute: Unknown attribute 1073676344 (0x3fff0038 -
VI_ATTR_TERMCHAR_EN)
Scope information: RIGOL TECHNOLOGIES,DS1104Z,DS1ZA192107675,00.04.04.SP3
2019-09-05 19:36:06,608 - pyvisa - ERROR - Unknown attribute 1073676310
(0x3fff0016 - VI_ATTR_SEND_END_EN)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py",
line 377, in get_attribute
    return self._get_attribute(attribute)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line
160, in _get_attribute
    raise UnknownAttribute(attribute)
UnknownAttribute: Unknown attribute 1073676310 (0x3fff0016 -
VI_ATTR_SEND_END_EN)
2019-09-05 19:36:07,611 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - reading 20480 bytes (last
status <StatusCode.success_max_count_read: 1073676294>)
2019-09-05 19:36:07,613 - pyvisa - ERROR - Unknown attribute 1073676312
(0x3fff0018 - VI_ATTR_TERMCHAR)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py",
line 377, in get_attribute
    return self._get_attribute(attribute)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line
160, in _get_attribute
    raise UnknownAttribute(attribute)
UnknownAttribute: Unknown attribute 1073676312 (0x3fff0018 -
VI_ATTR_TERMCHAR)
2019-09-05 19:36:07,614 - pyvisa - ERROR - Unknown attribute 1073676344
(0x3fff0038 - VI_ATTR_TERMCHAR_EN)
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py",
line 377, in get_attribute
    return self._get_attribute(attribute)
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa-py/usb.py", line
160, in _get_attribute
    raise UnknownAttribute(attribute)
UnknownAttribute: Unknown attribute 1073676344 (0x3fff0038 -
VI_ATTR_TERMCHAR_EN)
Wrote screen capture to filename "rigol_2019-09-05_19-36-07.png"
2019-09-05 19:36:12,710 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - closing
2019-09-05 19:36:12,713 - pyvisa - DEBUG -
USB0::6833::1230::DS1ZA192107675::0::INSTR - is closed

-Chark

bjaraujo, 2019-09-07:
Using the master branch does it help if you change chunk_size like so:

MatthieuDartiailh, 2019-09-11:
Can you try https://github.com/pyvisa/pyvisa-py/pull/207 ? It should the attribute issue on the USB backend.

charkster, 2019-09-12:
Hi Matthieu,
I installed #207 and on my RPi3 and it works just fine (like it did before), but on my RPi4 I am seeing a different error now. I also turned-on the pyusb logfile and see that the RPi4 is not detaching the kernel driver. I have tried to detach it using usb.core, but have been unsuccessful (I can detach the usb port on my Ubuntu x86 PC with the same code).
It's odd that simple SCPI commands work fine, but any time I am downloading killobytes of PNG data or binary oscilloscope settings I get the unexpected MsgID.

Specifying the chunk_size also did not help (Thx bjaraujo).

2019-09-11 21:00:10,494 - pyvisa - DEBUG - USB0::6833::1230::DS1ZA192107675::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.4.0.dev0-py2.7.egg/pyvisa-py/protocols/usbtmc.py:119: UserWarning: Unexpected MsgID format. Consider updating the device's firmware. See https://github.com/pyvisa/pyvisa-py/issues/20
  warnings.warn('Unexpected MsgID format. Consider updating the device\'s firmware. See https://github.com/pyvisa/pyvisa-py/issues/20')
Traceback (most recent call last):
  File "test_print_screen.py", line 15, in <module>
    scope.print_info()
  File "/home/pi/python_code/rigol/rigol_ds1054z.py", line 22, in print_info
    fullreading = self.oscilloscope.read_raw()
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 377, in read_raw
    return bytes(self._read_raw(size))
  File "/home/pi/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.4.0.dev0-py2.7.egg/pyvisa-py/highlevel.py", line 329, in read
    ret = self.sessions[session].read(count)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.4.0.dev0-py2.7.egg/pyvisa-py/usb.py", line 128, in read
    USBTimeoutException)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.4.0.dev0-py2.7.egg/pyvisa-py/sessions.py", line 478, in _read
    current = reader()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.4.0.dev0-py2.7.egg/pyvisa-py/usb.py", line 108, in _usb_reader
    return self.interface.read(count)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.4.0.dev0-py2.7.egg/pyvisa-py/protocols/usbtmc.py", line 419, in read
    response = BulkInMessage.from_bytes(resp)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.4.0.dev0-py2.7.egg/pyvisa-py/protocols/usbtmc.py", line 120, in from_bytes
    return BulkInMessage.from_quirky(data)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.4.0.dev0-py2.7.egg/pyvisa-py/protocols/usbtmc.py", line 135, in from_quirky
    if ';' in str(data):
UnicodeDecodeError: 'ascii' codec can't decode byte 0xda in position 3: ordinal not in range(128)
2019-09-11 21:00:10,562 - pyvisa - DEBUG - USB0::6833::1230::DS1ZA192107675::0::INSTR - closing
2019-09-11 21:00:10,565 - pyvisa - DEBUG - USB0::6833::1230::DS1ZA192107675::0::INSTR - is closed

bjaraujo, 2019-09-13:
It would be interesting to print those first 104 bytes to see the header. 
What is the inst.timeout value? 
Does this issue also happen using NI backend?

MatthieuDartiailh, 2019-09-13:
You can have a look at https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html but it looks lile there is still something wrong in the usb backend.

charkster, 2019-09-14:
@bjaraujo You are right about the 104 bytes being the header here they are:
<89>PNG
^Z
^@^@^@^MIHDR^@^@^C ^@^@^A<E0>^H^B^@^@^@<D2>e<9E><A2>^@^@^@^LtEXtAuthor^@RIGOL爌<C9>^@^@^@^QtEXtSource^@DSZ series?Z^]<97>^@^@^@)tEXtDescription

Here is the first 1000 bytes of a successful transfer:
<89>PNG
^Z
^@^@^@^MIHDR^@^@^C ^@^@^A<E0>^H^B^@^@^@<D2>e<9E><A2>^@^@^@^LtEXtAuthor^@RIGOL爌<C9>^@^@^@^QtEXtSource^@DSZ series?Z^]<97>^@^@^@)tEXtDescription^@DSZ_Printing equipment output<C1><A5>5<C1>^@^@^@^\tEXtCopyright^@RIGOL TECHNOLOGIES<E3>^Ta<AF>^@^@<86><9F>IDATx^A<ED><BD>-<BC><E4>ؑ훠@
<81>^F^E^Z^\Р@<83>^B^F^E^L
48<A0>A^C<83>^B^C^NhP<C0>`<C0>^@<83>^K^L^F^T^X0<E0><82>^A^O^X^L^X0<A0><81><81><C1>^C^F^F^M.00^X<F0><C0><80>^K^F^p<C1><C0>^B^M<DE>RF<E6><CA>
<C8><FD>!<85><B6>>2SZ<FA><C9><EA>P숽c<FF><B5>u<B4><AC><93><95><E7><F0>~g<DB>A<9B>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88>
<80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><80>^H<88><C0>^L^D<BE>>^\<DE>
^_^N<DF>^]^NϷ<DB>1:j@%<DA>D@^DD@^DD@^DD<E0><B1>  <BC>:^\<BE>=J<AB>;<99>^Fd^V<EA>AU<DA>D@^DD@^DD@^DD<E0>Q |}TW^P4<FF>z8<FC><F7><E1><F0><CB>
<E1><F0><E5>^V;<C6><C5><E8><A8>^A<95>|<A7><F7>X<8F><BA><9A>T<B7>^H<88><80>^H<88><80>^H<88><C0><91><C0><FB><E3><F1><FF>9^\<FE>z8<FC><D3><E1>
<F0>^O<87><C3>oo<B1>c\<8C><8E>^ZP   <B6><F7><C7><E3><E1><F0><C3>o~<F8>=<B6>^?<DC><CA><FE><FB><DF>cF<A7><B9>=<C8>^?<B6>v     6<B3><96>4<91>
<B1>^D<C6><DC>}Z<F6><8F><FD>Sw̵<C6>Ob]<EE>9/<F7>H<F8>^O<F2>$l+^S<AF><8B><B0><FD><E5>p<F8><C7><C3><E1>ӭwԀJ<B0>}<D7>^]~<F8><E1><87>O<BF><FD>
<F4><FC><FC><FC><FE>כ<F8><82><B2>_<BF><C7>\0#̫<9B><DE>y<FB><F2><CB>^W<EE><E6>KN<E1>L<8=g<9F><FE>K^O#<E9>I"y:^X<80>ȭ]<82>M<AC>#M<A2><85>@<E5>
<EE><E3><ED><E0>^M-<FB>^V<C2><F7><93>3<E6>Z<EB><A7><DC><CC><D7>m$|^?<DF>m<D1>~>N<EA><8F>ǷV?^^^N<B7><DD><F1><F2>^L<95>`{<EE>^N/?<BE>|<F8><EE><C3>̗<FF><D6><DD>aF<98>W7<BD><F3><96>^H^]^?jvѓ8<F3>Ht<EF>cΣ]<FE><DB><DF>jq^S/<C1><97><CF><CF><F9>~<EB>+<A0><F1><F7>K <BF><FB>.<F7><83><B3>&.
<FB><FD>򽧙^G<AF>5.<BB>.<F7><EC><D7>-^D<FF><E7>?^\z<F6>?<FD><CF><C3><E7><DF>^^<DE>^?<EB><EE>K<BC>^@<FA>M9<E5>*<E8>^\<F6><E5>o^G<EC><FF><FD>
<F3>^A]!^Q<BD>a<B8>7_%<B1><A7>ӷO<87>^?<F9><DD><E1><AF><FF>ޥ؎<E0>߽<A4><F1><C5>^B<90>Xݞ<8F>-<FF><8E>Uv^_;*<C1><F6><DC>^]><FE><DD>G\<F8>w<BF>z
^W߻<B4><FB><DE>0#̋5<E6>*<C7>{<CC>.z^Rg^^<C9>!`<A0><D5>v:<99><9E>71^FF<C3>%<F0>^W+WW<F0><F8>^@<D9>"<B0>&<81><E4><EE><F3>K<DD><DB>^S<97><BD>
<CD><C8>w({}^B<C1>k<8D><U+0082><97><FB><ED>۷<AF>_<BF>^^?"<8F>2"<E0>^@<91>-<FE>^P|J<99>~<E3>^O<FF>x!^@<91>T^Lf^D^D<D9>^?<FD><A9>^\c<89>PH<F9>^F<85>T<EC>^V<CE><FF><F3><97>N<9C>q+^V^@)Vݞ<8F>-<FF><8A>Uv^_;*<C1><F6><DC>^]>~<FC>8<EA><87><EF><D3>7O]<DA>}o<98>^Q<E6><C5>^Z<A9>o<BC><E2>
<F1><AD><B0><D9>D<DB><<F9><D1>^B|<S`x<DB><E7>&MvjǱ<97> <B9>^<90>S<BE>7<D8>^RX        "<9D><AE>L<C0><DF>}<C9><E2><E4><E9><C4>e<8F>^Y=<C4><CF>"<CE>w<93>^F<AE>B<E4>Zc<EE><91>˭^K^Z\$^@ew<F4>0<FC><DF><FF><F6><F0><D3><FF><BC>(<9B><FF><FC><D3><E1><CF>^?8<ED><90>5^<F1> Ҷ<E7><F7>^Gؾ^U=
<B0>^U<EA>

MatthieuDartiailh, 2019-09-16:
If you know beforehand how many bytes to expect and do not need to convert them the more straightforward way is actually `read_bytes` (I not realized you were reading a PNG, often binary transfer is used to transfer actual data and the header let you know what to expect, it does not appear to be so here). I need to find the time to dive into the USBTMC specification to figure out if we are doing something improperly here.

charkster, 2019-09-18:
I installed wireshark and have the log files for the good and bad transfers. I can see the data differences from the start of the communication between the host and the oscilloscope. It looks like it could be in pyusb or libusb. Is there any pyusb connection options that could be tried to help this?

bjaraujo, 2019-09-18:
I also thought this binary data had a header describing first the amount of bytes transferred but this data is has no header only the raw PNG data. In this case, I guess you have to be careful with the termination character as it will stop when it finds it. It may work or not depending on if it finds it or not. You could try setting inst.read_termination = None and then reading until it has no more bytes to read.

MatthieuDartiailh, 2019-09-18:
Could you share those wireshark results ?

charkster, 2019-09-19:
> I also thought this binary data had a header describing first the amount of bytes transferred but this data has no header, only the raw PNG data. In this case, I guess you have to be careful with the termination character as it will stop when it finds it. It may work or not depending on if it finds it or not. You could try setting inst.read_termination = None and then reading until it has no more bytes to read.

I strip the first 11 bytes when I do a raw_read (as seen above in my "write_screen_capture" function)... the amount of bytes is probably listed in there. I have been relying on pyvisa's raw_read to skip the termination characters and just pass on the entire data that the scope sends. It has been working just fine using my Raspberry Pi 3.

Here are the wireshark files. 
I use this display filter for the good RPi3:
usb.bus_id==001 and usb.device_address==007
I use this display filter for the bad RPi4:
usb.bus_id==001 and usb.device_address==003

[wireshark_print_screen_bad_rpi4.pcapng.gz](https://github.com/pyvisa/pyvisa/files/3628991/wireshark_print_screen_bad_rpi4.pcapng.gz)
[wireshark_print_screen_good.pcapng.gz](https://github.com/pyvisa/pyvisa/files/3628992/wireshark_print_screen_good.pcapng.gz)

I have noticed that there is a "SET ADDRESS Request" in the good logfile. I don't know if that is related to when pyusb does a detach_kernel_driver command.

Thanks for your help.

charkster, 2019-10-03:
I am thinking that this is a pyvisa-py issue as I have no problems running on WIndows 10 using the NI backend.

closed 2019-10-03 by charkster