pyvisa/pyvisa #459: :MMEMory:STORe:TRACe command doesn't work with Pyvisa

pawelzett, 2019-09-24:
Dear all,

I've got a problem with terminating one command. I'd like to save measurment on Antritsu spectrum analyzer with command `:MMEMory:STORe:TRACe\s0,\s"new"`. This one works when I'm sending it thorugh VISA interactive tool. Now I need to send it via pyvisa. Below you can find my code:

`import pyvisa
import numpy as np

rm = pyvisa.ResourceManager()
rm.list_resources()
inst = rm.open_resource('TCPIP0::10.44.128.110::inst0::INSTR')
print(inst.query("*IDN?"))
"Anritsu,MS2721B/9/44/35/65,943063,1.51"

print(inst.write(":MMEMory:STORe:TRACe\s0,\s"new")") 
SyntaxError: invalid syntax
`
I'm sending :MMEMory:STORe:TRACe\s0,\s"new" via Visa interactive tool and it works. Can you help me with this problem? Thank you!

pawelzett, 2019-09-24:
I tried also this and still nothing: 

```
values = list(range(100))
inst.write_ascii_values(':MMEMory:STORe:TRACe\s new,\s', values)
```

MatthieuDartiailh, 2019-09-24:
The issue is that you are using the same quote to delimit your string as the one you need to use in the message. The proper way is :
```python
print(inst.write(':MMEMory:STORe:TRACe\s0,\s"new")') 
```

pawelzett, 2019-09-24:
I still can't save this file:
[image]
 Maybe I'm doing something wrong. Thank you.

MatthieuDartiailh, 2019-09-24:
My bad I misplaced the closing `'` with respect to the parenthesis ! It should be :
```python
print(inst.write(':MMEMory:STORe:TRACe\s0,\s"new"')) 
```

pawelzett, 2019-09-25:
Still doesn't work, when I open telnet connection in cmd I can see this:
[image]

pawelzett, 2019-09-25:
Ok, I got this it shoud be `print(inst.write(':MMEMory:STORe:TRACe 0,"new"'))`
Thank you for your help!

pawelzett, 2019-09-25:
One more thing I missed, I try to save measurment on my pc using ftplib, but at the first time it can't see the file, which was just saved on the device. When i run this code next time it saves last measuremnt, not the new one I want. Also maybe you know how to name this file by actual date and time? Sorry for all these questions, but I'm new in python :) Thank you.
```

import pyvisa
from ftplib import FTP
import time
from datetime import datetime

rm = pyvisa.ResourceManager()
inst = rm.open_resource('TCPIP0::10.44.128.110::inst0::INSTR')

print(inst.query("*IDN?")) #printing analyzer name, model etc. to check if it is connected correctly

print(inst.write(':MMEMory:STORe:TRACe 0, "name"')) #Downloading measurment file to device

ftp = FTP('10.44.128.120')
ftp.login()
ftp.retrbinary('RETR name.spa', open('name.spa', 'wb').write)
ftp.quit()

```

pawelzett, 2019-09-26:
Ok, I got the save issue from ftp. I just added:
`time.sleep(5)` befroe the ftp lines. I still don't know how to save the file with name as actual date.

MatthieuDartiailh, 2019-09-26:
You can just do something like:
```python
import datetime

print(inst.write(':MMEMory:STORe:TRACe 0, "%s"' % datetime.date.today()))
```

I am closing since PyVISA is working as expected.

closed 2019-09-26 by MatthieuDartiailh

pawelzett, 2019-09-27:
Thank you! Have a nice day :)